tsduck/tsduck #1533: Not get a valid Json output

TheRitikk, 2024-10-08:
When i run this cmd 
**_tsp -v -I hls http://192.168.1.66/files/stream/streaming_channel_0.m3u8 -P analyze --json -O outputhls.json_**     
Here i have multiaudio hls that time this cmd not give me all audio autput in my json file
for single audio also have not all output

lelegard, 2024-10-08:
Hi @TheRitikk 

Could you be more specific? What is the exact command you used? (the one you provided is syntactically incorrect, so it cannot be the one).

If you think that the analysis is incorrect, we need a sample of the input file. Otherwise, it is impossible to decide if this is correct or not.

TheRitikk, 2024-10-08:
Hii @lelegard 
when i run this cmd -> "tsp -I hls http://192.168.1.66/files/stream/streaming_channel_0.m3u8 -P analyze -i 1 --json -o hlsdata.json -O drop >> hlsdata.json"               
that i store hls data, here if multi audio hls, that time not get all audio output with it's details only get one audio output

closed 2024-10-23 by TheRitikk