tsduck/tsduck #1535: [Q] Unable to include tsduck header in C++.

the-atharva, 2024-10-12:
I tried simply including tsduck header file & am getting following error in the my code: 

`main.cc:2:10: fatal error: tsduck.h: No such file or directory
    2 | #include "tsduck.h"
      |          ^~~~~~~~~~
compilation terminated.`

I am new to using 3rd party libraries, build systems & package managers in C++. I am working on project which uses CMake build system & it was unable to find tsduck package. So I thought maybe there is problem with my cmake commands & tried simply including tsduck in main file in different program.

lelegard, 2024-10-12:
Hi @the-atharva 

Please read the chapter ["developing applications with TSDuck"](https://tsduck.io/download/docs/tsduck-dev.html#chap-application) in the TSDuck developer's guide.

TSDuck uses the same method as many other third-party libraries: a dedicated command `tsconfig` ([doc here](https://tsduck.io/download/docs/tsduck.html#tsconfig-ref)) generates the right compilation and link options.

the-atharva, 2024-10-13:
Hi @lelegard 

I referred to the developers guide & documentation. I am trying to run the code through terminal
For simply including tsVersion header files & printing the versions the code works

```
#include <iostream>
#include "tsVersion.h"
using namespace std;
int main(){
    cout<<"Major version: "<<TS_VERSION_MAJOR<<endl;
    cout<<"Minor version: "<<TS_VERSION_MINOR<<endl;
    return 0;
}
```
This is what I get
```
a@peace:~/Projects/tsduckTrial$ g++ $(tsconfig --cflags) app.cpp
a@peace:~/Projects/tsduckTrial$ ./a.out
Major version: 3
Minor version: 39
```
If I try this code:
```
#include "tsduck.h"
#include <iostream>
using namespace std;
int main(){
    ts::AsyncReport report;
    cout<<"Major version: "<<TS_VERSION_MAJOR<<endl;
    cout<<"Minor version: "<<TS_VERSION_MINOR<<endl;
    return 0;
}
```
This is the error I get:
```
a@peace:~/Projects/tsduckTrial$ g++ $(tsconfig --cflags) app.cpp
/usr/bin/ld: /tmp/cc4tV4ta.o: in function `main':
app.cpp:(.text+0x5e): undefined reference to `ts::AsyncReport::AsyncReport(int, ts::AsyncReportArgs const&)'
/usr/bin/ld: app.cpp:(.text+0xe2): undefined reference to `ts::AsyncReport::~AsyncReport()'
/usr/bin/ld: app.cpp:(.text+0x10b): undefined reference to `ts::AsyncReport::~AsyncReport()'
collect2: error: ld returned 1 exit status
a@peace:~/Projects/tsduckTrial$ 
```
If I try to run exact command like that in the documentation I get this:
```
a@peace:~/Projects/tsduckTrial$ gcc $(tsconfig --cflags) app.cpp -o app.o
/usr/bin/ld: /tmp/ccllu41M.o: warning: relocation against `_ZSt4cout' in read-only section `.text'
/usr/bin/ld: /tmp/ccllu41M.o: in function `main':
app.cpp:(.text+0x5e): undefined reference to `ts::AsyncReport::AsyncReport(int, ts::AsyncReportArgs const&)'
/usr/bin/ld: app.cpp:(.text+0x6f): undefined reference to `std::cout'
/usr/bin/ld: app.cpp:(.text+0x77): undefined reference to `std::basic_ostream<char, std::char_traits<char> >& std::operator<< <std::char_traits<char> >(std::basic_ostream<char, std::char_traits<char> >&, char const*)'
/usr/bin/ld: app.cpp:(.text+0x84): undefined reference to `std::ostream::operator<<(int)'
/usr/bin/ld: app.cpp:(.text+0x8b): undefined reference to `std::basic_ostream<char, std::char_traits<char> >& std::endl<char, std::char_traits<char> >(std::basic_ostream<char, std::char_traits<char> >&)'
/usr/bin/ld: app.cpp:(.text+0x96): undefined reference to `std::ostream::operator<<(std::ostream& (*)(std::ostream&))'
/usr/bin/ld: app.cpp:(.text+0xa7): undefined reference to `std::cout'
/usr/bin/ld: app.cpp:(.text+0xaf): undefined reference to `std::basic_ostream<char, std::char_traits<char> >& std::operator<< <std::char_traits<char> >(std::basic_ostream<char, std::char_traits<char> >&, char const*)'
/usr/bin/ld: app.cpp:(.text+0xbc): undefined reference to `std::ostream::operator<<(int)'
/usr/bin/ld: app.cpp:(.text+0xc3): undefined reference to `std::basic_ostream<char, std::char_traits<char> >& std::endl<char, std::char_traits<char> >(std::basic_ostream<char, std::char_traits<char> >&)'
/usr/bin/ld: app.cpp:(.text+0xce): undefined reference to `std::ostream::operator<<(std::ostream& (*)(std::ostream&))'
/usr/bin/ld: app.cpp:(.text+0xe2): undefined reference to `ts::AsyncReport::~AsyncReport()'
/usr/bin/ld: app.cpp:(.text+0x10b): undefined reference to `ts::AsyncReport::~AsyncReport()'
/usr/bin/ld: /tmp/ccllu41M.o: in function `__static_initialization_and_destruction_0(int, int)':
app.cpp:(.text+0x151): undefined reference to `std::ios_base::Init::Init()'
/usr/bin/ld: app.cpp:(.text+0x16c): undefined reference to `std::ios_base::Init::~Init()'
/usr/bin/ld: /tmp/ccllu41M.o:(.data.rel.local.DW.ref.__gxx_personality_v0[DW.ref.__gxx_personality_v0]+0x0): undefined reference to `__gxx_personality_v0'
/usr/bin/ld: warning: creating DT_TEXTREL in a PIE
collect2: error: ld returned 1 exit status
a@peace:~/Projects/tsduckTrial$ 
```

lelegard, 2024-10-13:
On your build command, you compile and link at the same time but you only provided the compile flags, not the link flags (which specifies the TSDuck library). 

In the first example, you only use compile-time symbols and you don't reference the library. And it works. 

In the second example, you reference a class of the library. Therefore you get an undefined symbol.

the-atharva, 2024-10-13:
Thanks for prompt reply. I was able to execute the program after some changes.
I may be wrong but I think the official documentation for tsconfig [here](https://tsduck.io/download/docs/tsduck.html#tsconfig-ref) for building application us should be changed to this:
```
g++ $(tsconfig --cflags) app.cpp -c app.o
g++ app.o $(tsconfig --libs) -o app
```

lelegard, 2024-10-13:
Do you mean `g++` instead of `gcc`?
This is correct, I will fix it.

the-atharva, 2024-10-13:
Yes `g++` instead of `gcc` & `-c` instead of `-o` in 1st line for compiling & building only.

closed 2024-10-14 by the-atharva