tsduck/tsduck #1536: Failed to successfully insert SCTE-35

Avencheng, 2024-10-15:
Greeting~
I'm trying to insert SCTE-35 into a udp stream and out put it as a ts file. 
Here is my tsp command:
```
tsp -I ip 235.35.3.5:3535 \
        -P pmt --service 1 --add-programinfo-id 0x43554549 --add-pid 600/0x86 \
        -P spliceinject --service 1 --files 'xml/splice-1.xml' --udp 3535 \
        -O file scte35_test.ts
```
The xml file is already present at the time the processing starts. The content of `splice-1.xml` as following:
```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
    <splice_information_table>
      <splice_insert splice_event_id="100" unique_program_id="1"
                     out_of_network="true" pts_time="1061686"/>
    </splice_information_table>
</tsduck>
```
I used the `tsanalyze scte35_test.ts` to exame the `scte35_test.ts`. The pid 600/0x86 seems added, but there is no splice info?
```
|  PID  Usage                                     Access          Bitrate  |
|   Total  Digital television service ................... C    7,679,321 b/s  |
|  0x0100  AVC video (1920x1080, high profile, level 4.0  C    7,214,877 b/s  |
|  0x0101  MPEG-1 Audio (chi, Audio layer II, 384 kb/s,   C      447,401 b/s  |
|  0x0258  SCTE 35 Splice Info .......................... C            0 b/s  |
|  0x1000  PMT .......................................... C       17,042 b/s  |
|          (C=Clear, S=Scrambled, +=Shared)                                   |
```
I seems missing something very simple, please correct me.

lelegard, 2024-10-15:
Hi,

Do you have "free space" (a.k.a. stuffing, a.k.a. null packets) in your input stream? A transport stream is like a file system; when it's full you can't create new files. Inserting packets means replacing null packets with new packets; just like creating a file means replacing free blocks in the file system with data for the new file.

If your TS is CBR (broadcast), it should contain some stuffing. You can insert packets, but not more that the input free space.

If your TS is VBR (streaming), you probably have no input stuffing. You need to create some arbitrary amount of stuffing at input, do the insertion, remove the extra stuffing before the output plugin.

Please read the [reference documentation for tsp](https://tsduck.io/download/docs/tsduck.html#tsp-reference), especially the section named "Modifying, inserting and deleting packets".

lelegard, 2024-10-15:
Also in the case of SCTE-35 insertion, you will see the cue messages only a few seconds before PTS 1061686 (as seen in your file). Otherwise, nothing is inserted, even with free space. See the documentation of the plugin `spliceinject` for the insertion policy.

Avencheng, 2024-10-18:
Hi Thank you for your detailed explanation!

I was using a ts file to create streams via FFmpeg for testing phase:
`ffmpeg -re -i input.ts -vcodec copy -f mpegts 'udp://235.35.3.5:3535?pkt_size=1316'`
Now i have changed my `tsp` command as bellow:
```
tsp --add-input-stuffing 1/10 \
        -I ip 235.35.3.5:3535 \
        -P pmt --service 1 --add-programinfo-id 0x43554549 --add-pid 600/0x86 \
        -P spliceinject --service 1 --files 'xml/splice-1.xml' \
	-P filter --negate --pid 0x1FFF \
        -O file tsp_out.ts
```
The splice info was successfully inserted.
[image]

By the way, i will use the signal from IRD(broadcast) in real case. If it already contain some stuffing, but not more that the input free space as you said. Does it mean that will be better if I add stuffing in that case?

Thanks for your time to reply my question.

lelegard, 2024-10-18:
>. By the way, i will use the signal from IRD(broadcast) in real case. If it already contain some stuffing, but not more that the input free space as you said. Does it mean that will be better if I add stuffing in that case?

It depends on what you want to do with the _output_ of your command. CBR or VBR?

If you want to reinject the output as a broadcast stream with the same modulation parameters, you need to exactly preserve the input bitrate. This means 1) don't add input stuffings, 2) don't remove stuffing before output. If there is not enough input stuffing (which is doubtful), then you can't add anything in that stream, no way.

Just check if the IRD actually transmits the complete TS, including stuffing. I have seen some tuners that removed stuffing at low level.

If you reroute the output to streaming for instance, then you don't care about staying CBR and you may remove output stuffing (and add input stuffing if there is not enough in input).

Avencheng, 2024-10-21:
Thanks again for your detailed explanation!
I'll close this topic, cheers~

closed 2025-03-23 by github-actions[bot]