tsduck/tsduck #1538: [BUG] TSDuck doesn't work under Windows Server 2016

baerlein, 2024-10-28:
#### Bug description:
I have tried both the latest setup binary and the portable version and when running any of the tsduck tools (e.g. tsp.exe), I'm getting the following error in a popup window:

`The procedure entry point SetThreadDescription could not be located in the dynamic link library C:\Program Files\TSDuck\bin\tsduck.dll.
`

I have also tried running tsduck under Windows Server 2019 and 2022 and it works.

#### How to reproduce:
Install TSDuck-Win64-3.38-3822.exe in Windows Server 2016 Standard and run tsp.exe.

#### Expected behavior:
tsp.exe should work.

#### Errors and logs:
The procedure entry point SetThreadDescription could not be located in the dynamic link library C:\Program Files\TSDuck\bin\tsduck.dll.

#### Environment:
 - OS: Windows 
 - OS version: Windows Server 2016 Standard (x64)
 - TSDuck full version: 3.38-3822
 - Installation type: official binary TSDuck-Win64-3.38-3822.exe

#### Additional information:

baerlein, 2024-10-28:
Adding information that might be useful: 
I just verified that version 3.30-2710 is the last release version that works in Windows Server 2016.

lelegard, 2024-10-28:
To be precise, TSDuck now needs to call the system function `SetThreadDescription` and this function did not exist in Windows Server 2016.

baerlein, 2024-10-28:
Makes sense.
Thanks.

closed 2025-03-30 by github-actions[bot]