tsduck/tsduck #1540: Build without docs, Make variable e.g. NODOCS

GerryHickman, 2024-11-03:
Previous build 3.36-3528, no dependency on asciidoctor, no problem

New build 3.38-3822, wants asciidoctor (and might want ruby)

But I don't want docs right now, and don't want tsduck to have ruby dependency.

I was hoping there'd be a Make variable, something like NODOCS. There's a similar Make variable NODEKTEC that I use, but I can't find a way to skip docs generation? In the end I had to comment our some lines in the main Makefile.

lelegard, 2024-11-03:
Hi @GerryHickman,

Which `make` command did you run? The command `make` alone (or `make -j10` or more for the sake of speed) builds the binaries only, not the documentation.

The command `make docs` is used to build the documentation, HTML and PDF.

The command `make install` or `make installer` (which builds the `.rpm` or `.deb`) builds the HTML docs because these files are part of the installer and need to be built.

Finally, the script `scripts/install-prerequisites.sh` installs all required build tools, including Asciidoctor and Ruby.

Note that the _build_ of TSDuck depends on many things, compiler, Java, Python, and Ruby (for Asciidoctor) but the installed TSDuck does not depend on them.

GerryHickman, 2024-11-03:
The original build command was
`make -j10 NOTEST=1 NODEKTEC=1 NOHIDES=1 NOVATEK=1 NOPCSC=1 NOSRT=1 NORIST=1 NOGITHUB=1`
and actually this was fine
then I ran
`make NOGITHUB=1 install SYSPREFIX=/data/build/tsduck-3.38-3822/inst`
and this is where it broke
I didn't run the prerequisites script, as I was worried it might try to install something (or many things)
I'll just have to get used to the new way...

lelegard, 2024-11-03:
I will add `NODOC=1` to `make install` to install without documentation.

lelegard, 2024-11-03:
Note that, for consistency, you should use exactly the same list of variables to all make commands, for instance the build and the installation. For a direct installation, there are just a few differences, for instance if you use `pkgconfig` later.

GerryHickman, 2024-11-03:
Thanks for the update.
I'll make sure I use the full list next time I run make install.
With the new NODOC variable, I assume I can add it to both make commands?
You probably guessed I'm doing the whole thing without needing root, and therefore the prerequisites script probably would not work.

lelegard, 2024-11-03:
The option NODOC has been added.

> With the new NODOC variable, I assume I can add it to both make commands?

Yes, generally speaking, use the exact same set of make variables to build and install. This enforces the consistency of the installation with what was built.

Since you do not need `sudo` for `make install`, you don't even need to run the two commands separately. Because installing requires building first, the `make install` command will rebuild first if necessary and then install.

So, you can just use one single command:
~~~
make -j10 install NODOC=1 NOTEST=1 ...
~~~

I recommend `-j10` to reduce the build time. It basically depends on your number of CPU's. On my 8-core iMax, `make -j10` takes 2 minutes. Without it, no way...

GerryHickman, 2024-11-04:
Thanks for the additional details.
I'm hoping to test after work.
I'm looking at yesterdays changes and I see
`Various fixes on make-config`
has an 'x' next to it in github, not sure what that means or if it affects me using make...

GerryHickman, 2024-11-04:
Tested NODOC with repo from 04 Nov 2024.
Worked first time.
I also found NOJAVA and NOPYTHON options.

closed 2024-11-04 by GerryHickman