tsduck/tsduck #1541: [Q] How to diagnose broken MGT / tVCT

faroutliving, 2024-11-04:
With a strong interest in software defined radios (SDR) and working with ffmpeg/libav & streaming content on my day job, I thought I would build a ATSC encoder for SDR. First step was creating a single service and that worked fine. TV could discover the channel and it would play fine. So I thought it would be a simple matter to multiplex two services into a single TS. 

Sadly, TV will not discover the channels 100%. w_scan seems to pick them out fine. The TV recognizes the channels, but gets the major/minor channel wrong and shortname is omitted. Ie, no tVCT. So after pouring over the output by hand again and again, I started my search for something to help analyze what I was generating and discovered TSDuck. TSDuck seems to not recognize the VCT either! My first guess is that the MGT is not properly pointing to the tVCT, but w_scan just looks for the tVCT in the stream and doesn't parse the MGT?

What I can't seem to figure out (besides the original error in my stream), is how TSDuck can help me find the error. Here is output of the first 3 packets with PAT, MGT, and tVCT using tsdump, and tsp -P tables will only dump the PAT and MGT. It will dump the tVCT for ota captures so I know it can do so.

I'm sure the error is a simple one, and I am betting TSDuck can help, but I just can't see the forest for the trees.

Thanks for any suggestions,
Deron

##################################################################
# tsdump -a test.ts

* Packet 0
  ---- TS Header ----
  PID: 0x0000 (0), header size: 4, sync: 0x47
  Error: 0, unit start: 1, priority: 0
  Scrambling: 0, continuity counter: 1
  Adaptation field: no (0 bytes), payload: yes (184 bytes)
  ---- Full TS Packet Content ----
  47 40 00 11 00 00 B0 11 21 C2 C1 00 00 00 03 E0 30 00  G@......!.......0.
  04 E0 40 2A BD E5 D8 FF FF FF FF FF FF FF FF FF FF FF  ..@*..............
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF                                ........

* Packet 1
  ---- TS Header ----
  PID: 0x1FFB (8187), header size: 4, sync: 0x47
  Error: 0, unit start: 1, priority: 0
  Scrambling: 0, continuity counter: 1
  Adaptation field: no (0 bytes), payload: yes (184 bytes)
  ---- Full TS Packet Content ----
  47 5F FB 11 00 C7 F0 19 00 00 C1 00 00 00 00 01 00 00  G_................
  FF FB E0 00 00 00 72 F0 00 F0 00 9A 57 61 57 FF FF FF  ......r.....WaW...
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF                                ........

* Packet 2
  ---- TS Header ----
  PID: 0x1FFB (8187), header size: 4, sync: 0x47
  Error: 0, unit start: 1, priority: 0
  Scrambling: 0, continuity counter: 1
  Adaptation field: no (0 bytes), payload: yes (184 bytes)
  ---- Full TS Packet Content ----
  47 5F FB 11 00 C8 F0 6F 21 C2 C1 00 00 00 02 00 54 00  G_.....o!.......T.
  45 00 53 00 54 00 00 00 00 00 00 F0 80 01 04 00 00 00  E.S.T.............
  00 21 C2 00 03 0D C2 00 03 FC 11 A1 0F E0 31 02 02 E0  .!............1...
  31 00 00 00 81 E0 34 65 6E 67 00 44 00 4F 00 4C 00 4C  1.....4eng.D.O.L.L 
  00 53 00 00 00 00 F0 80 02 04 00 00 00 00 21 C2 00 04  .S............!...
  0D C2 00 04 FC 11 A1 0F E0 41 02 02 E0 41 00 00 00 81  .........A...A....
  E0 44 65 6E 67 FC 00 09 7E 62 F5 FF FF FF FF FF FF FF  .Deng...~b........
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ..................
  FF FF FF FF FF FF FF FF                                ........
[... 94 lines cut ...]
	0x50: 04 FC 11 A1 0F E0 41 02 02 E0 41 00 00 00 81 E0 :       A   A     
	0x60: 44 65 6E 67 FC 00                               : Deng  
	========================================================================
	===================== parse_psip_descriptors ===========================
	len = 17
	0x00: A1 0F E0 31 02 02 E0 31 00 00 00 81 E0 34 65 6E :    1   1     4en
	0x10: 67                                              : g
	========================================================================
  VIDEO     : PID 0x0031
	AUDIO	: PID 0x0034 lang: eng
service is running. Channel number: 32:1. Name: 'TEST'
	===================== parse_psip_descriptors ===========================
	len = 17
	0x00: A1 0F E0 41 02 02 E0 41 00 00 00 81 E0 44 65 6E :    A   A     Den
	0x10: 67                                              : g
	========================================================================
  VIDEO     : PID 0x0041
	AUDIO	: PID 0x0044 lang: eng
service is running. Channel number: 32:2. Name: 'DOLLS'
read_filters:1985: filter success: pid 0x1ffb
remove_filter:1941: remove_filter: pid 8187 (0x1ffb)
stop_filter:1913: stop_filter: pid 8187 (0x1ffb)
..
TEST;(null):545000:M10:A:0:49=2:52=eng;52:0:0:3:0:0:0
DOLLS;(null):545000:M10:A:0:65=2:68=eng;68:0:0:4:0:0:0

lelegard, 2024-11-04:
Hi @faroutliving 

TSDuck is the right tool to analyze that kind of problem. It can also generate the signalization, by the way.

The command `tstables` (or the plugin `tables`) scans sections and tables on all PID's. So, if a valid TVCT was there, somewhere, even on an incorrect PID, it would have found it. Maybe it is not injected at all or the packetization is incorrect.

You could also try the options `--include-next --invalid-sections --invalid-versions` with `tstables`. They report additional error situations.

You can also try the command `tsanalyze` to get a global view of the TS.

See the user's guide here: https://tsduck.io/download/docs/tsduck.html#_tstables

You can also have a look at a few real-world ATSC capture here: https://tsduck.io/streams/?name=usa-atsc

faroutliving, 2024-11-04:
That was it! And -P continuity does indeed complain about it. I was tracking continuity per packet type, not PID. I have fixed that, and so far so good in testing.

I can only presume that earlier single service streams had discontinuous CC values instead of serial duplicate values that the tv's ignored or accepted in some way. Either way, I am happy to have a fix and glad to have learned something more about TSDuck.

Might I request that TSDuck not quietly discard packets, but at least have an error mode when something is wrong. I am afraid it would have been a very very long time before I had discovered this on my own.

Thanks so much for your time for TSDuck and for this discussion!

lelegard, 2024-11-05:
> Might I request that TSDuck not quietly discard packets, but at least have an error mode when something is wrong.

TSDuck is very versatile and there is rarely one single solution. It depends on what is wrong, how much does it cost to detect and what are the performance requirements. 

By default, we detect the minimum functional conditions. When it comes to demuxing sections for instance, we track continuity counters to make sure we use a continuous stream. We don't check if contiguous packets with identical CC have same content (true duplicate) or not (invalid stream) because this is too costly: need to keep a full copy of the last packet of all PIDs, need to compare and copy, at every packet in the stream. But if you want these error detection and error messages, just plug a `continuity` plugin in the command. 

TSDuck is a Lego set. It provides the building blocks. You build the functions and error detection you need, case by case.

faroutliving, 2024-11-07:
Just a request. Obviously some part of the code made a comparison and skipped. Would be a good time to (optionally) print an error message which would then prompt someone to dig further. Would not consume memory or require any complicated code. I agree that built in deeper exploration would require too much resources.

Thanks again for your help. I will close this now.

lelegard, 2024-11-09:
> Obviously some part of the code made a comparison and skipped. Would be a good time to (optionally) print an error message which would then prompt someone to dig further. Would not consume memory or require any complicated code. 

The continuity counters are checked when demuxing a PID. Duplicate packets are silently ignored. Missing packets create a hole, the currently deserialised data structure is dropped and resynchronisation will be done on next PUSI.

Displaying a message would not consume computing resources but it would consume the user's patience 😄 

Many captured broadcast streams have missing packets. This creates discontinuities. Unconditionally displaying a message for each continuity issue is annoying (see the ffmpeg example). Therefore, these messages have been suppressed. If you want them, use the `continuity` plugin.

closed 2025-04-13 by github-actions[bot]