tsduck/tsduck #1542: [Q] Playing ISDB-Tb (BTS 204 bytes) capture with tsp

mariossmoura, 2024-11-04:
Hello TSDuck community,
 
First of all, thank you for providing this tool; it has been very valuable for my testing in the lab.

I am currently running tests with a .ts capture from a 204-byte ISDB-T stream (Brazilian standard) but I'm encountering some issues.

I'm using the following command to stream this capture via multicast IP:

tsp -I file -i /root/MK/BTS204.ts -P pcrbitrate --min-pcr 256 -P regulate --packet-burst 128 -O ip 239.10.10.10:5001 &

Here are the main issues I’m observing:

1.	The output packet size is 188 bytes instead of the expected 204 bytes.
2.	The IIP PID (TMCC) appears to be passing through, but its characteristics are changing unexpectedly.
3.	The dummy8 packets are being filtered out.

For reference, I've included a comparison below that highlights the differences between the original capture and the TSDuck-generated stream.

Original capture:
[image]

TSDuck streaming:

 [image]
 
Any guidance on maintaining the 204-byte structure (TMCC+dummy8) and preserving all packet types and characteristics would be greatly appreciated.
 
Regards,
 
Mario Moura

lelegard, 2024-11-05:
Hi,

TSDuck works on standard ISO 13818-1 transport streams where all packets have exactly 188 bytes. The extra 16 bytes are apparently specific to ISDB-Tb. The input plugins extract 188-byte packets from the specification of their input transport. In the case of files of 204-byte packets, the extra 16 bytes are discarded. Usually, files of 204-byte packets are raw broadcast captures and the extra 16 bytes are just a Reed-Solomon outer FEC which is useless.

Internally, the large packet buffer over which all plugins work is made of 188-byte packets. It is not possible to change this without breaking virtually everything.

However, there is also a buffer of associated metadata. Each packet has its metadata which are currently made of timestamps, labels and a few flags. In theory, it could be possible to extend those metadata to include ISDB-Tb data. However, we need to know more about this specific packet trailer to determine if this is appropriate.

- What is the exact content of the trailer?
- Any link to a reference document?
- Are all trailers identical in a given stream?
- Or dependent on the position in the stream?
- Or dependent on the content of the packet, such as PID or payload?

In the processing of a stream in a chain of plugins, packets can be deleted, replaced, swapped, transformed in many ways. We need to understand the relationship between a packet and its trailer.

Please explain what are:
- IIP PID
- dummy8 packets.

Never heard of that, must be specific to ISDB-Tb.

In your commands, all packets are passed, without omission, without modification. The only difference is that the 16-byte trailers are discarded.

In your screenshot, the PID 8176 is identified as IIP with 204-byte packets and unknown with 188-byte packet. Could you confirm if this identification is stored in the 16-byte trailer?

There is a [stream repository](https://tsduck.io/streams/) where TSDuck users have provided streams from their region. There are [two ISDB-Tb streams from Brazil](https://tsduck.io/streams/?name=brazil-isdb-tb) but the analysis shows no unknow PID which could be the IIP.

lelegard, 2024-11-05:
> I downloaded one of the ISDB-Tb streams from Brazil, but it is 188 bytes, not 204.

They were provided by a Brazilian user and are published without modification. They don't contain any PID 8176 either. So, no IIP. According to their TOT, they were captured in December 2015.

> Do you know how I can upload my capture?

Upload it on some public file transfer service and post the URL here. I will publish it on the site if this is a real-world stream (not a test stream in a lab).

mariossmoura, 2024-11-05:
Hello,

Here is the link to downlod the file: 

https://1drv.ms/v/c/9851636b7b4c3f8b/EQ-ifproHgpNuulDWTVPao8BcLEXpuSk_w0AFL2ASdxdLw?e=Z1citu

I'll try to find any documentation about this IIP PID.

lelegard, 2024-11-05:
Thanks for the file. The 16-byte trailers are, for the first packets:
~~~
A9 0F E7 CC FF FF FF FF 69 F9 92 95 D6 A0 16 CD
A9 0F E7 CD FF FF FF FF B2 80 D9 AF 60 AC BD 0E
A9 0F E7 CE FF FF FF FF C2 0B 04 E1 A7 B8 5D 56
A9 0F E7 CF FF FF FF FF 19 72 4F DB 11 B4 F6 95
A9 0F E7 D0 FF FF FF FF 85 2C 05 37 42 30 D3 1C
A9 0F E7 D1 FF FF FF FF 5E 55 4E 0D F4 3C 78 DF
A9 0F E7 D2 FF FF FF FF 2E DE 93 43 33 28 98 87
A9 0F E7 D3 FF FF FF FF F5 A7 D8 79 85 24 33 44
A9 2F E7 D4 FF FF FF FF DC E4 6B BF 53 27 A7 17
A9 2F E7 D5 FF FF FF FF E5 7D 6B 24 5D E6 1F 71
A9 2F E7 D6 FF FF FF FF 24 6A 3C 20 3F 21 DA 7A
A9 2F E7 D7 FF FF FF FF 51 95 BB 80 9B E0 07 6F
A9 2F E7 D8 FF FF FF FF 63 59 1F 73 72 B9 07 4B
A9 2F E7 D9 FF FF FF FF ED 1D 9F 2C 79 29 47 0C
A9 2F E7 DA FF FF FF FF F3 0D 09 E2 8D 9F 0F FB
A9 2F E7 DB FF FF FF FF 5D 9B 2E 82 96 E8 8F C9
A9 2F E7 DC FF FF FF FF 61 72 AB C6 AB 20 5B BB
A9 2F E7 DD FF FF FF FF DD C8 06 9A A8 A1 94 08
A9 2F E7 DE FF FF FF FF 08 60 1A F6 68 73 AE DC
A9 2F E7 DF FF FF FF FF 58 C1 5C 80 34 8B 0D FE
A9 2F E7 E0 FF FF FF FF A6 04 83 7F 24 92 EC 32
A9 0F E7 E1 FF FF FF FF 0D 0D 1F 99 18 61 DE 36
A9 0F E7 E2 FF FF FF FF 7D 86 C2 D7 DF 75 3E 6E
A9 0F E7 E3 FF FF FF FF A6 FF 89 ED 69 79 95 AD
A9 0F E7 E4 FF FF FF FF 9D 8D 65 4B 4C 5D E3 DE
A9 0F E7 E5 FF FF FF FF 46 F4 2E 71 FA 51 48 1D
A9 0F E7 E6 FF FF FF FF 36 7F F3 3F 3D 45 A8 45
A9 0F E7 E7 FF FF FF FF ED 06 B8 05 8B 49 03 86
A9 0F E7 E8 FF FF FF FF 40 9B 36 6E 77 0D 44 A3
A9 1F E7 E9 FF FF FF FF F9 33 58 74 9E ED 47 B4
A9 2F E7 EA FF FF FF FF 65 C1 37 C7 2A 74 8B BA
A9 2F E7 EB FF FF FF FF A5 79 76 34 2F B6 04 CA
A9 2F E7 EC FF FF FF FF 32 D4 76 7B 90 68 3A D4
A9 2F E7 ED FF FF FF FF 3A 3F 02 F6 22 29 CB 62
A9 2F E7 EE FF FF FF FF F3 FC C8 4A 63 D6 B5 6D
A9 2F E7 EF FF FF FF FF EC 39 4D 8E DC A1 78 0B
A9 2F E7 F0 FF FF FF FF 13 C9 26 A3 FE BF 80 FF
A9 2F E7 F1 FF FF FF FF 5C 83 2B 10 F7 8D 6E 64
~~~

We see patterns:
- 4 bytes: constant, except last byte incrementing, 2nd byte with 3 possible values.
- 4 bytes: FF FF FF FF, probably unused
- 8 bytes: apparent random, possibly a shortened Reed Solomon

Looking forward to get the spec.

lelegard, 2024-11-06:
The paper here, from the Brazilian Society of Television Engineering, contains some additional information: https://www.set.org.br/wp-content/uploads/2016/05/04_IEEE-format-IP-Transmission-over-ISDB-TB.pdf

"This output signal is called the Broadcast Transport Stream (BTS)" 

- First time I see the acronym "BTS". Good to know...

"The final sixteen bytes of the BTS, i.e., the dummy byte, has a counter and a packet layer identifier [13], and can optionally have a Reed Solomon block code (RS) shortened to (204,196,4) with a correction capacity of up to 4 bytes in a TS packet."

- The "dummy byte" (singular??) is the 16-byte trailer.
- The final 8 bytes are indeed a shortened Reed Solomon.

"Identified in the final 16 bytes of the BTS, the corresponding hierarchical layer of each packet coming from the encoders are configured in ISDB-TB multiplexer so that the modulator can transmit that packet in the layer correctly."

- There are some modulation information in the first 4 bytes of the 16-byte trailer.

lelegard, 2024-11-06:
More information on the "dummy byte" structure in ARIB STD-B31, section 5.5:
https://www.arib.or.jp/english/html/overview/doc/6-STD-B31v2_2-E1.pdf

Section 5.5.2 gives the format of the 8 bytes in the trailer (excluding the final RS). The "unused" 4 bytes can also contain "AC data".
- TMCC means Transmission and Multiplexing Configuration Control
- AC means Auxiliary Channel

Section 5.5.4 gives the format of TS packets in the IIP PID. 
- IIP means "ISDB-T Information Packet".

mariossmoura, 2024-11-07:
Great!

Thanks for sharing.

Do you think something can be implemented on TS Duck to handle BTS 204 bytes?

lelegard, 2024-11-07:
I have added support for this. Now, it is possible to integrate the 16-byte trailer, when present, in the metadata of each packet. This means that the trailer is preserved from input to output.

The core mechanism is in place and can be used by all plugins. It is the responsibility of the input and output plugin to read/write them or not.

Currently, as a start and a test, this is implemented in the `file` input and output plugin. On the output side, the default remains `--format ts` (188-byte packets). Use `--format rs204` to write the trailer. The option already existed but the generated trailer was a dummy one, a placeholder. Now, if a trailer has been read on input and propagated all along the chain, it is saved. Thus, the command `tsp -I file ... -O file --format rs204 ...` now preserves the input 16-byte trailers, when they exist. If there is no input trailer, 16 x 0xFF are generated.

I will add support in other input and output plugins, starting with `ip`.

The interesting point is that the packet metadata are accessible to all packet processing plugins. It will now be possible to develop plugins to analyze, modify, create or delete the 16-byte trailer. No idea of what can be useful now.

Support has been added in `tsdump` to dump the trailer with option `--rs204`. With option `--isdb`, the trailer is interpreted according to ARIB STD-B31 and formatted accordingly.

Example on your file:

~~~
$ tsdump BTS204.ts --isdb --rs204
...
* Packet 8
  ---- TS Header ----
  PID: 0x0111 (273), header size: 4, sync: 0x47
  Error: 0, unit start: 0, priority: 0
  Scrambling: 0, continuity counter: 1
  Adaptation field: no (0 bytes), payload: yes (184 bytes)
  ---- Full TS Packet Content ----
  47 01 11 11 8B EB 96 42 D1 5D 53 05 23 A8 2E 87 F4 55 E5 D9 E0 DC 68 04
  7F 2B 3F 27 A5 75 08 48 D6 B4 78 72 43 3F 88 37 8E 3A 69 B1 1E 87 A7 A7
  D1 85 09 6B 27 90 44 B7 66 EB 33 D2 90 96 CB 15 04 1C 8F 77 B5 3D 5E E4
  F9 B4 FB 3C 51 65 0C 81 2D 7E BF 85 18 F7 74 34 1C E9 7E ED CD A9 6A 41
  BA 10 45 27 B1 95 38 DE 56 55 6A 8A 64 41 66 51 FE DA 1A F4 F5 50 3A D1
  C6 92 C8 86 24 EB 7B 3C D4 F8 1E B2 5C 5A 59 BE 25 EB F8 6B 50 EF 26 27
  B7 E7 EE 89 DF B3 6C 51 D2 5B B4 2D 62 CB 7D 23 CA 2B A7 59 11 A6 67 0F
  71 7D 51 D0 1B 14 DD 36 A2 01 E4 91 17 AC 10 EB EE 39 33 25
  ---- ISDB-T information ----
  TMCC identifier: 0x2 (Digital terrestrial TV)
  buffer_reset_control_flag: 0
  switch-on_control_flag_for_emergency_broadcasting: 1
  initialization_timing_head_packet_flag: 0
  frame_head_packet_flag: 0
  frame_indicator: 1 (odd)
  layer_indicator: 0x2 (TSP transmitted by hierarchical layer B)
  count_down_index: 15
  TSP_counter: 0x07D4 (2004)
  ---- Packet trailer (16 bytes) ----
  A9 2F E7 D4 FF FF FF FF DC E4 6B BF 53 27 A7 17
...
~~~

One thing surprises me. I thought that the various layers A, B, C transport distinct TS. However, since this is an ISDB specificity, I have never used or observed it.

In the case of your file, see this:
~~~
$ tsdump BTS204.ts --isdb --rs204 | grep layer_indicator: | sort -u
  layer_indicator: 0x0 (NULL-TSP which is not transmitted by any layer)
  layer_indicator: 0x1 (TSP transmitted by hierarchical layer A)
  layer_indicator: 0x2 (TSP transmitted by hierarchical layer B)
  layer_indicator: 0x8 (TSP which transmits the IIP but not transmitted by any layer)
~~~

So, it seems that the same TS contains packets from distinct layers. Any explanation ?

mariossmoura, 2024-11-07:
Hello @lelegard thank you for your help!

> So, it seems that the same TS contains packets from distinct layers. Any explanation ?

Apologies, I don't have the in-depth knowledge required to answer that question thoroughly.

lelegard, 2024-11-08:
To analyze what is inside each layer, I have added an option `--isdb_layer` to the plugin `filter`.

See the result:
~~~
$ tsdump BTS204.ts --isdb --rs204 | grep layer_indicator: | sort -u
  layer_indicator: 0x0 (NULL-TSP which is not transmitted by any layer)
  layer_indicator: 0x1 (TSP transmitted by hierarchical layer A)
  layer_indicator: 0x2 (TSP transmitted by hierarchical layer B)
  layer_indicator: 0x8 (TSP which transmits the IIP but not transmitted by any layer)
$
$ for layer in 0 8 1 2; do echo ==== Layer $layer; tsp -I file BTS204.ts -P filter --isdb-layer $layer -P count -O drop; done
==== Layer 0
* count: PID 8191 (0x1FFF):    230,362 packets
==== Layer 8
* count: PID 8176 (0x1FF0):        136 packets
==== Layer 1
* count: PID    1 (0x0001):         38 packets
* count: PID   16 (0x0010):         45 packets
* count: PID   17 (0x0011):         44 packets
* count: PID   20 (0x0014):          6 packets
* count: PID   36 (0x0024):         39 packets
* count: PID   39 (0x0027):         60 packets
* count: PID  281 (0x0119):         44 packets
* count: PID  512 (0x0200):        143 packets
* count: PID  529 (0x0211):      6,143 packets
* count: PID  530 (0x0212):        994 packets
* count: PID 8136 (0x1FC8):        484 packets
* count: PID 8191 (0x1FFF):        658 packets
==== Layer 2
* count: PID    0 (0x0000):        495 packets
* count: PID   18 (0x0012):        987 packets
* count: PID  256 (0x0100):      1,179 packets
* count: PID  257 (0x0101):        495 packets
* count: PID  273 (0x0111):    289,199 packets
* count: PID  274 (0x0112):      2,116 packets
* count: PID  275 (0x0113):      2,141 packets
* count: PID  276 (0x0114):      2,142 packets
* count: PID  277 (0x0115):      5,267 packets
* count: PID  278 (0x0116):         42 packets
* count: PID  500 (0x01F4):         60 packets
* count: PID  900 (0x0384):     11,846 packets
* count: PID 8191 (0x1FFF):     36,293 packets
$ 
~~~

Summary:
- In layer 0 (NULL-TSP which is not transmitted by any layer), there are only null packets. However, we can also find null packets in layers 1 and 2.
- In layer 8 (TSP which transmits the IIP but not transmitted by any layer), there is one single PID, the IIP. This is the only layer carrying IIP packets. No IIP packet is found in other layers.
- Layers 1 and 2 (A and B) contains a mixture of PID's.
  - Neither layer A or B is a consistent TS alone, they contain only a subset of the TS.
  - Each PID is completely inside only one of the two layers. Except null packets which can be found in both, all packets from a given PID are in one single layer.
  - Some global PID are in layer A (CAT, NIT, SDT, TOT, BIT, EIT), some in layer B (PAT, EIT).
  - Layer A contains all PID's for service "1-SEG".
  - Layer B contains all PID's for service "HD".

So, there is some kind of "organization" but the details are intriguing.
- The fact that most stuffing and all IIP are on their own layer makes sense.
- The fact that each service is on its own layer also (although an MPTS can contain much more services than available layers).
- However, the spread of the global PID's on distinct layers, in an apparently arbitrary manner, is intriguing.

Is anyone here familiar with the rules of operation of ISDB-T or ISDB-Tb?

lelegard, 2024-11-11:
Hi @mariossmoura 

I have completed support for 204-byte packets all along the chain, for plugins where this is relevant.

More details in the user's guide here: https://tsduck.io/download/docs/tsduck.html#support204

As a general rule, streams with 204-byte packets are automatically detected on input, the trailer is loaded and propagated in the packets' metadata. On output, the default is to generate 188-byte packets, a standard TS packet, but options exist to output 204-byte packets.

For contiguous streams (file and pipes), the option `--format` can be used with the 4 supported formats. For datagram-oriented streams (ip, pcap), the only two alternatives are 188 (default) and 204 (option `--rs204`) bytes.

One open point remains, the plugin `dvb` (misnamed since it supports all types of tuners, DVB, ATSC, ISDB). With ISDB-T tuners, I don't know if it is possible to extract 204-byte packets, including the modulation information. Do you have any USB ISDB-T tuner that you can use with TSDuck to capture a stream from the antenna?

mariossmoura, 2024-11-11:
Hello @lelegard 

First of all, thank you very much for your effort to make it happen. The possibility of playing BTS204 bytes content in my lab will be very helpful for troubleshooting. 

Unfortunately, I don't have any ISDB-T tuner to do this capture, sorry.

lelegard, 2024-11-19:
FYI, the new plugin `isdbinfo` (first step in the v3.40 cycle, after releasing v3.39 last week-end) analyzes ISDB modulation information, either from the 16-byte trailer in 204-byte packets or from the IIP PID.

I tested it with the file you provided. This is the only one I have with ISDB-specific 204-byte packets.

riquecouto, 2025-01-14:
> To analyze what is inside each layer, I have added an option `--isdb_layer` to the plugin `filter`.
>
> See the result:
>
> ```
> $ tsdump BTS204.ts --isdb --rs204 | grep layer_indicator: | sort -u [...]

ISDB-T and ISDB-TB adopt the Band Segmented Transmission Orthogonal Frequency Division Multiplex (BST-OFDM) modulation principle, in which the 6 MHz channel is split into 13 segments of 428.58 kHz, grouped to form three distinct hierarchical layers. Each layer can transmit a program with different levels of robustness, depending on the service needs. This format allows a service subject to a higher level of interference to be configured with more robust modulation (the 1-SEG service, i.e _one segment_, meant for mobile reception).

This might make things clearer:
[https://www.teleco.com.br/tutoriais/tutorialtvdentr2/pagina_2.asp](https://www-teleco-com-br.translate.goog/tutoriais/tutorialtvdentr2/pagina_2.asp?_x_tr_sl=pt&_x_tr_tl=en&_x_tr_hl=pt-BR&_x_tr_pto=wapp)

riquecouto, 2025-01-14:
> One open point remains, the plugin `dvb` (misnamed since it supports all types of tuners, DVB, ATSC, ISDB). With ISDB-T tuners, I don't know if it is possible to extract 204-byte packets, including [...]

Is this what you need @lelegard?  Feel free to add to the repository.
[https://drive.google.com/file/d/1kLWrGaBou6KlIAteEgZqoio4QJmU5EKC](https://drive.google.com/file/d/1kLWrGaBou6KlIAteEgZqoio4QJmU5EKC)

Source: TV Globo Rio de Janeiro
`tsp --verbose  -I dvb --brazil --uhf-channel 29  -O file --max-duration 45 --format RS204 GloboRJ.ts`

`tslsdvb
0: /dev/dvb/adapter0 ("Siano Mobile Digital MDTV Receiver", ISDB-T)`
`tsversion
3.39-3956`

Let me know if I can be of any further help!

lelegard, 2025-01-15:
@riquecouto, thanks for the TV Globo stream. However, currently, there is no way to perform a capture on a Linux tuner with 204-byte packets. Last time I checked (it was quite a while ago), the API could only return 188-byte packets. The result is that, in your file, all 16-byte trailers are FF stuffing. All ISDB-Tb information are gone.

The only real [204-byte ISDB-Tb stream](https://tsduck.io/streams/?name=brazil-bts204) I have was provided by @mariossmoura and it was not captured from a physical tuner in a Linux system.