tsduck/tsduck #1545: [Q] 5.2.6. Extracting selected packets (and packet index)

GerryHickman, 2024-11-08:
Usage Examples
5.2.6. Extracting selected packets
it shows how to dump specific packets

I'm trying to find the packet index numbers for the first few intra frames.

`bin/tsp -I file /mmedia/rec/dvb/myfile.ts -P until --packets 4096 -P filter --pid 0x18A1 -P filter --intra-frame -P dump -O drop`

this "works" and finds five packets, BUT the packet numbers shown in 'dump' look like '0,1,2,3,4', instead of the original packet index numbers?

lelegard, 2024-11-08:
Hi @GerryHickman 

The plugin `filter`, well..., filters packets. This means that all packets which satisfy one of the specified conditions are passed and all others are dropped.

This means that:
- `-P filter --intra-frame` drops all packets which don't contain the start of an I-frame.
- `-P dump` only receives the remaining packets which are therefore numbered 0, 1, 2, etc.

I don't see an obvious way to display the packet index of selected packets in the original complete TS.

What is the most important information for you? Having the packet index of all I-frame start? Or having a packet dump with a precise packet number?

lelegard, 2024-11-08:
> How can I print the packet indexes of the first five intra frames to stdout?

Well, since the request is legitimate, I have added a few things to solve it in different ways.

First, I have added an option `--intra-frame` to plugin `history`. It can be interesting to locate intra-frames in the history.

In your case:
~~~
$ tsp -I file test.ts \
      -P history --intra-frame \
      -O drop \
      2>&1 | grep intra-frame | head -5
* history: 12758: PID 0x03F2 (1010), new intra-frame, AVC H.264, service 0x226A (8810)
* history: 17691: PID 0x00D2 (210), new intra-frame, AVC H.264, service 0x2262 (8802)
* history: 49878: PID 0x03F2 (1010), new intra-frame, AVC H.264, service 0x226A (8810)
* history: 58258: PID 0x00D2 (210), new intra-frame, AVC H.264, service 0x2262 (8802)
* history: 77958: PID 0x03F2 (1010), new intra-frame, AVC H.264, service 0x226A (8810)
~~~

Then, I have added a new plugin named `trace` which traces selected packets using a custom message with `%` fields, just like `printf`. Currently, there aren't many options but it can be extended later. Amongst the `%` fields, `%i` is the index of the packet in the stream, as seen by the plugin, just like the packet index in the plugin `dump`. The field `%a` is the absolute index of the packet in the input transport stream, independently of which packet were previously filtered-out, removed, not transmitted to the plugin, or when the plugin was suspended. This value is available internally to plugins but rarely used.

So, here are two equal solutions:
~~~
$ tsp -I file test.ts \
      -P filter --intra-frame --set-label 1 \
      -P until --only-label 1 --packet 5 \
      -P trace --label 1 --format '%i' \
      -O drop
* trace: 12758
* trace: 17691
* trace: 49878
* trace: 58258
* trace: 77958
~~~

~~~
$ tsp -I file test.ts \
      -P filter --intra-frame --set-label 1 \
      -P until --only-label 1 --packet 5 \
      -P trace --only-label 1 --format '%a' \
      -O drop
* trace: 12758
* trace: 17691
* trace: 49878
* trace: 58258
* trace: 77958
~~~

I let your RTMF for the details.

GerryHickman, 2024-11-08:
Testing the new 'history' plugin. Working as advertised with the new --intra-frame option.

Would it make sense to have the first/last intra frame included in the default 'history' output?

Moving on to testing the new 'trace' plugin...

GerryHickman, 2024-11-08:
Testing the new 'trace' plugin. Works really well.

Both styles work, and show the same results (in my case).

I also tried adding an additional filter to restrict to a single pid, and got the same results.

```
bin/tsp -I file /mmedia/rec/dvb/myfile.ts -P filter --pid 0x18A1 --only-label 2 -P filter --intra-frame --set-label 2 -P until --only-label 2 --packet 5 -P trace --label 2 --format '%i' -O drop
* trace: 1189
* trace: 1889
* trace: 2627
* trace: 3179
* trace: 4028

```

This also agrees with my earlier test where I was seeing five packets in the first 4096.

lelegard, 2024-11-08:
Your first plugin `-P filter --pid 0x18A1 --only-label 2` is ineffective. It receives only packets which have label 2. Since this the first plugin, no packet has label 2, the packet receives nothing, it does nothing, all packets fly around it.

GerryHickman, 2024-11-08:
Thanks for pointing out my incorrect use the trying to filter by pid. I did try it without --only-label, but then I ended up with wrong packet numbers. Anyway your example works.

lelegard, 2024-11-08:
No need to "try", just RTFM as usual 😄 

The option `--only-label` is generic and accepted by all packet processing plugins (not input or output plugins):
~~~
  --only-label label1[-label2]
      Invoke this plugin only for packets with any of the specified labels.
      Other packets are transparently passed to the next plugin, without going
      through this one. Several --only-label options may be specified. See also
      option --except-label. This is a generic option which is defined in all
      packet processing plugins.
~~~

The option `--set-label` is specific to the plugin `filter`:
~~~
  --set-label label1[-label2]
      Set the specified labels on the selected packets. Do not drop unselected
      packets, simply mark selected ones. Several --set-label options may be
      specified.
~~~
There are also similar options `--set-permanent-label`, `--reset-label`, `--reset-permanent-label` in plugin `filter`.

Finally, the option `--label` is specific to the plugin `trace`:
~~~
  -l label1[-label2]
  --label label1[-label2]
      Trace packets with any of these label values. Labels should have
      typically been set by a previous plugin in the chain. Several --label
      options may be specified.

      Note that the option --label is different from the generic option
      --only-label. The generic option --only-label acts at tsp level and
      controls which packets are passed to the plugin. All other packets are
      directly passed to the next plugin without going through this plugin. The
      option --label, on the other hand, is specific to the trace plugin and
      selects packets with specific labels among the packets which are passed
      to this plugin.
~~~

Just keep this in mind when building a chain of plugins.

GerryHickman, 2024-11-09:
Thanks for the additional details about labels. I did actually read the docs before trying to use labels, but obviously didn't understand it well enough.

Anyway, the new features are working well.

Let me know about the idea of showing first / last intra frame in the default 'history' output, otherwise this can be closed.

lelegard, 2024-11-09:
> Let me know about the idea of showing first / last intra frame in the default 'history' output, otherwise this can be closed.

Do you mean without showing them all as with `--intra-frame`, using another specific option ?

The idea of "first" and "last" are useful only if your don't show them all. Having the last packet in a PID is useful because we don't show all packets. With `--intra-frame`, the last intra-frame is just the last displayed one.

GerryHickman, 2024-11-09:
> Do you mean without showing them all

Yes, and no new option is needed.

What I mean is to add the first and last intra frame to the default output of the 'history' plugin.

It would look a bit like this

```
* history: 0: PID 0x18A1 (6305) first packet, clear
* history: 3: PID 0x01A8 (424) first packet, clear
* history: 5: PID 0x18A2 (6306) first packet, clear
* history: 18: PID 0x18A1 (6305), PES stream_id is 0xE0 (Video 0)
* history: 50: PID 0x0000 (0) first packet, clear
* history: 51: PAT v5, TS 0x3006 (12294)
* history: 64: PID 0x18A2 (6306), PES stream_id is 0xC0 (Audio 0)
* history: 150: PMT v23, service 0x3910 (14608)
* history: 1189: PID 0x18A1 (6305), first intra-frame, MPEG-2 Video, service 0x3910 (14608)
* history: 4028: PID 0x18A1 (6305), last intra-frame, MPEG-2 Video, service 0x3910 (14608)
* history: 4083: PID 0x0000 (0) last packet, clear
* history: 4004: PID 0x01A8 (424) last packet, clear
* history: 4095: PID 0x18A1 (6305) last packet, clear
* history: 4089: PID 0x18A2 (6306) last packet, clear

```

lelegard, 2024-11-09:
OK, why not. I see an extra CPU load because we need to analyse the start of all PES packets according to all known video codecs to look for an I-frame. With `--intra-frame`, you have that extra CPU load but you asked for it. Now, you will get it by default. Let's assume nobody will complain.

lelegard, 2024-11-10:
Reporting first and last I-frames in each PID is now implemented by default.

GerryHickman, 2024-11-11:
Testing default output of the new 'history' plugin.

Looks good, but the packet numbers at the end are are not in order?

```
bin/tsp -I file /mmedia/rec/dvb/myfile.ts -P history -O drop
* history: 0: PID 0x18A1 (6305) first packet, clear
* history: 3: PID 0x01A8 (424) first packet, clear
* history: 5: PID 0x18A2 (6306) first packet, clear
* history: 18: PID 0x18A1 (6305), PES stream_id is 0xE0 (Video 0)
* history: 50: PID 0x0000 (0) first packet, clear
* history: 51: PAT v5, TS 0x3006 (12294)
* history: 64: PID 0x18A2 (6306), PES stream_id is 0xC0 (Audio 0)
* history: 150: PMT v23, service 0x3910 (14608)
* history: 1189: PID 0x18A1 (6305), first intra-frame, MPEG-2 Video, service 0x3910 (14608)
* history: 3771730: PID 0x0000 (0) last packet, clear
* history: 3771605: PID 0x01A8 (424) last packet, clear
* history: 3771374: PID 0x18A1 (6305), last intra-frame, MPEG-2 Video, service 0x3910 (14608)
* history: 3771770: PID 0x18A1 (6305) last packet, clear
* history: 3771769: PID 0x18A2 (6306) last packet, clear
```

lelegard, 2024-11-11:
> the packet numbers at the end are are not in order

This is normal, even before the latest modifications, inevitable, and documented (including how to sort them back):

~~~
$ tsp -P history --help
...
  -o filename
  --output-file filename
      Specify the output file for reporting history lines. By default, report
      history lines on standard error using the tsp logging mechanism.

      Without option --output-file, output is formated as:
      * history: PKT#: MESSAGE

      Some messages may be out of sync. To sort messages according to their
      packet numbers, use a command like:
      tsp -P history ...  2>&1 | grep '* history:' | sort -t : -k 2 -n

      When an output file is specified using --output-file, the sort command
      becomes:
      sort -n output-file-name
...
~~~
The reason is simple. History events are displayed as they occur. We don't know which packet or intra-frame will be last in its PID, until the end of the stream. When the end of the stream occurs, several additional events have been reported after these "last" events. After the end, we loop back into internal tables and we display the "last" events in each PID.

This is the same for declaring a PID "suspended". We need to wait for some time without any packet of that PID to declare it suspended (60 seconds by default) and many events may have been reported in the meantime.

GerryHickman, 2024-11-11:
Thanks for the update about the order of packet numbers.
Updated plugin is working well.

closed 2024-11-11 by GerryHickman