tsduck/tsduck #1546: Error : spliceinject : pts time must be in range 0 to 8589934591

Avencheng, 2024-11-11:
Recently, i'm testing my project of detect frames and spliceinject, basiclly it has those two process working at the same time:
(1) detect frames and create xml file: Input IRD UDP stream, the process will detect frames and write the pts of the target frame into an xml file.
(2) spliceinject: Inputting the same IRD UDP stream as the above process, I delayed the stream for 5 seconds to allow enough time to inject the splicing information, here is my `tsp` command:
```
tsp --synchronous-log --add-input-stuffing 1/10 -I ip 229.0.0.5:9000 -P timeshift --time 5000 \
    -P pmt --service 24 --add-programinfo-id 0x43554549 --add-pid 600/0x86 \
    -P spliceinject --service 24 --files 'xml/splice-*.xml' --delete-files\
    -P filter --negate --pid 0x1FFF \
    -O ip 230.0.0.14:3539
```
I always got an error like:
```
* Error: spliceinject: Error in table <splice_information_table> at line 2
* Error: spliceinject: '8641593013' must be in range 0 to 8,589,934,591 for attribute 'pts_time' in <splice_insert
```
When this error happens I will restart the whole program (1 and 2) and then the problem goes away since the pts starts with a smaller value, but when the pts increases greater than 8,589,934,591 the same error appears again.

Is it possible to have spliceinject read pts which larger than 8,589,934,591? Thanks!

lelegard, 2024-11-11:
Hi @Avencheng 

8,589,934,591 is 0x1FFFFFFFF, the maximum value on 33 bits. PTS and DTS values are encoded on 33 bits. Therefore, it is impossible by design to have higher values of PTS. The next value after 8,589,934,591 is 0. That's the way it is designed.

Avencheng, 2024-11-12:
OK ~ 
Furthermore, if i don't restart the `tsp` command to reload the stream, does the pts will starts fram 0 after 8,589,934,591 ? Thanks!

lelegard, 2024-11-12:
> does the pts will starts fram 0 after 8,589,934,591 

Yes, all time stamps, PTS, DTS, PCR, use modular arthmetic and wrap up to zero every 26h 30mn.

Avencheng, 2024-11-13:
Hi @lelegard 
I've observed that the PTS for the IRD stream seems to reset every 26h 30mn.
But strangely the PTS doesn't reset on my side. Here's how I confirmed it:
First, I record the input stream as a ts file.
`tsp -I 229.0.0.5:9000 -O file iNews2.ts`

Then I check the changes of PTS in recorded ts file. The output of the second command shows that pts has not been reset. PTS would continue to increase after 8,589,934,591.
```
test@cuda:~/test$ ffprobe -f lavfi -i "movie=iNews2.ts,select=eq(n\,332000)" -v quiet -of csv=p=0 -show_entries packet=pts:packet=pts_time
8589749460,95441.660667

test@cuda:~/test$ ffprobe -f lavfi -i "movie=iNews2.ts,select=eq(n\,333000)" -v quiet -of csv=p=0 -show_entries packet=pts:packet=pts_time
8592752460,95475.027333
```
So what can i do to let the PTS starts from 0 after 8,589,934,591? Is there any option can fix this?

lelegard, 2024-11-13:
> So what can i do to let the PTS starts from 0 after 8,589,934,591? Is there any option can fix this?

PTS is a generic term for Presentation Time Stamp. It exists in all video and container formats. However, each format has its own definition, unit and size. In MPEG-TS streams, the PTS uses a unit of 1/90000 second (90 kHz) and is stored on 33 bits. Therefore, no PTS value, when directly extracted from a PES packet, can have a larger value then 2^33-1. It is _physically impossible_.

If you see a larger value, then this is _not_ a PTS value which was extracted from a PES packet. If this value is displayed by ffprobe, it is quite possible that this tool displays its own internal PTS value which continues to increment on 64 bits, even when the 33-bit extracted value wraps to zero. In that case, this PTS value is a synthetic ffprobe internal counter, not a MPEG-TS PTS value. This is a question to ask in an ffmpeg forum.

If you want to see the actual PTS values from the PES packets in the MPEG-TS, use `tsp -P pcrextract --pts`.

lelegard, 2024-11-13:
From ISO 13818-1 and H.222:

[image]
[image]

You can see that the PTS and DTS are stored in 3 parts, for a total of 33 bits.

An the description explicitly says that the PTS value is modulo 2^33:

[image]

So, it anyone knows a way to store a value larger than 2^33-1 on 33 bits, please proceed 😄

Avencheng, 2024-11-18:
Hi @lelegard 
Thank you so much for the informations. The problem has been solved.
I still have so much to learn.

closed 2024-11-18 by Avencheng