tsduck/tsduck #1547: [Q] Trying to test RIST using TSDuck

paragon-yamm2, 2024-11-13:
I am currently trying to test RIST telemetry using TSDuck but I can't seem to get it installed properly. I have followed [Issue 620](https://github.com/tsduck/tsduck/issues/620) to the best of my ability but some of the links there are not going through.

Currently I am using Ubuntu 20.04.6 LTS and I get to the point where I've compiled the binaries but when I go run "tsp -I rist" i get the error: "tsp: error while loading shared libraries: libsrt.so.1.5: cannot open shared object file: No such file or directory" I have followed Issue 620 to the furthest I can but would like some help on why this is the case 

I did read somewhere on that issue something about LTS, am I correct to think that TSDuck would not work on an Ubuntu LTS version? Should I look elsewhere for a tool to test RIST connections?

lelegard, 2024-11-13:
Hi @fluffy-celery 

> am I correct to think that TSDuck would not work on an Ubuntu LTS version?

Of course, TSDuck works on Ubuntu LTS !

Now it depends on what you call "TSDuck". If you mean "a prebuilt binary that you can install", well no, not for all OS versions. TSDuck is a free open source project. We don't have the infrastructure of Canonical, Red Hat and the like. Each version of TSDuck is built only on the latest version of each distro. We don't have the infrastructure to build 40 binaries for the latest 4 or 5 versions of Ubuntu, Debian, Fedora, Red Hat EL, etc.

On macOS and Windows, the binary distributions are ready to install on any "not too old" OS version. However, Linux is known for building binaries which are compatible with only one version of only one distro. With projects such as TSDuck which are not delivered with the distro, you need to rebuild most of the time. Building TSDuck is easy and [explained here](https://tsduck.io/download/docs/tsduck-dev.html#unixbuild).

Now, using RIST - a recent protocol with many recent updates - on a 4 years old distro such as Ubuntu 20.04 is looking for troubles. You won't get the required features. Or you will have to rebuild everything. The fact that this is a "LTS" can be misleading. It is "supported" by Canonical. But this is a supported "old stuff" with old features.

On the other hand, installing Ubuntu 24.04 is a matter of 5 minutes of human time, plus 10 or 15 minutes of unassisted processing. You will get the latest versions of RIST and you can directly install the latest [nightly builds](https://tsduck.io/download/prerelease/) of TSDuck. The nightly binaries are built on GitHub CI infrastructure which, by chance, uses Ubuntu 24.04 LTS.

If you choose the hard way of using an older distro, you will probably spend much more time analysing and fixing problems, finding that you need a more recent librist, rebuilding it, fixing build problems, etc.

Or you devote 5 minutes of your time to install Ubuntu 24.04, maybe in a virtual machine, or upgrade your old Ubuntu 20. Your choice...

fluffy-celery, 2024-11-13:
Got it, I will update to 24.04 and reattempt again. Thank you very much. Hopefully I will not run into more issues but will keep this line open if I to. Thank you again

lelegard, 2024-11-13:
> would running the install on 22.04 be ill advised?

You mean the TSDuck nightly builds? They won't run for sure because they are built against libraries versions of Ubuntu 24.04 which are more recent by 2 years. You would have to rebuild yourself.

But it takes time. TSDuck is pretty big, nearly 400,000 lines of C++. Using 10 parallel compilations, it takes only 2 minutes on my M3 iMac but, if you look at the [latest nightly build logs](https://github.com/tsduck/tsduck/actions/runs/11809218163), it takes 20 minutes on Linux and 1 hour on Windows, 64-bit Intel, even with parallel builds.

fluffy-celery, 2024-11-14:
Good morning @lelegard , I wanted a little more advise as currently I am building and installing on 24.04 in an emulator. My set up is a test bench for the real system that is in RHEL. I assume you would suggest that RHEL be the most updated version as well for the build to go smoothly?

lelegard, 2024-11-14:
> I assume you would suggest that RHEL be the most updated version as well for the build to go smoothly?

Yes, indeed. Especially because RHEL is a "stable" distro, meaning that everything is outdated, even in the most recent version of RHEL. AFAIK, the most recent RHEL is 9.4. However, it does not even include RIST. You have to use the community-driven EPEL (Extra Packages for EL). And there, in EPEL 9, you only have librist version 0.2.7 (released in Apr 2022) while the latest version of the project, librist 0.2.10, has been available for more than a year now (Oct 2023).

RHEL is stable in the sense it is appropriate for stable production systems running old but stable software. I must admit that the Red Hat world is very late about RIST. Even Fedora 41, the most bleeding edge distro, running on kernel 6.11, is using the same 0.2.7 version. A 2-years old package on Fedora! Quite unusual.

It seems that Ubuntu 24.04 remains the best choice for RIST.

lelegard, 2024-11-14:
> i cd into the installers folder there are no binaries in there. 

That is not normal. It you run `make installer` and don't get the installer, you must have an error message and need to investigate.

>  emulator 

Just curious, what do you mean by "emulator"? Something like qemu or a virtual machine?

fluffy-celery, 2024-11-14:
currently I am attempting to stand up an instance on EVE-NG, I also do not get an error when i run `make installer`. Currently the emulator environment has been updated to 24.04

fluffy-celery, 2024-11-14:
username@server:~$ git clone https://github.com/tsduck/tsduck.git
Cloning into 'tsduck'...
remote: Enumerating objects: 84927, done.
remote: Counting objects: 100% (113/113), done.
remote: Compressing objects: 100% (85/85), done.
remote: Total 84927 (delta 52), reused 49 (delta 28), pack-reused 84814 (from 1)
Receiving objects: 100% (84927/84927), 40.80 MiB | 9.75 MiB/s, done.
Resolving deltas: 100% (74164/74164), done.
username@server:~$ cd tsduck
username@server:~/tsduck$ scripts/install-prerequisites.sh
Packages: git g++ cmake flex bison dos2unix curl tar zip linux-libc-dev dpkg-dev python3 libssl-dev libedit-dev pcscd libpcsclite-dev libcurl4 libcurl4-openssl-dev libsrt-dev libusb-1.0-0-dev default-jdk doxygen graphviz asciidoctor qpdf
[sudo] password for username: 
Sorry, try again.
[sudo] password for username: 
Hit:1 http://us.archive.ubuntu.com/ubuntu focal InRelease
Get:2 http://us.archive.ubuntu.com/ubuntu focal-updates InRelease [128 kB]                                       
Hit:3 https://dl.google.com/linux/chrome/deb stable InRelease                                                    
Ign:4 https://download.docker.com/linux/ubuntu $ InRelease                                                       
Hit:5 http://security.ubuntu.com/ubuntu focal-security InRelease                                                 
Hit:6 https://download.docker.com/linux/ubuntu focal InRelease                                                   
Err:7 https://download.docker.com/linux/ubuntu $ Release                                                         
  404  Not Found [IP: 13.32.151.52 443]
Hit:8 http://us.archive.ubuntu.com/ubuntu focal-backports InRelease
Hit:9 http://ppa.launchpad.net/smartfinn/eve-ng-integration/ubuntu focal InRelease
Reading package lists... Done
E: The repository 'https://download.docker.com/linux/ubuntu $ Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
bison is already the newest version (2:3.5.1+dfsg-1).
default-jdk is already the newest version (2:1.11-72).
flex is already the newest version (2.6.4-6.2).
g++ is already the newest version (4:9.3.0-1ubuntu2).
libedit-dev is already the newest version (3.1-20191231-1).
libpcsclite-dev is already the newest version (1.8.26-3).
libusb-1.0-0-dev is already the newest version (2:1.0.23-2build1).
python3 is already the newest version (3.8.2-0ubuntu2).
zip is already the newest version (3.0-11build1).
asciidoctor is already the newest version (2.0.10-2).
dos2unix is already the newest version (7.4.0-2).
doxygen is already the newest version (1.8.17-0ubuntu2).
graphviz is already the newest version (2.42.2-3build2).
libsrt-dev is already the newest version (1.4.0-1build1).
pcscd is already the newest version (1.8.26-3).
cmake is already the newest version (3.16.3-1ubuntu1.20.04.1).
curl is already the newest version (7.68.0-1ubuntu2.24).
dpkg-dev is already the newest version (1.19.7ubuntu3.2).
git is already the newest version (1:2.25.1-1ubuntu3.13).
libcurl4 is already the newest version (7.68.0-1ubuntu2.24).
libcurl4-openssl-dev is already the newest version (7.68.0-1ubuntu2.24).
libssl-dev is already the newest version (1.1.1f-1ubuntu2.23).
linux-libc-dev is already the newest version (5.4.0-200.220).
tar is already the newest version (1.30+dfsg-7ubuntu0.20.04.4).
qpdf is already the newest version (9.1.1-1ubuntu0.1).
0 upgraded, 0 newly installed, 0 to remove and 63 not upgraded.
ERROR:  Error installing asciidoctor-pdf:
	The last version of public_suffix (>= 2.0.2, < 7.0) to support your Ruby & RubyGems was 5.1.1. Try installing it with `gem install public_suffix -v 5.1.1` and then running the current command again
	public_suffix requires Ruby version >= 3.0. The current ruby version is 2.7.0.0.
Successfully installed rouge-4.5.1
Parsing documentation for rouge-4.5.1
Done installing documentation for rouge after 3 seconds
1 gem installed
username@server:~/tsduck$ gem install public_suffix -v 5.1.1
Fetching public_suffix-5.1.1.gem
ERROR:  While executing gem ... (Gem::FilePermissionError)
    You don't have write permissions for the /var/lib/gems/2.7.0 directory.
username@server:~/tsduck$ sudo su
root@server:/home/username/tsduck# gem install public_suffix -v 5.1.1
Fetching public_suffix-5.1.1.gem
Successfully installed public_suffix-5.1.1
Parsing documentation for public_suffix-5.1.1
Installing ri documentation for public_suffix-5.1.1
Done installing documentation for public_suffix after 0 seconds
1 gem installed
root@server:/home/username/tsduck# ls
CHANGELOG.txt  CONTRIBUTORS.txt  images       Makefile      OTHERS.txt  README.md  scripts  TODO.txt
CONFIG.txt     doc               LICENSE.txt  Makefile.inc  pkg         sample     src
root@server:/home/username/tsduck# cd pkg
root@server:/home/username/tsduck/pkg# ls
build-remote.sh  deb  docker  github  homebrew  installers  Makefile  nsis  README.md  rpm  winget
root@server:/home/username/tsduck/pkg# cd installers
root@server:/home/username/tsduck/pkg/installers# ls
README.md

I can not copy anything from the emulator but this was what happened on my ubuntu 20.04 machine yesterday when i redid the build and installation process, you can see only the README is in the installers folder

lelegard, 2024-11-14:
You did not run `make installer` so you do not get any installer.

Why do you use an emulator? This is really looking for trouble. From the error messages during the installation of the prerequisites, this system does not look like a standard Ubuntu 24.04.

Either you use a standard Ubuntu on a standard PC, possibly in a VirtualBox virtual machine for simplicity, and you install the TSDuck binaries, and everything will run fine.

Or you use this weird EVE-NG emulator and I cannot help you to configure it because  I do not know it or its peculiarities.

fluffy-celery, 2024-11-14:
This is the only instance that I have from the previous night on my 20.04 machine (I can not update this machine for specific reasons hence the usage of EVE) and I can say trust me I did run `make installer` this particular instance does not show it but I have done that and the installer is still empty with no error messages. The same thing happened in the emulator despite being on 24.04. The emulator is the exact same thing as a VM and a computer it allows me complete control over any environment. View it as just a bare metal server running 24.04 which it essentially is.

Could it be because I have attempted to install in the past, `rm -rf tsduck` to redo that process and messed something up?

lelegard, 2024-11-14:
When you run `make installer`, you have hundredths of lines of display, indicating what is compiled. Please run `make installer` and post all its output.

I do not know what is EVE. If this is a virtualization environment with native code, you may get it right. However, if this is a real emulator, in the sense of qemu for instance, then rebuilding may take hours, maybe half a day. If it runs a real Ubuntu 24.04 on 64-bit Intel ISA, then use the binaries, don't try to rebuild.

fluffy-celery, 2024-11-14:
[image]

completely not sure why it has decided to behave THIS time, I ran the exact same thing as I did many times before. 

Now I assume that running a RIST test would simply just be `tsp -I rist url remote-ip:port`? would I also need `tsduck` and `librist` on that destination?

closed 2024-11-18 by fluffy-celery