tsduck/tsduck #1551: [FEAT]

dbhat-peacock, 2024-11-17:
#### Is your feature request related to a problem?
TSDuck does not seem to support ipv6 addresses in tools like tslatencymonitor. But recently there is need to be able to support ipv6 addresses. While TSDuck seems to support the porgramming structures for ipv6, it is not part of tools. I mention tslatencymonitor specifically but I think ti is not support with tsp when used with -P analyze etc.

#### Proposed solution:
Support the use of ipv6 addresses explicitly and DNS tied to ipv6 addresses.

lelegard, 2024-11-17:
Hi @dbhat-peacock 

Currently, as you noticed, IPv6 adresses are supported in data structures, essentially because there are DVB descriptors which contain IPv6 addresses.

However, for its networking features, TSDuck uses IPv4 only. I understand that, in 2024, this is probably insane and needs to be fixed.

There is no excuse for that, only historical reasons.
- Even though IPv6 is 30 years old, its wide adoption is recent. When the ancestor of TSDuck started in 2005, no one used it.
- TSDuck is a test tool, for lab experiments, and internal networks of all labs support IPv4, sometimes alongside IPv6.
- Outside point-to-point communication, IP multicast for TV always use IPv4 in practice. In theory, it can use IPv6 but I have never seen this so far.

So, no excuse again, this needs to be done. Now, it is a matter of resources. If you look at the code, the impact is spread all over the project (nearly 400,000 lines of C++). It also needs an IPv6 infrastructure to test.

TSDuck is a free and open source project which is maintained on spare time only. No company is funding it, unfortunately. So, general migration to IPv6 should be done. Will it happen in the short term? Unfortunately, I do not think so, for resource reasons.

dbhat-peacock, 2024-11-18:
Thanks @lelegard for the clarification. I can see the tremendous amount of work needed for this feature.

ArianitC, 2024-11-21:
After spending some time on this task, I must report that I have not come up with a successful solution and probably not a very good suggestion either. Previous solution was using _tslatency_ monitor, a plugin of _TSDuck_. That unfortunately does not support IPv6 addressing that was confirmed by _TSDuck_ when our colleague, **Dinkar Bhat**, asked on their discussion board. It can be seen, and that was confirmed by them, that this is significant piece of work: 

> “the impact is spread all over the project (nearly 400,000 lines of C++). It also needs an IPv6 infrastructure to test.”.

I have tried to make use of pithing script, using _ffprobe_ - _ffmpeg_ tool, but was not successful. Reason being that “probing” is done separately on each stream/source. Depending on the machine, although multithreading is advertised as being parallel processing, depending on a machine, it is never truly parallel. So, difference will never be a true representation. I have used _ffprobe_ command (below) to get _pkt_dts pkt_pos best_effort_timestamp_ of video Key IDR frames of one SRT input.
`ffprobe -v 0 -show_entries frame=pkt_dts,pkt_pos,best_effort_timestamp -skip_frame nokey -of compact=p=0:nk=1 -select_streams v:0 "srt://egress-somehexvalue.lrs.vos360.video:7086?passphrase=obfuscatedpassphrase"`
this produced values as in the snippet below: 
**Central**
pkt_dts|	        best_effort_timestamp|	pkt_pos
825406384|	825053584|	                 5249524
825586384|	825233584|	                 12028616
825766384|	825413584|	                 19332416
**West**
pkt_dts|	        best_effort_timestamp|	pkt_pos
825406384|	825053584|	                 1787128
825586384|	825233584|	                 8566220
825766384|	825413584|	                 15870020

Aligning them with respect to _best_effort_timestamp_ and then finding the difference, it produces nothing meaningful (very much expected) as below:
pkt_dts|	best_effort_timestamp|   pkt_pos
0|	                            0|                    3462396
0|	                            0|                    3462396
0|	                            0|                    3462396
If there is no common reference time point with which the PTS values of each side will be compared against the above is of no use.
Was expected that maybe Packet position to provide constant reference but it is shown that it cannot be used as reference point of comparison. 
**Suggestion for progressing**: 
I believe that our best chance is the news that our partner Harmonic do seem to have a tool that does exactly what we need. Been informed that it is not for external consumption. If we think we truly need this tool maybe exercising of some persuasion is needed by higher management. If only revealed to us as a form of API that we can use.

lelegard, 2024-11-28:
@dbhat-peacock, it seems that the job is not so big. After some refactoring in the base IP classes, we can have a reasonable IPv6 support.

IPv6 unicast works. IPv6 multicast is implemented but without a proper infrastructure to test, it is difficult to give a reliable conclusion. The problem is the combination of operating systems which all have their specificities, Linux, macOS and Windows. The 4 BSD systems have not been tested yet. The IPv6 interface is very different from the IPv4. In practice, everything but basic send/receive is different and needed to be modified. Even UDP receive is different (if you want to get the destination address). However, because everything is encapsulated in a small number of classes, the modification were relatively local.

If you are interested, please try the latest state of the repo ([building instructions here](https://tsduck.io/download/docs/tsduck-dev.html#building)). I cannot guarantee that IPv6 multicast will work at once, but we precisely need users' feedback to improve it.

My understanding is that "source-specific multicast" (SSM) does not exist in IPv6. Can anyone confirm this?

Which operating system do you use?

NB: Have a look [here](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/base/network/tsUDPSocket.cpp) to see how IPv6 programming is different from IPv4. Some functions need to separately process four cases, the combinations of IPv4/v6 and unicast/multicast, not including the `#ifdef` per operating system.

lelegard, 2024-11-30:
@dbhat-peacock, I have successfully tested various usages of IPv6 with TSDuck, including TCP connections and UDP unicast streaming. UDP multicast has not been tested, mainly because of a lack of experience on these infrastructure. IPv6 pcap captures are also supported.

There is no support so far for SRT and RIST with IPv6. We should see with the two teams if and how this is possible.

I close this issue as "IPv6 support" is mostly done. Please test it and open dedicated issues for specific problems you may have with IPv6.

closed 2024-11-30 by lelegard