tsduck/tsduck #1553: bin/tsfclean can be destructive

GerryHickman, 2024-11-23:
I have some *.ts files containing dvb recordings; these files play fine in "all" players, but I wanted to run 'tsfclean' on them.
The first one worked ok, but the second one broke in three different ways:

1. Thumbnail broken
2. File Properties blank
3. File won't play

Original file:

```
bin/tsp -I file /mmedia/rec/dvb/myfile.ts -P history -O drop
* history: 0: PID 0x18A1 (6305) first packet, clear
* history: 3: PID 0x01A8 (424) first packet, clear
* history: 5: PID 0x18A2 (6306) first packet, clear
* history: 18: PID 0x18A1 (6305), PES stream_id is 0xE0 (Video 0)
* history: 50: PID 0x0000 (0) first packet, clear
* history: 51: PAT v5, TS 0x3006 (12294)
* history: 64: PID 0x18A2 (6306), PES stream_id is 0xC0 (Audio 0)
* history: 150: PMT v23, service 0x3910 (14608)
* history: 1189: PID 0x18A1 (6305), first intra-frame, MPEG-2 Video, service 0x3910 (14608)

```

the video pid starts at packet index 0, and the first intra-frame is at 1189
the audio starts at packet 5, and the first pusi is at 64
there are 107 audio packets before the first video intra-frame

When tsfclean runs, it copied 107 audio packets into the new stream, before the first video packet, so the video starts "late"

Clean file

```
bin/tsp -I file /mmedia/rec/dvb/avclean.ts -P history -O drop
* history: 0: PID 0x0000 (0) first packet, clear
* history: 1: PAT v0, TS 0x3006 (12294)
* history: 2: PID 0x01A8 (424) first packet, clear
* history: 2: PMT v0, service 0x3910 (14608)
* history: 6: PID 0x18A2 (6306) first packet, clear
* history: 6: PID 0x18A2 (6306), PES stream_id is 0xC0 (Audio 0)
* history: 131: PID 0x18A1 (6305) first packet, clear
* history: 131: PID 0x18A1 (6305), PES stream_id is 0xE0 (Video 0)
* history: 131: PID 0x18A1 (6305), first intra-frame, MPEG-2 Video, service 0x3910 (14608)

```

The audio now starts at packet index 6, and the video starts at packet index 131, there are still 107 audio packets before the video starts (just like before, but they're squashed together now).

Turns out, this is affecting gstreamer with the playbin element.
It starts reading packets, but there's only audio packets, and it can't find the PCR.
AVFormat players do some extra seeking looking for the PCR

I did an experiment with tsfclean, where it waits for the first video intra-frame before writing the audio packets. This did fix the issue, but it's only an experiment, it's not a real fix.

```
diff --git a/src/tstools/tsfclean.cpp b/src/tstools/tsfclean.cpp
index e92147fd1..7951f2160 100644
--- a/src/tstools/tsfclean.cpp
+++ b/src/tstools/tsfclean.cpp
@@ -10,6 +10,8 @@
 //
 //----------------------------------------------------------------------------
 
+// GH test version
+
 #include "tsMain.h"
 #include "tsDuckContext.h"
 #include "tsSignalizationDemux.h"
@@ -208,6 +210,7 @@ ts::FileCleaner::FileCleaner(FileCleanOptions& opt, const fs::path& infile_name)
 
     // In second pass, count input packets per PID.
     std::map<PID,PacketCounter> pkt_count;
+    bool bVideoStarted = false;
 
     // Second pass: read input file again, write output file.
     while (_success && _in_file.readPackets(&pkt, nullptr, 1, _opt) == 1) {
@@ -240,7 +243,7 @@ ts::FileCleaner::FileCleaner(FileCleanOptions& opt, const fs::path& infile_name)
         else if (pid_class == PIDClass::AUDIO || pid_class == PIDClass::SUBTITLES || pid_class == PIDClass::DATA) {
             // Write these packets transparently after the first payload unit start.
             const PacketCounter first_index = sig.pusiFirstIndex(pid);
-            if (first_index == INVALID_PACKET_COUNTER || pkt_index >= first_index) {
+            if ((first_index == INVALID_PACKET_COUNTER || pkt_index >= first_index) && bVideoStarted) {
                 _success = _success && _out_file.writePackets(&pkt, nullptr, 1, _opt);
             }
         }
@@ -251,6 +254,7 @@ ts::FileCleaner::FileCleaner(FileCleanOptions& opt, const fs::path& infile_name)
                 first_index = sig.pusiFirstIndex(pid);
             }
             if (first_index == INVALID_PACKET_COUNTER || pkt_index >= first_index) {
+                bVideoStarted = true;
                 _success = _success && _out_file.writePackets(&pkt, nullptr, 1, _opt);
             }
         }

```

lelegard, 2024-11-24:
Hi @GerryHickman 

Thank you for this analysis. 

I would say that the output file is "correct" and that the player is to blame for not being able to play it. The PCR clock is unchanged, the PTS and DTS remain unchanged either. The order of TS packets is unchanged (for packets which remain) and conditions and timing of reception and processing are unchanged.

Concerning the two other points, this is not a matter of file content but how external applications can see the file:

- "Thumbnail broken". The thumbnail is not part of the file. This is something your file viewer is building. I don't know under which conditions it can recreate one or not. Again, in that case, the file viewer is to blame, not the file itself.
- "File Properties blank". What do you name "file properties"? Ownership and protection? In that case, the output file is created by `tsfclean` just like any other output file, with the default properties of the user (`umask` on all Unix systems).

All of that being said, we may try to improve the way some players or file viewers can process the output file.

About initial clock synchronisation: We could keep the video packets which contain a PCR before the first intra-frame. Of course, we delete the payload. However, because the video must be set as "PCR PID" in the PMT, and the PMT comes before any content PID, that may help. These packets are typically found at the start of each initial non-intra frames.

About audio and video synchronisation: The reference is the PTS, not the order of packets in the file. For a given PTS, audio usually comes after video in streams. So, instead of dropping audio packets before the first video intra-frame, it would be better to remove all audio packets with a PTS which is lower (minus some epsilon) than the PTS of the first video intra-frame. If course, this works only if the audio and video are in the clear.

PS: Just a pedantic stylish note, please don't use Hungarian naming, it has been universally hated for decades. Only Microsoft is still using it (which may be a reason for that universal detestation 😉, but not only IMHO).

lelegard, 2024-11-24:
Hi @GerryHickman 

I have implemented the synchronisation of each audio PID on its corresponding video PID. The audio PID's start on next PUSI with PTS at or after the PTS of the first intra-frame of the video of the same service. If no PTS can be found, the audio starts at the next PUSI after the associated video started.

Please try the latest state of the repo (v3.40-3969) and let us know the result in your environment.

Tested tonight on macOS, other platforms will be tested in the GitHub CI during the night.

GerryHickman, 2024-11-25:
Here's a more simplistic way of looking at the original problem with the original files (history above) and original tsfclean.

```
PCR pid 0x18A1
bin/tsp -I file /mmedia/rec/dvb/myfile.ts -P until --packet 256 -P pcrextract --pid 0x18A1 --pcr -O drop
6305,18,15,PCR,1,1182525816220,0,
6305,42,36,PCR,2,1182526731767,915547,
6305,69,58,PCR,3,1182527683224,1867004,
6305,94,80,PCR,4,1182528622711,2806491,
6305,120,103,PCR,5,1182529583143,3766923,
6305,145,125,PCR,6,1182530522631,4706411,
6305,171,147,PCR,7,1182531453144,5636924,
6305,200,170,PCR,8,1182532442001,6625781,
6305,225,192,PCR,9,1182533375505,7559285,
6305,249,213,PCR,10,1182534276098,8459878,

```
gstreamer starts reading packets, and "instantly" finds the PCR at packet 18

```
PCR pid 0x18A1
bin/tsp -I file /mmedia/rec/dvb/avclean.ts -P until --packet 256 -P pcrextract --pid 0x18A1 --pcr -O drop
6305,153,20,PCR,1,1182560746328,0,
6305,186,50,PCR,2,1182561715736,969408,
6305,218,79,PCR,3,1182562646247,1899919,
6305,253,110,PCR,4,1182563573766,2827438,

```

gstreamer starts reading (and reading), but yikes, no PCR until packet 153

lelegard, 2024-11-26:
To cope with the situation where PCR are not in PUSI packets (start of PES packets), `tsfclean` now passes video packets preceding the first intra-frame when they contain a PCR. In that case, the payload is deleted (PCR-only packet).

You need version 3.40-3975 or higher.

GerryHickman, 2024-11-30:
I'm looking forward to testing the new version of 'tsfclean', but I still don't know if there was a problem with the old one. The tests above are using 'tsfclean' from tsduck 3.38-3822. In your first reply, you said the output file was correct and the PCR had not changed.

Do you still have access to 'tsfclean' from tsduck 3.38-3822?

I'll try to make a smaller version of the test file above, I didn't know you could upload things here.

lelegard, 2024-11-30:
> Do you still have access to 'tsfclean' from tsduck 3.38-3822?

All versions are available here: https://github.com/tsduck/tsduck/releases

And with the magic of git, we can rebuild any older version.

I can only say that `tsfclean` has never, ever, modified a packet, especially not a PCR value. There is simply no code for that in `tsfclean`. It deletes some packets at the beginning of the file and it replaces the PAT/PMT/EIT signalisation, that's all.

Now, I agree that the previous version, up to v3.39, let pass some audio PES before the video in a service. Since the PCR are in the video PID, the player receives some audio PES before the first PCR of the service. If this is a problem for the player, this is a player's issue, not a file issue. That being said, TSDuck remains pragmatic and implements ways of fixing bugs, much more important bugs, in various receivers (there are a lot of options for that). So, the new version (the current development state of the repo) changes this and now tries to align the start of audio with the start of video.

I will probably not reinstall / rebuild v3.38 to test a file. However, to solve your issue, I recommend that you test the new version on your file and player.

GerryHickman, 2024-11-30:
I understand where you say you don't want to test with older version.

So my understanding of the original issue is not that tsfclean changed a PCR packet, but rather it created a gap before the first PCR (e.g. audio packets first). It does raise an interesting question; how many ts packets is a player expected to read before it decides there's no PCR?

lelegard, 2024-11-30:
> So my understanding of the original issue is not that tsfclean changed a PCR packet, but rather it created a gap before the first PCR (e.g. audio packets first).

Exactly.

> It does raise an interesting question; how many ts packets is a player expected to read before it decides there's no PCR?

And the answer is most probably "it depends".

It depends on the receiver or player. I have seen so many different behaviours in receivers, many of them being buggy. The desktop software players are usually more resilient. VLC, for instance, is ubiquitous. It decodes virtually everything. If VLC doesn't decode, it is probably a problem with the stream. If a set-top box doesn't decode, well, usually the set-top box is to blame.

GerryHickman, 2024-12-01:
This raises a different question:

Is there any way to determine if a transport stream is "well formed".

e.g. imagine a stream where there's 10 minutes of audio before the PCR and video.

'tsanalyze' already has the --error-analysis option, but that's slightly different.

lelegard, 2024-12-01:
> Is there any way to determine if a transport stream is "well formed".

What is the definition of "well formed"?

I don't think that this definition exists in the most general sense. As long as it is conformant with ISO 13818-1, we can say it is "well formed". Then, DVB added a few standards for "conformant broadcast stream" (which are sometimes incompatible with other broadcasting standards such as SCTE-35). However, ISO 13818-1 implicitly defined TS as continuous broadcast format, without begin or end. In that context "starting audio before video" is meaningless because there is no "beginning". A storage format has a beginning. However, for ISO 13818-1, the storage format is program stream (DVD), not transport stream. And after DVD, the BluRay storage format switched to TS (or "M2TS"), which was not anticipated. BluRay must have its own set of requirements, I guess (I don't know them). And then HLS specified the requirement for the start of a TS segment file (PAT, PMT, intra-frame, but nothing about audio/video starting order as far as I remember). 

So, what is a "TS file"? A slice of broadcast stream? A BluRay M2TS? A HLS segment? Stuff that was generated by ffmpeg or other tools, without formal definition? 

I am afraid that the quest for the universal "well formed" transport stream is vain...

Let's be pragmatic and let's generate something that is widely accepted.

GerryHickman, 2024-12-01:
Thanks for the update about trying to define what "well formed" means.

I've tried to re-build at this commit:

```
commit 1eb1bea64eac0f1e30620e43b6c2c2f23f5f5459 (HEAD -> gh_build, origin/master, origin/HEAD, master)
Author: Thierry Lelegard <thierry@lelegard.fr>
Date:   Sun Dec 1 00:49:35 2024 +0100
    Fixed issue in AVC and EVC descriptors (issue #1558)
```

but I'm getting an error from Make

make[2]: *** No rule to make target '/data/build/tsduck/src/libtsduck/base/network/tsIPv4Address.h',
needed by '/data/build/tsduck/bin/release-x86_64-gh01/objs-libtsduck/tsATSCMultipleString.o'.  Stop.

this has never happened before

lelegard, 2024-12-01:
Run `make clean` first. You probably still have `.dep` files from builds of previous versions. Some recent updates have unified IPv4 and IPv6 classes. The distinct classes for IPv4 and IPv6 have disappeared but you clearly still have dependencies on them.

GerryHickman, 2024-12-01:
After running 'make clean', the build now completes, and all tsduck programs are working.

A quick test on a broken file from yesterday, now works.

- Thumbnail is now created (was broken)
- Duration is now correct (was zero)
- File Properties, now populated (were empty)
- File Seeking was broken (now working accurate)
- File is able to play.

I'll now re-run the more detailed 'tsfclean' tests above.

GerryHickman, 2024-12-01:
Regarding "well formed" transport streams, I should have said containing DVB in SD.

I found this document

https://download.tek.com/document/25W_14706_0.pdf

On Page 2, it says
#"PCRs
are snapshots of a counter, driven by this program clock, that are
inserted into packets within the Transport Stream (TS) at more or less
regular intervals (ISO/IEC13818-1 specifies an interval of 100 ms,
while DVB recommends 40 ms)."

lelegard, 2024-12-02:
> Regarding "well formed" transport streams, I should have said containing DVB in SD.

Precisely. DVB is a broadcast standard (sometimes IP multicast). There is no concept of "DVB file". A DVB transport stream is a continuous stream without beginning or end. So, there is nothing about "starting" audio before video. Anyway, the concept of "TS file" (DVB or not) is undefined. TV engineers don't manipulate "TS files", they manipulate "slices of broadcast" which are meaningless on ends. We just have tools, with their limitations. And we try to adapt when necessary.

GerryHickman, 2024-12-03:
I have now carried out more detailed testing with tsfclean v3.39-3956.
I used the exact same file as in the original post.
The original post shows the history before running tsfclean, and then after running the _old_ tsfclean.
The old tsfclean was causing files to become unplayable due to skipping some PCR packets

The new tsfclean v3.39-3956

```
* history: 0: PID 0x0000 (0) first packet, clear
* history: 1: PAT v0, TS 0x3006 (12294)
* history: 2: PID 0x01A8 (424) first packet, clear
* history: 2: PMT v0, service 0x3910 (14608)
* history: 4: PID 0x18A1 (6305) first packet, clear
* history: 61: PID 0x18A1 (6305), PES stream_id is 0xE0 (Video 0)
* history: 61: PID 0x18A1 (6305), first intra-frame, MPEG-2 Video, service 0x3910 (14608)
* history: 853: PID 0x18A2 (6306) first packet, clear
* history: 853: PID 0x18A2 (6306), PES stream_id is 0xC0 (Audio 0)
```
pcr pid 0x18A1
video pid 0x18A1 now starts much earlier at packet index 4, was 131
audio pid 0x18A2 now starts much later at packet index 853, was 64

```
pcr pid 0x18A1
6305,4,0,PCR,1,1182525816220,0,
6305,5,1,PCR,2,1182526731767,915547,
6305,8,2,PCR,3,1182527683224,1867004,
6305,9,3,PCR,4,1182528622711,2806491,
6305,10,4,PCR,5,1182529583143,3766923,
6305,11,5,PCR,6,1182530522631,4706411,
6305,13,6,PCR,7,1182531453144,5636924,
6305,16,7,PCR,8,1182532442001,6625781,
```
The file is able to play

BUT

bin/tsp -I file /mmedia/rec/dvb/avclean.ts -P continuity -O drop

```
* continuity: packet index: 5, PID: 0x18A1 (6305), missing 4 packets
* continuity: packet index: 8, PID: 0x18A1 (6305), missing 5 packets
* continuity: packet index: 9, PID: 0x18A1 (6305), missing 5 packets
* continuity: packet index: 10, PID: 0x18A1 (6305), missing 6 packets
* continuity: packet index: 11, PID: 0x18A1 (6305), missing 5 packets
* continuity: packet index: 13, PID: 0x18A1 (6305), missing 5 packets
* continuity: packet index: 16, PID: 0x18A1 (6305), missing 6 packets
* continuity: packet index: 17, PID: 0x18A1 (6305), missing 5 packets
* continuity: packet index: 18, PID: 0x18A1 (6305), missing 4 packets
* continuity: packet index: 19, PID: 0x18A1 (6305), missing 6 packets
* continuity: packet index: 21, PID: 0x18A1 (6305), incorrect CC increment without payload
* continuity: packet index: 22, PID: 0x18A1 (6305), incorrect CC increment without payload
* continuity: packet index: 28, PID: 0x18A1 (6305), incorrect CC increment without payload
* continuity: packet index: 30, PID: 0x18A1 (6305), incorrect CC increment without payload
* continuity: packet index: 31, PID: 0x18A1 (6305), incorrect CC increment without payload
* continuity: packet index: 34, PID: 0x18A1 (6305), incorrect CC increment without payload
* continuity: packet index: 35, PID: 0x18A1 (6305), missing 14 packets
* continuity: packet index: 36, PID: 0x18A1 (6305), missing 13 packets
* continuity: packet index: 37, PID: 0x18A1 (6305), missing 14 packets
* continuity: packet index: 39, PID: 0x18A1 (6305), missing 13 packets
* continuity: packet index: 40, PID: 0x18A1 (6305), missing 14 packets
* continuity: packet index: 43, PID: 0x18A1 (6305), missing 13 packets
* continuity: packet index: 44, PID: 0x18A1 (6305), missing 13 packets
* continuity: packet index: 45, PID: 0x18A1 (6305), missing 13 packets
* continuity: packet index: 46, PID: 0x18A1 (6305), missing 14 packets
* continuity: packet index: 48, PID: 0x18A1 (6305), missing 13 packets
* continuity: packet index: 51, PID: 0x18A1 (6305), missing 14 packets
* continuity: packet index: 52, PID: 0x18A1 (6305), missing 13 packets
* continuity: packet index: 53, PID: 0x18A1 (6305), missing 14 packets
* continuity: packet index: 54, PID: 0x18A1 (6305), missing 13 packets
* continuity: packet index: 56, PID: 0x18A1 (6305), missing 13 packets
* continuity: packet index: 57, PID: 0x18A1 (6305), missing 13 packets
* continuity: packet index: 60, PID: 0x18A1 (6305), missing 12 packets
* continuity: packet index: 61, PID: 0x18A1 (6305), missing 8 packets
```

lelegard, 2024-12-03:
I assume that this PID 6305 is the video PID, the one with PCR's. This is clearly the early PCR packets which were kept (with payload removed), while other packets were removed. If your listing is complete, the discontinuities stop when the video packets start. So, because intermediate packets were removed, this is the expected behavior. 

Now, is it a problem ? Why would this be a problem ?

The CC are here to detect missing packets in an elementary stream of PES or section content. The discontinuities indicate that the flow of payload content becomes invalid because there is some missing part. All those packets have no payload, no content. So, there is no missing part and no problem. 

If you are visually displeased by the output of the plugin `continuity`, you can can use the same plugin to "fix" them. However, so-called "fixing" discontinuities is sometimes worse than keeping them when they hide actual problems. If your input file has discontinuities later, then "fixing" the CC will hide them and the video codec may display garbage instead of skipping a frame or part of one, which is invisible to the human eye. 

Many people focus on CC errors and want them to disappear but they are useful.

GerryHickman, 2024-12-05:
The packets with the continuity errors are between 0 and 61, which agrees with the history above where PCR pid starts at 61, so yes they can be ignored, but the original file shows no errors, and running a "cleaning" tool creates a file with errors.

I need to re-think what tsfclean is all about. The placing of the PSI/SI works well. I'm not so sure about the PES packets.

lelegard, 2024-12-06:
I have aligned all continuity counters at the beginning of the video PID, the packets with PCR only and no payload, with the continuity counter of the first packet of the intra-frame. Therefore, there is no longer any reported discontinuity at the beginning of the output. It does not change the usage of the file but it will avoid future remarks about reported discontinuities (although harmless in that context). 😉 

Note that if there is any further discontinuity in the input file, there are propagated unchanged to the output, _because this is the right thing to do_.

GerryHickman, 2024-12-08:
I've now tested the updated version of 'tsfclean'.
History and PCR offsets (on the clean file) are the same as above.
Continuity errors are gone.
File is able to play.
Issue can be closed.

Before posting this issue, I was going to run 'tsfclean' in batch mode, and delete the original files, but then noticed the "clean" files were unable to play, luckily I kept the originals.

closed 2024-12-08 by lelegard