tsduck/tsduck #1554: [Q] cc errors after tsp process

ToniTram81, 2024-11-25:
Hello,
I change a mpeg ts with tsp (injecting/replacing 2 tables, merging 1 service). Before the tsp process the ts has 0 cc errors, after the tsp process about 1000 per day. The ts stream has enough null bits (>500 kbit/s). The load of the server is about 9.
Even if I just change name als provider in the stream, cc errors occur.
How can I improve the output signal?

lelegard, 2024-11-25:
From that description, we haven't the slightest idea of what you are doing. Please provide the exact commands you use and the analysis of the symptoms.

If you use UDP output, the common cause is the burst effect which has been described many times here, with possible solutions.

lelegard, 2024-12-04:
It may be the internal buffering which creates latencies and bursts. Try `tsp --max-flushed-packets 7 --max-input-packets 7 --max-output-packets 7` in the two commands, the main one and the one in the merge plugin. This forces the transmission from one plugin to the next one every 7 packets (the output granularity). By default, it waits for more packets. Transmitting packets more often eats more CPU but it reduces the latency.

closed 2025-05-18 by github-actions[bot]