tsduck/tsduck #1555: [Q] error running test command

fluffy-celery, 2024-11-26:
Hi again, I am getting this error when im running the rist input/output test on the server. Been scratching my head for a while wondering how to resolve this? Any pointers would be appreciated!

root@ubuntu:~# tsp -I rist url rist://(server-ip-address):8194
* Error: rist: Failure resolving host url: Temporary failure in name resolution
* Error: rist: Host url cannot be resolved
* Error: rist: Send failed: errno=88, reason=Socket operation on non-socket, ret=-1, socket=0
* Error: rist: Send failed: errno=88, reason=Socket operation on non-socket, ret=-1, socket=0
* Error: rist: Send failed: errno=88, reason=Socket operation on non-socket, ret=-1, socket=0
* Error: rist: Send failed: errno=88, reason=Socket operation on non-socket, ret=-1, socket=0
* Error: rist: Send failed: errno=88, reason=Socket operation on non-socket, ret=-1, socket=0
* Error: rist: Send failed: errno=88, reason=Socket operation on non-socket, ret=-1, socket=0
* Error: rist: rist_send_common_rtcp failed
* Error: rist: rist_send_common_rtcp failed
* Error: rist: rist_send_common_rtcp failed
* Error: rist: rist_send_common_rtcp failed
* Error: rist: rist_send_common_rtcp failed

fluffy-celery, 2024-11-26:
Follow up: for the output I keep having to terminate the command because I keep getting this

root@ubuntu:~# tsp -O rist url rist://(server-ipa-address):8194
(literally nothing happens when i enter the command)
^C* tsp: user interrupt, terminating...
^C

lelegard, 2024-11-26:
@geoffroutledge, yes you got it. I didn't notice the extra `url` in ` -I rist url rist://(server-ip-address):8194`

The RIST URL is defined at https://code.videolan.org/rist/librist/-/wikis/risturl-Syntax-as-of-v.-0.2.0

fluffy-celery, 2024-11-26:
I will give this a try once I get my environment up and running again thank you!

closed 2025-05-04 by github-actions[bot]