tsduck/tsduck #1556: [BUG]

akkermansadriaan, 2024-11-27:
#### Bug description:
When using tsswitch to switch between files and the regulate plugin afterwards before sending the output over UDP, the regulate plugin stops regulating when using a video file made with ffmpeg.

This only seems to happen if the 2nd input is the ffmpeg file. If tsswitch starts with the ffmpeg file there's no issue.

#### How to reproduce:
First I use FFmpeg to create the ts file. I have noticed that -c copy doesn't have any effect. I have used big_buck_bunny_720_stereo-720p.mp4 as an example
```
ffmpeg -i big_buck_bunny_720_stereo-720p.mp4 -f mpegts big_buck_bunny_720_stereo-720p.ts
```
Afterwards 
```
tsswitch -I file input1.ts -I file big_buck_bunny_720_stereo-720p.ts | tsp -P regulate --pcr-synchronous -O ip 127.0.0.1:1234
```
If I switch `input1.ts` and `big_buck_bunny_720_stereo-720p.ts` the issue doesn't seem to occur.

#### Expected behavior:
The output remains regulated.

#### Errors and logs:
No relevant logs, even with verbose or debug=3. Log output is the same, except that the file is played way too fast, and consequently logs are produced at that (too fast) rate.

#### Environment:
 - OS: macOS 15.1.1 (but the same result on Linux machines)
 - TSDuck full version: 3.39-3956
 - Installation type: using brew

#### Additional information:
- FFmpeg version: ffmpeg version 7.1

lelegard, 2024-11-27:
Hi @akkermansadriaan 

The main question is: how are the two TS files compatible in terms of structure? Do they use the same TS id, same services with same PID's, same PMT, etc.

The tsp command receives a continuous stream of TS packets from the output of the pipe. When the `tsswitch` command switches from one file to another, the TS is completely replaced. However, at the input of tsp, this is still a continuous stream of TS packets. There is no way to tell that the input content was completely replaced.

So, if your two files do not have the same structure (not same services or PID's), the structure of the stream is completely changed without notice in the middle of the processing.

Now, look at how the plugin `regulate` works:
~~~
$ tsp -P regulate --help
...
  --pcr-synchronous
      Regulate the flow based on the Program Clock Reference from the transport
      stream. By default, use a bitrate, not PCR's.

  --pid-pcr value
      With --pcr-synchronous, specify the reference PID for PCR's. By default,
      use the first PID containing PCR's.
~~~

When `regulate` is initialized, it selects a reference PID for PCR's. If no option `--pid-pcr` is given, it lets packets pass until one PCR is found in one packet. The PID of that packet becomes the PCR reference PID. The regulation is based on PCR values on that PID.

So, if the stucture of your input suddenly changes in the middle of the stream and the PID which was previously selected as PCR reference disappears or no longer carries PCR's, the regulation is broken.

This is why I asked the question about the compatibility of the file structures. The second file being generated by ffmpeg, it contains only one service and the video PID (with PCR's) is probably 0x100, the default on ffmpeg.

In your first file, does PID 0x100 contain PCR's? More importantly, does it contain _other_ PID's with PCR's? Let's assume that the first file contains PCR's on several PID's, 0x100 and at least another one.
- If you start with the ffmeg-generated file, the plugin `regulate` finds a PCR in PID 0x100 and uses that PID for PCR reference. When you switch to the first file, PID 0x100 contains PCR and regulation will continue. There will be a glitch at the transition since the PCR values are suddenly discontinuous, but the regulation will rapidly stabilizes.
- On the other hand, if you start with the first file, and the first PCR in the file is in another PID than 0x100, this other PID will be used as PCR reference. When you switch to the second file, that PID no longer exists or does not contain PCR's, and the regulation is broken.

Could you please post the output of `tsanalyze` for each TS file so that we can check? Or check by yourself if you prefer.

The situation or explanation can be different but there is at least one plausible scenario for what you see.

akkermansadriaan, 2024-12-03:
Thanks @lelegard 

I updated the ffmpeg command to set the correct PID numbers and this resolved the issue. I included it for reference if someone encounters the same.
```
ffmpeg -i big_buck_bunny_720_stereo-720p.mp4 -f mpegts -mpegts_pmt_start_pid '0x01E0' -mpegts_start_pid '0x01E1' mpegts big_buck_bunny_720_stereo-720p.ts
```

closed 2024-12-03 by akkermansadriaan