tsduck/tsduck #1558: [BUG] Incorrect logic in AVC_timing_and_HRD_descriptor and EVC_timing_and_HRD_descriptor

EricBerendsen, 2024-11-30:
#### Bug description:
This is bug report is identical to #1065 (Incorrect logic in HEVCTimingAndHRDDescriptor), only this time it is about the AVC_timing_and_HRD_descriptor and EVC_timing_and_HRD_descriptor.

These descriptors have the same inverted logic for the 90kHz flag as the HEVCTimingAndHRDDescriptor.

As far as I can see the same problem has already been fixed for the VVC_timing_and_HRD_descriptor.

lelegard, 2024-11-30:
Thank you @EricBerendsen for reporting this. I just fixed it.

As you noticed, the VVC_timing_and_HRD_descriptor doesn't have the issue. I guess that the reason it that it was developed by @paulhiggs, who also reported issue #1065. I assume he correctly anticipated and avoided the problem.

paulhiggs, 2024-12-01:
I was about to push a fix for this as week. The 90kHz_flag indicates that the time base is 90khz. N and K are used to calculate the time base for other frequencies.

Generally the has_90khz should be interpreted ad is_90kHz

commit a24c39b5 (2024-12-02): handle timing and HRD descriptor for AVC, HEVC, EVC and VVC in a consistent way; N & K values only specified if not 90kHz. Fixes #1558
commit be923986 (2024-12-02): handle timing and HRD descriptor for AVC, HEVC, EVC and VVC in a consistent way; N & K values only specified if not 90kHz. Fixes #1558
closed 2024-12-02 by lelegard