tsduck/tsduck #1564: [BUG] Ambiguous access of '=='

paulhiggs, 2024-12-21:
#### Bug description:
Cannot build in Visual Studio 2022 using project supplied defaults.
Following errors are reported during the build process

Offending lines: 
tsSignalizationDemux.h(736,12):  
   [return ctx == _pids.end() ? no_value : (*ctx->second).*field;](https://github.com/tsduck/tsduck/blob/2ea0810edc921b838b34dbdfea5022d8c181e1e6/src/libtsduck/dtv/demux/tsSignalizationDemux.h#L736)

tsSignalizationDemux.h(743,12):
   [return ctx == _pids.end() ? no_value : (*ctx->second).*pp.*field;](https://github.com/tsduck/tsduck/blob/2ea0810edc921b838b34dbdfea5022d8c181e1e6/src/libtsduck/dtv/demux/tsSignalizationDemux.h#L743)

probably related to the `auto ctx =` declaration

#### Errors and logs:
~~~
89>D:\Documents\code\TSDuck\tsduck\src\libtsduck\dtv\demux\tsSignalizationDemux.h(736,12): error C2385: ambiguous access of '=='
89>(compiling source file '../../src/tsplugins/tsplugin_isdbinfo.cpp')
89>    D:\Documents\code\TSDuck\tsduck\src\libtsduck\dtv\demux\tsSignalizationDemux.h(736,12):
89>    could be the '==' in base 'ts::TableHandlerInterface'
89>    D:\Documents\code\TSDuck\tsduck\src\libtsduck\dtv\demux\tsSignalizationDemux.h(736,12):
89>    or could be the '==' in base 'ts::SectionHandlerInterface'
89>    D:\Documents\code\TSDuck\tsduck\src\libtsduck\dtv\demux\tsSignalizationDemux.h(719,11):
89>    This diagnostic occurred in the compiler generated function 'T ts::SignalizationDemux::getPIDContextField(ts::PID,const T &,T ts::SignalizationDemux::PIDContext::* ) const'
89>D:\Documents\code\TSDuck\tsduck\src\libtsduck\dtv\demux\tsSignalizationDemux.h(743,12): error C2385: ambiguous access of '=='
89>(compiling source file '../../src/tsplugins/tsplugin_isdbinfo.cpp')
89>    D:\Documents\code\TSDuck\tsduck\src\libtsduck\dtv\demux\tsSignalizationDemux.h(743,12):
89>    could be the '==' in base 'ts::TableHandlerInterface'
89>    D:\Documents\code\TSDuck\tsduck\src\libtsduck\dtv\demux\tsSignalizationDemux.h(743,12):
89>    or could be the '==' in base 'ts::SectionHandlerInterface'
89>    D:\Documents\code\TSDuck\tsduck\src\libtsduck\dtv\demux\tsSignalizationDemux.h(723,11):
89>    This diagnostic occurred in the compiler generated function 'T ts::SignalizationDemux::getPIDPointField(ts::PID,const T &,ts::SignalizationDemux::PIDPoint ts::SignalizationDemux::PIDContext::* ,T ts::SignalizationDemux::PIDPoint::* ) const'
~~~

#### Environment:
 - OS: Windows
 - OS version: 10
 - TSDuck full version: HEAD
 - Visual Studio:
   - Microsoft Visual Studio Community 2022 (2)
   - Version 17.11.5
   - VisualStudio.17.Release/17.11.5+35327.3
   - Microsoft .NET Framework
     - Version 4.8.09037
   - Installed Version: Community
   - Visual C++ 2022   00482-90000-00000-AA121   
   - Microsoft Visual C++ 2022

lelegard, 2024-12-21:
Hi @paulhiggs 

This is clearly a Visual Studio 17.11.5 bug. Not only it compiles with GCC and Clang, but it also compiles with Visual Studio 17.12.3 (the latest version). It also compiles in the Windows job of the Github CI (not sure which version of Visual Studio they use but they are generally reasonably up to date).

To help the compiler, I replaced the `auto ctx` with `const PIDContextMap::const_iterator ctx`. I hope it will work for you. Otherwise, I can only recommend to upgrade Visual Studio to the latest version.

paulhiggs, 2024-12-22:
Upgraded to latest Visual Studio to resolve the issue.

paulhiggs, 2024-12-22:
Could also add some specific workaround handling, i.e.
````
//WORKAROUND for bug in Microsoft Visual Studio 17.11
#if defined (_MSC_VER) && _MSC_VER <= 1941
    #pragma message("using workaround for Microsoft Visual Studio 17.11")
    #define _res_type const PIDContextMap::const_iterator
#else
    #define _res_type auto
#endif

template<typename T>
T ts::SignalizationDemux::getPIDContextField(PID pid, const T& no_value, T PIDContext::* field) const
{
    _res_type ctx = _pids.find(pid);
    return ctx == _pids.end() ? no_value : (*ctx->second).*field;
}

template<typename T>
T ts::SignalizationDemux::getPIDPointField(PID pid, const T& no_value, PIDPoint PIDContext::* pp, T PIDPoint::* field) const
{
    _res_type ctx = _pids.find(pid);
    return ctx == _pids.end() ? no_value : (*ctx->second).*pp.*field;
}
````

lelegard, 2024-12-22:
Hi Paul,

I don't think it is necessary to add that complexity. The usage of `auto` is really useful when the type depends on the context and is complicated to evaluate or express. Otherwise, it is just a way to compensate for the developer's laziness. If you confirm that providing the exact type solves the issue with the previous version of VS, I would prefer to keep it that way.

paulhiggs, 2024-12-22:
Sounds good @lelegard.
`const PIDContextMap::const_iterator` is a good solution - although I can admit to being lazy and going with `auto`

closed 2024-12-22 by paulhiggs