tsduck/tsduck #1566: [Q] Is it normal for the `scrambler` plugin to ignore the crypto-period when given fixed control words ?

zappynou, 2024-12-27:
Hello everyone !

I must mention that I've been testing this subject on versions 3.36, 3.37, and 3.39 on a Debian Sid x86_64 using the official .deb files and also after building the 3.39 myself. All commands below are using the official .deb of Debian x86_64, version 3.39.

The documentation from version 3.36 says "The next control word is used each time a new crypto-period is started", while more recently the version 3.39 only mentions the "During descrambling" case even for the `scrambler` plugin, nothing for scrambling : **I'm assuming the `scrambler` starts a new crypto-period by itself every `--cp-duration` seconds, is this correct ?**

In order to illustrate my question, I have a homemade clear TS which is 50+ seconds long (~16MB, made with ffmpeg, one program with three elementary streams -- 1 video on PID=0x100 and 2 audios). My goal is to only scramble the video for this example, with 10s crypto-period.

So, first I create a "cw" file containing a list of random values:

```
$ cat cw
0266839afcdda18acdc6e9228d343335
15879fa9072376acb957ca350035cbac
5971523e42ead2bbfdbd1d94e4209a5d
f714c8fdfd11e2f598e64bf44ed359f5
b4bbf412eb8d1f5388766ad71055872b
```

Then, I invoke tsp with the `scrambler` plugin:

```
$ tsp --debug -I file clear.ts -P scrambler --aes-cbc --cw-file cw --cp-duration 10 --pid 0x100 -O file scrambled.ts
tsp: Debug: ====> tsp --debug
* Debug: file: ====> tsp -I file clear.ts
* Debug: file: ====> tsp -O file scrambled.ts
* Debug: trying to load "/usr/bin/tsplugin_scrambler.so"
* Debug: /usr/bin/tsplugin_scrambler.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/bin/scrambler.so"
* Debug: /usr/bin/scrambler.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/tsplugin_scrambler.so"
* Debug: /usr/lib64/tsduck/tsplugin_scrambler.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/scrambler.so"
* Debug: /usr/lib64/tsduck/scrambler.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsplugin_scrambler.so"
* Debug: /usr/lib64/tsplugin_scrambler.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/scrambler.so"
* Debug: /usr/lib64/scrambler.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib/tsduck/tsplugin_scrambler.so"
* Debug: registering processor plugin "scrambler", status: ok
* Debug: scrambler: ====> tsp -P scrambler --aes-cbc --cw-file cw --cp-duration 10 --pid 0x100
* Debug: loading names file /usr/share/tsduck/tsduck.dtv.names
* Debug: scrambler: switching scrambling type from DVB-CSA2 to AES-CBC with externally-defined IV (TSDuck-specific)
* scrambler: loaded 6 control words
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: file: detected TS file format TS
* Debug: file: initial buffer load: 44,620 packets, 8,388,560 bytes
* file: initial input bitrate is 2,128,041 b/s
* file: creating file scrambled.ts
* Debug: scrambler: packet processing thread started
* Debug: file: output thread started
* Debug: control commands: found 1 packet processor plugins
* Debug: file: input thread started
* Debug: scrambler: using scrambling key: 02 66 83 9A FC DD A1 8A CD C6 E9 22 8D 34 33 35
* Debug: scrambler: starting using CW 02 66 83 9A FC DD A1 8A CD C6 E9 22 8D 34 33 35 (even)
* Debug: file: stopping the input plugin
* Debug: file: input thread terminated after 82,811 packets
* Debug: scrambler: packet processing thread terminated after 82,811 packets, 82,811 passed, 0 dropped, 0 nullified
* Debug: scrambler: stopping the plugin
* Debug: scrambler: scrambled 49,333 packets in 1 PID's
* Debug: file: stopping the output plugin
* Debug: file: output thread terminated after 82,811 packets (82,811 output)
* Debug: Report logging thread terminated
```

As the report tells, only the first control word was used : a single "even" period seems to have been created for the entire stream. It is confirmed if I add the options `--output-cw-file outcw`: only the first control word appears.

I can check the scrambling using `tsanalyse`:

```
$ tsanalyze scrambled.ts
...
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
...
|  Broadcast time: ................................... 59 sec (0 min 59 sec)  |
...
|  PIDS ANALYSIS REPORT                                                       |
...
|  PID: 0x0100 (256)                                            MPEG-2 Video  |
|  Service: 0x0001 (1) Prog#1                                                 |
|-----------------------------------------------------------------------------|
|  Single Service PID        Transport:                Discontinuities:       |
|  Bitrate: . 1,237,003 b/s  Packets: ........ 49,333  Expected: ......... 0  |
|  Access: ...... Scrambled  Adapt.F.: ........ 2,382  Unexpect: ......... 0  |
|  Crypto-Per: .... Unknown  Duplicated: .......... 0  PES:                   |
|  Inv.scramb.: ......... 0  TSrate: .. 2,076,449 b/s  Packets: ...... 1,800  |
|                                                      Inv.Start: ........ 0  |
|  Clock values range:                                                        |
|  PCR: ............... 600                                                   |
|  from ........ 18,900,000                                                   |
|  to ....... 1,636,200,000                                                   |
|  Leaps: ............... 0                                                   |

```

The PID is in fact scrambled, but the tool fails to tell the duration of the crypto-period. Sure enough, if I take a look using `dvbsnoop`, it is clear there is only "even" in there:

```
$ dvbsnoop -s ts -if scrambled.ts -tssubdecode -ph 2 0x100|grep 'transport_scrambling_control:'|sort|uniq -c
  49333 transport_scrambling_control: 2 (0x02)  [= TS packet scrambled with Even Key]
```

* **Is it normal for the `scrambler` plugin to only create one crypto-period, regardless of the `--cp-duration` option ?**
* Is the first crypto-period it creates always "even" ? (ie. I don't see any mention of this in the documentation, I had to check manually through testing)
* would it be possible to request to start with "odd" instead ? (ie. through the command line)

lelegard, 2024-12-28:
Hi @zappynou,

This is an unusual use case, so there is nothing really "normal" or not.

As a short answer, cryto-periods are applied when the scrambling is performed on a service, not a selected list of PID's. In your example, if you replace `--pid 0x100` by the name or id of a service, or simply `-` for the first or only service in the TS, then crypto-periods are applied.

In the real world, crypto-periods are linked to ECM's because ECM's are the only practical way to transmit CW's to the receiver. Specifying a list of CW's in a file is only a lab use case. ECM's are transported in a PID which is signaled in the PMT of the service. This is why, in the real world, there is no practical way to apply crypto-periods outside a service, because we need a service to tell the receiver where to find ECM's and CW's.

In theory, it could be possible to separately address the use case of multiple CW's on a set of PID's, outside any service. Since this is an unexpected use case, the code is not written this way. Currently, the crypto-period and ECM scheduling is initialized when the PMT of the service is found. This is done this way because initializing the ECM scheduling involves a modification of the PMT, adding CA descriptors.

Modifying this is possible but it is risky and requires extensive testing to make sure we don't break anything. Does it worth it? Is your use case just a test or a real use case with practical applications?

> Is the first crypto-period it creates always "even" ? (ie. I don't see any mention of this in the documentation, I had to check manually through testing)
> would it be possible to request to start with "odd" instead ? (ie. through the command line)

In the general case, the scrambler does not choose the crypto-period parity. It is imposed by the ECMG. In the unusual case of a fixed list of CW's without ECMG, the first parity is simply zero, meaning even.

The parity value, even or odd, is a reference to a CW inside the ECM. In the absence of ECM, the parity value is meaningless in itself because there is nothing to reference. Only the change of parity value is meaningful because it is a trigger to change the CW.

Do you have a practical use case to specify to start with an odd parity?

lelegard, 2024-12-28:
Additionally, as you mentioned, the documentation was not clear enough and I just updated it in commit 9a1d98e3af539cc11596f65885ed927c20981973

zappynou, 2024-12-29:
Thank you for your answer.

> Modifying this is possible but it is risky and requires extensive testing to make sure we don't break anything. Does it worth it? Is your use case just a test or a real use case with practical applications?

It is a lab test, my goal is to scramble each elementary stream of a program individually in order to test the limits of hardware/software. My tests so far were expecting the CW to arrive through another way than the ECM (ie. hard coded in the binary, entered by the tester, etc), but with what you explained, I have no problem to switch the tests to ECMs for simplicity.

> In your example, if you replace --pid 0x100 by the name or id of a service, or simply - for the first or only service in the TS, then crypto-periods are applied.

I figured the `--component-level` option from the `scrambler` plugin only specifies _where_ the CA_descriptor is inserted (into ES_info instead of program_info when the option is set) but does not tell if the plugin generates CWs per component which is why I went for the `--pid` option. _Is the `scrambler` plugin able to target individual ES specifically if I specify the "service" ?_

I have tested the tsecmg tool along with the `--ecmg` options of the `scrambler` plugin in order to perform the scrambling operation, it was working quite well but still it didn't seem to allow per-component scrambling.

> Do you have a practical use case to specify to start with an odd parity?

The practical use was only in order to ugly-work-around the issue of having only Even parity in the resulting stream, but you may ignore this idea now.

> Additionally, as you mentioned, the documentation was not clear enough and I just updated it in commit https://github.com/tsduck/tsduck/commit/9a1d98e3af539cc11596f65885ed927c20981973

Thank you, it is very clear now.

*  _Is there a way to per-component scramble using the service to scramble ?_

Thank you for your help.

lelegard, 2024-12-29:
If you scramble in two passes (two consecutive `scrambler` plugins in the same command), both with `--component-level`, one with `--no-audio` and the other one with `--no-video`, you should be able to scramble audio and video PID's with distinct streams of CW. However, if you have several audio streams, they will be scrambled with the same CW.

It could be possible to enhance the `scrambler` plugin to scramble only one PID in a service, as long as it is referenced in the specified service.

zappynou, 2024-12-30:
I expect to have several audio streams per program indeed, so I'll go for the plugin enhancement.

So far, I have added an `--only-pid` option to the plugin which filters out the other PIDs if it is provided, or does the "same as before" if it is not. The commit is ready, the pull request will be opened as soon as I manage to get a new test in tsduck-test with that new option.

lelegard, 2024-12-30:
Good. Waiting for your PR then.

lelegard, 2024-12-31:
PR #1568 has been merged.
Thank you for your contribution.

closed 2024-12-31 by lelegard