tsduck/tsduck #1567: [BUG] Makefile.inc broken under Linux

fcartegnie, 2024-12-30:
#### Bug description:

Can not build due to bash evaluation errors

#### Errors and logs:
~~~
bash: -c: line 1: unexpected EOF while looking for matching `"'
bash: line 1: /git-hook-update.py: No such file or directory
make: *** [Makefile:106: git-hooks] Error 127

~~~

#### Environment:
 - OS: Fedora 40

#### Additional information:
```
v3.38-3822 OK
v3.39-3956 KO

6a571d95c284e5ed33586f70df5830aaba44e432 is the first bad commit
commit 6a571d95c284e5ed33586f70df5830aaba44e432
Author: Thierry Lelegard <thierry@lelegard.fr>
Date:   Tue Oct 15 22:13:46 2024 +0200

```

lelegard, 2024-12-30:
Hi @fcartegnie 

As you may imagine, if this was a general problem, hundredths of continuous integration workflows would have failed, as well as all my compilations and all users who rebuild TSDuck on their own.

There must be something - or something missing - in your system which interferes with the build.

Did you try with the latest state of the repo (version 3.40-something)?

The error message is `bash: line 1: /git-hook-update.py: No such file or directory` when the command in the makefile is `$(PYTHON) $(SCRIPTSDIR)/git-hook-update.py`. So, it seems that neither `PYTHON` nor `SCRIPTSDIR` were defined. The error is probably in the way `make-config.sh` is called from `Makefile.inc`. Since environment variables are passed on the command line, I would bet that one of your environment variables is the cause of the problem. Please start a shell with a clean environment, using `env -i bash` for instance, and rebuild. If the problem persists, unset the remaining environment variables, just keep `HOME` and `PATH` and retry.

Once we will have identified the type of variables which creates trouble, we may be able to make the build resistant to that kind of "rogue" variable.

fcartegnie, 2024-12-31:
> Once we will have identified the type of variables which creates trouble, we may be able to make the build resistant to that kind of "rogue" variable.

As far as I've been able to go, the issue was in the eval() call
https://github.com/tsduck/tsduck/blob/master/Makefile.inc#L61

lelegard, 2024-12-31:
> As far as I've been able to go, the issue was in the eval() call

Yes, that is exactly what I meant in my previous comment. The problem must be in the expansion of one of your environment variables. Did you try to build in a clean / empty environment as I suggested?