tsduck/tsduck #1570: [Q] Big Thank you and ATSC 3.0 queastion.

8bitsten, 2025-01-02:
Hey everyone,

I’ve been using TSDuck to modify PMT tables and inject the AIT table for HbbTV app testing. After that, I broadcast the modified files in a loop with StreamXpress and the DekTec modulator.

TSDuck is an awesome tool, and I’m really grateful to Thierry Lelegard for creating it – saying **thank you** is actually the main reason I’m posting this message. 😊 I just made also a small donation to support his work.

I have a question about ATSC 3.0/NextGen and A/337 app signaling (I’m completely new to this). I recently got a couple of US ATSC 3.0 boxes that claim to support it, and I noticed that the DTU-315 seems to support ATSC 3.0. I want to dig into this, but since it uses MPEG-DASH, I’m guessing TSDuck isn’t the right tool for it. If anyone here has experience with ATSC 3.0 and app signaling, I’d really appreciate any insights you could share! It’ll save me a lot of time. 🙏

Thanks again for this amazing tool, and I really hope it keeps evolving!

Cheers,
Petr | [www.tvgam.es](http://www.tvgam.es/)

lelegard, 2025-01-02:
Hi @8bitsten,

Thank you for your support, this is much appreciated.

As you guessed, TSDuck is a transport stream tool, it deals with all sorts of TS processing. Since ATSC 3.0 is not based on TS transport, TSDuck cannot manipulate it.

I hope that other users here have some ATSC 3.0 experience and may help.

8bitsten, 2025-01-04:
After some initial research, it appears that the process of getting input for ATSC 3.0 broadcast is more complicated. 

In layman terms: For the StreamXpress player, which is used by DekTek modulators, I cannot use a file type like ****MPEG-2 TS**** as an input like in DVB. Instead, I need to use a Wireshark trace (because I guess all ATSC 3.0 necessary broadcasting information is transmitted via multicast from different ports) of the **STLTP stream** (kind of similar to MPEG-2 TS in DVB) that aggregates all the **UDP/ROUTE** streams for individual channels into a single multicast.

In addition to the MPEG/DASH data, there is also another UDP-based data in the so called **SLT (Service List Table)**, which carries signaling information called **LLS (Low-Level Signaling)**. This is likely similar to the **PAT** in MPEG-2 TS. The **SLT** points to the **Service Layer Signaling**, which is transmitted via **ROUTE**, and I perceive it as analogous to the **PMT** in **MPEG-2 TS**.

While **HbbTV** signaling was delivered via the **AIT table** in **DVB**, in **ATSC 3.0**, the signaling for interactive apps as described in **A/331** ([Link to A/331](https://www.atsc.org/wp-content/uploads/2024/04/A331-2024-04-Signaling-Delivery-Sync-FEC.pdf/)) using **HELD** metadata referenced from **ESG**, which I see as and umbrella for various metadata (captions, EPG, etc.). **HELD** for interactive app signaling carries similar data to what is used in HbbTV's MPEG-2 TS **AIT** (e.g., app name, app URL, launch mechanism, etc.). 

Unfortunately, I don't have any trace pcap files to try or modify. An alternative could, of course, be some form of "ATSC Caster", which I do not have access as well.

Also I guess I misunderstand some concepts as there is much more than a 2-hour search can reveal.

If anyone has some tips let me know please.

jensk-dk, 2025-05-14:
Hey @8bitsten 
I don't know if it's still relevant, but Dektec has STLTP and other ATSC3 files on their ftp server. Mail their support for login.

Wireshark also has some ATSC3 pcap examples, but they're in ALP packet format and not STLTP:
https://wiki.wireshark.org/samplecaptures#atsc3-protocols