tsduck/tsduck #1571: [Q] Help need to create a EIT for each channel (Who has the correct way to structure the EIT?)

SonnyWalkman, 2025-01-03:
Howdy all broadcaster  TS guru's.

I'm developing a basic EPG scheduler in GO and are needing clarification on how to structure my EIT.xml template.

I require a EPG for each (x) channels and basically iterate in a for loop the meta data from a SQLite database. 
I also require how to add the <long_event_descriptor></long_event_descriptor> to each event.

What other metadata do I need to consider ?  version increments etc?

EIT.xml continually updated by eitinject plugin..  

Anyone out there has more than likely done what I'm doing. Rather get the right way than having to persevere in silence.

```
const eitTemplate = `<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
{{range .Channels}}
<!-- Eit channel {{.ServiceID}} -->
<EIT actual="true" current="true" last_table_id="0x4E" original_network_id="0x20FA" service_id="{{.ServiceID}}" transport_stream_id="0x0001" type="pf" version="14">
    {{range .Events}}
    <event event_id="0x{{printf "%04x" .EventID}}" start_time="{{.StartTime.Format "2006-01-02 15:04:05"}}" duration="{{.Duration}}" running_status="running" CA_mode="false">
        <short_event_descriptor language_code="eng">
            <event_name>{{.Description}}</event_name>
            <text>Rating: {{.Rating}}</text>
        </short_event_descriptor>
    </event>
    {{end}}
</EIT>
{{end}}
</tsduck>`

```

lelegard, 2025-01-03:
Hi @SonnyWalkman 

The first point is that the plugin `eitinject` loads events and then dynamically generates the EIT's. The input EIT's, the one you are talking about, are just containers for events. So, you don't care about the EIT structure, just care about the events inside it. When the XML files are loaded, the plugin extracts all events from the input EIT's, stores them in an internal in-memory database and drops the input EIT containers. Output EIT's will be regenerated later with the "right" events.

So, the only question that matters is how to build the events, not the EIT's.

Then, there is no "right" way to create events. This is an editorial matter which is decided with the TV operator (or yourself if you are acting as such in a private network).

This is also driven by the capabilities of the receivers, what they display, what kind of EIT descriptor they manage, etc.

The standards are voluntarily vague about what _should_ be set in an event to keep some editorial freedom. The standard just mention what can technically be set in an event. See ETSI EN 300 468 section 6.1 for the list of descriptors which are allowed in an EIT (in practice in the events). ETSI TS 101 211 is only technical, it is the reference for the design of the plugin `eitinject` but it won't help you much about the content of the events.

Specific information such as rating have specialized descriptors such as the `parental_rating_descriptor`. However, the receiver must handle it correctly: 1) technically to manage access to adult (or over N years) contents using a PIN code, 2) editorially to present the information in the EPG UI.

The `extended_event_descriptor` may contain raw information or structured information using item lists ("director", "actor", etc.) Again, this is a joint decision between the TV operator, the receiver provider (or regional specification of it), the program information editor. No "right" way, just "your" way.

SonnyWalkman, 2025-01-05:
Hello @lelegard,
Thank you for your prompt feedback.   I apologise for asking about EIT. I'll follow your comments. 
After further investigations I have found a few OP documents from Free TV Australia which sheds light on what is required here in Australia as far as EIT. EPG.

These documents maybe of some help to others.

[OP-44-Implementation-of-DVB-EIT-p-f-Issue-6-December-2014.pdf](https://github.com/user-attachments/files/18312153/OP-44-Implementation-of-DVB-EIT-p-f-Issue-6-December-2014.pdf)
[OP-58-Implementation-Guide-for-DVB-EIT-Schedule-Information-Issue-3-December-2014.pdf](https://github.com/user-attachments/files/18312156/OP-58-Implementation-Guide-for-DVB-EIT-Schedule-Information-Issue-3-December-2014.pdf)
[OP-72-Implementation-of-Content-Reference-IDs-by-Australian-Broadcasters-Issue-1-December-2014.pdf](https://github.com/user-attachments/files/18312157/OP-72-Implementation-of-Content-Reference-IDs-by-Australian-Broadcasters-Issue-1-December-2014.pdf)

lelegard, 2025-01-06:
That is exactly the kind of local operator / regional regulator documents which provide the guidelines you are looking for.

closed 2025-01-23 by SonnyWalkman