tsduck/tsduck #1573: [BUG] C++17 / C++20 cleanup: old workarounds are still in place

barsnick, 2025-01-08:
#### Bug description:
A lot of obsolete workarounds for buggy GCC versions are still in place.

#### How to reproduce:
`git grep -Fn "old gcc"`
and possibly others.

#### Expected behavior:
No compilation workarounds in the code base. Code still complies cleanly.

#### Errors and logs:

#### Environment:
 - OS: any
 - OS version: any
 - TSDuck full version: branch `develop` as of https://github.com/tsduck/tsduck/commit/6fb14e798695aacbcd071482381f108b49e7b774
 - Installation type: none

#### Additional information:
A lot of workarounds are introduced to fix compilation with older compilers, notably those default compilers on some relevant platforms such as RedHat (7/8).

E.g. I was involved in this one: https://github.com/tsduck/tsduck/issues/1285 which resulted in https://github.com/tsduck/tsduck/commit/8831aea69ac4f698bc0a07cb89bfdca37022d74b.

Since only C++20 is now supported in the code base (or C++17 in the last release 3.39), and the code base is being massively cleaned up, I'm sure it make sense to drop these workarounds. This should be an "easy" fix.

BTW, I appreciate the good documentation on the build process and how to get hold of proper compilers.

lelegard, 2025-01-08:
Hi @barsnick 

You are right, that makes sense. The idea is to always simplify the TSDuck source code to reduce the risks. I will do the cleanup.

I would hardly qualify this as a "bug", but certainly an enhancement.

lelegard, 2025-01-10:
@barsnick, I have cleanup up the old workarounds you mentioned. Maybe some others are still around. Do not hesitate to report them if you find some. Thank you for your watchful code review.

closed 2025-06-15 by github-actions[bot]