tsduck/tsduck #1578: [Q] Inserting SCTE-35 splices to TS file

tab1293, 2025-01-17:
I am attempting to insert SCTE-35 splices in to a TS file using `spliceinject`. I would like to generate a file that can be used for testing `splicemonitor` and my media pipeline's downstream playback synchronization. Ideally I could create a file that has many SCTE-35 splices that occur on a frequent basis, say every 30s.  Here is an example of how I am trying to accomplish this:

Create a minute long video:
```gen.sh
ffmpeg -re -f lavfi -i testsrc=size=1280x720:rate=30 -f lavfi -i sine=frequency=1000:duration=60 -t 60 \
    -c:v libx264 -preset veryfast -b:v 1500k -force_key_frames 'expr:gte(t,n_forced*3)' -c:a aac -b:a 128k -f mpegts gen.ts
```
Define a manifest that contains a splice insert at 30 seconds.
```scte.xml
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
    <splice_information_table>
        <splice_insert splice_event_id="100" unique_program_id="1"
            out_of_network="true" pts_time="2700000" splice_immediate="false" />
    </splice_information_table>
</tsduck>
```
Use `spliceinject` to inject the splices and output a new TS file
```spliceinject.sh
tsp -v --add-input-stuffing 1/10 \
    -I file gen.ts \
    -P pmt --service 1 --add-programinfo-id 0x43554549 --add-pid 600/0x86 \
    -P spliceinject --service 1 --files scte.xml --wait-first-batch \
    -O file gen_scte.ts
```
Resulting tables for the file look like:
```

* PAT, TID 0 (0x00), PID 0 (0x0000)
  Version: 0, sections: 1, total size: 16 bytes
  - Section 0:
    TS id:       1 (0x0001)
    Program:     1 (0x0001)  PID: 4096 (0x1000)

* PMT, TID 2 (0x02), PID 4096 (0x1000)
  Version: 0, sections: 1, total size: 37 bytes
  - Section 0:
    Program: 1 (0x0001), PCR PID: 256 (0x0100)
    Program information:
    - Descriptor 0: Registration (0x05, 5), 4 bytes
      Format identifier: 0x43554549 ("CUEI")
    Elementary stream: type 0x1B (AVC video), PID: 256 (0x0100)
    Elementary stream: type 0x0F (MPEG-2 AAC Audio), PID: 257 (0x0101)
    Elementary stream: type 0x86 (SCTE 35 Splice Info), PID: 600 (0x0258)

* SDT Actual, TID 66 (0x42), PID 17 (0x0011)
  Version: 0, sections: 1, total size: 40 bytes
  - Section 0:
    Transport Stream Id: 1 (0x0001)
    Original Network Id: 65281 (0xFF01)
    Service Id: 1 (0x0001), EITs: no, EITp/f: no, CA mode: free
    Running status: running
    - Descriptor 0: Service (0x48, 72), 18 bytes
      Service type: 0x01 (Digital television service)
      Service: "Service01", Provider: "FFmpeg"

* SCTE 35 Splice Information, TID 252 (0xFC), PID 600 (0x0258)
  Short section, total size: 35 bytes
  - Section 0:
    Protocol version: 0
    Encryption: none
    PTS adjustment: 0x000000000
    CW index: 0x00 (0), tier: 0xFFF (4095)
    Command type: 0x05 (SpliceInsert), size: 15 bytes
    Splice event id: 0x00000064 (100), cancel: 0
    Out of network: yes, program splice: yes, duration set: no, immediate: no
    Time PTS: 0x0002932E0 (2,700,000, 30,000 ms)
    Unique program id: 0x0001 (1), avail: 0x00 (0), avails expected: 0
    CRC32: 0xFBE630C9 (OK)

* SCTE 35 Splice Information, TID 252 (0xFC), PID 600 (0x0258)
  Short section, total size: 35 bytes
  - Section 0:
    Protocol version: 0
    Encryption: none
    PTS adjustment: 0x000000000
    CW index: 0x00 (0), tier: 0xFFF (4095)
    Command type: 0x05 (SpliceInsert), size: 15 bytes
    Splice event id: 0x00000064 (100), cancel: 0
    Out of network: yes, program splice: yes, duration set: no, immediate: no
    Time PTS: 0x0002932E0 (2,700,000, 30,000 ms)
    Unique program id: 0x0001 (1), avail: 0x00 (0), avails expected: 0
    CRC32: 0xFBE630C9 (OK)
```
But when I play the file with `splicemonitor` enabled, the SCTE splices seem to be loaded immediately at the start of playback. The `splicemonitor` plugin is not logging anything 30s in to playback, only right at the start, as seen in the logs below:
```
tsp -v -I file gen_scte.ts  -P splicemonitor -O play
* file: initial input bitrate is 1,132,039 b/s
* play: using media player command: "/Applications/VLC.app/Contents/MacOS/VLC" - --play-and-exit
* splicemonitor: starting monitoring splice PID 0x0258 (600)
* splicemonitor: splice PID 0x0258 (600), event 0x00000064 (100) out, occurrence #1, time to event: 1,860 ms
* splicemonitor: splice PID 0x0258 (600), event 0x00000064 (100) out, occurrence #2, time to event: 1,130 ms
* splicemonitor: splice PID 0x0258 (600), event 0x00000064 (100) out, occurred, actual pre-roll time: 1,980 ms
VLC media player 3.0.21 Vetinari (revision 3.0.21-0-gdd8bfdbabe)
[00000001461055c0] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
[0000000134626ff0] videotoolbox decoder: Using Video Toolbox to decode 'h264'
[0000000144608e60] mpeg4audio packetizer: AAC channels: 1 samplerate: 44100 
```
My questions are:
1. Why are there 2 SCTE splices reported in the tables output when I only have one defined in the XML?
2. Why are the splices reported at the start of playback rather than 30s in to the video? 

I would appreciate any insight in to how I can get a proper test file generated! Thanks!

lelegard, 2025-01-17:
Hi @tab1293

> Why are there 2 SCTE splices reported in the tables output when I only have one defined in the XML?

Splice commands such as splice_insert are always repeated several times before the event. This is the principle of broadcast. You can always miss a packet. Therefore, all data are repeated, PAT, PMT, CAS ECM and EMM, splice commands, everything.

You need to separate the SCTE-35 commands and the way there are broadcasted. You define one single command but this single command is broadcasted several times.

The way a command is broadcasted, starting how much time before the event, how many times it is repeated, is driven by command-line options in the `spliceinject` plugin.

> Why are the splices reported at the start of playback rather than 30s in to the video?

I think the file is correct, the commands are really at 30s (minus pre-roll time) in the file, but your perception of "time" has been deceived. More exactly, the `splicemonitor` plugin and VLC rendering do not use the same wall-clock time.

First, let's assume that the file begins with PTS starting at zero because it was generated from scratch by ffmpeg. In that case, PTS 2,700,000 is indeed 30s after the beginning of the file. The displayed bitrate is 1,132,039 b/s, meaning that 30 seconds of contents use 4.2 MB (bytes, not bits).

Now, please read the reference documentation of `tsp` in the TSDuck user's guide.

Then, consider your command:
~~~
tsp -I file gen_scte.ts -P splicemonitor -O play
~~~

- There is a global buffer where all packets are stored and all threads work on it. More details on a [user perspective](https://tsduck.io/download/docs/tsduck.html#tsp-reference) and on a [developer's perspective](https://tsduck.io/download/docs/tsduck-dev.html#pdevdesign). The default buffer size is 16 MB, modifiable through command-line options.
- Each plugin executes in a separate thread. Each thread progresses on the global buffer until it is blocked by the previous thread. This is a global flow control which ensures that all packets are processed by all threads, in the order of the command line.
- Typically, some threads are faster than others.
  - The `file` input plugin fills the buffer at the speed of the disk, way much faster then receiving a live stream at the playout speed. The input plugin has to wait when the global buffer is full, waiting for the output plugin to remove packets and free space.
  - The `splicemonitor` plugin just reads and analyzes packets. It runs at the CPU speed. When all packets in the buffer are processed, it waits for the input plugin to provide more.
  - The `play` output plugins starts a VLC process and sends the packets to it through a pipe. VLC has some input buffer but it essentially processes packets at the playout speed, very slowly compared to the disk reading speed.
  - After an initial phase where all buffers are filled, the `tsp` global buffer, the pipe kernel buffer, the VLC input buffer, a global flow control applies and the output plugin slowly frees packets from the global buffer, at playout speed.

Now, in practice, the consequences are:
- The input plugin reads the file at disk speed and fills the buffer when space is available. The default buffer size is 16 MB, meaning 2 minutes of content at your bitrate. The first 16 MB are read at disk speed to fill the buffer. Then the input plugin is slowed down to the playout speed, when packets are freed in the buffer.
- The `splicemonitor` plugin processes packets as soon as they are available in the buffer. This means that the first 16 MB, 2 minutes of content, are immediately analyzed at the beginning. Then, the analysis proceeds at playout speed, when packets are provided by the input plugin.

This is why you immediately see the messages.

Note that the messages say "occurrence #1, time to event: 1,860 ms" and "occurrence #2, time to event: 1,130 ms". This means that the plugin has compared the PTS in the video packets with the value in the splice command and computed the time difference. Therefore, the commands are at the right place in the file.

tab1293, 2025-01-17:
Thanks @lelegard for your response!

I have a few of follow up questions...

Since `splicemonitor` processes packets as soon as they are available in the buffer, that means that splice reporting for files will never be synchronized with the pipeline's output, is that correct? So for example if I'm using the `-O ip` output plugin, `splicemonitor` will likely print splice information before PTS 2700000 is output over the network?

If that's the case, how can splices with `immediate: yes` ever be synchronized? Since these splices do not contain any PTS information and `splicemonitor` does not report the splice in sync with the output plugin, how can these slices ever be associated with the correct timing of when ad insertions should start/end?

In the live scenario (ie reading from a SRT listener), will `splicemonitor` reporting of splice insertions more or less be aligned with the output plugin? So as soon as `splicemonitor` outputs a splice insert with `immediate: yes`, does that mean that the splice should be honored downstream immediately?

At a higher level I am just trying to understand how `splicemonitor` can be used in order to synchronize splice insertion downstream with what an end user is seeing during playback.

lelegard, 2025-01-17:
The important point is not the output of the plugins. The important point is the placement of the splice commands in the stream.

The plugin `splicemonitor` reports the message at the time it sees them and the ad-inserter/splicer/transrater/whatever processes the splice commands at the time it sees them in the stream. There is no synchronization between the two systems. And there is no need or reason for it.

The ad-inserter must be an independent system. The stream flows through the ad-inserter. When the ad-inserter sees a splice command for the first time, it monitors the A/V PID's for the target PTS and starts replacing these PID's when the event PTS is reached. If the command is immediate, the ad-inserter starts replacing the PID's immediately.

If you wanted to use the output of the plugin `splicemonitor`, this is of course definitely not an appropriate approach.

tab1293, 2025-02-13:
Okay thanks for you follow up @lelegard . Closing this as we are opting for a different approach to trigger ad insertions outside of `splicemonitor`

closed 2025-02-13 by tab1293