tsduck/tsduck #1579: [BUG]  tsswitch duplicates receiving data to output

DvarfInkviz, 2025-02-08:
Hi, Thierry.

Thanks for so usefull software.

We find a issue using any utility from TSDuck package, that works with udp multicast as a source. The issue is 100% reproduced under Windows 10 OS. Version of TSDuck does not matter.

When tsswitch is not running, both multicasts are stable and decodable. MPTS bitrate is about 27 Mbit/s.

[image]

When tsswitch is started, it duplicate receiving data to output with the same multicast address. We can see in tsreader, that both source multicasts grow up twice (look at Last sec value) and become undecodable with lots of CC errors.

[image]

Is it possible to solve this problem by tsswich|tsp settings, or it’s a bug?

Thx
#### Bug description:
tsswitch duplicates receiving data to output

#### How to reproduce:
1. Add udp multicast as a source.
2. run tsswitch
3. When tsswitch is started, it duplicate receiving data to output with the same multicast address

#### Expected behavior:
Stable multicasts at output without CC errors

#### Errors and logs:
Continuity Errors: 215097
[image]
~~~
[be sure to surround verbatim log lines with "~~~" lines]
~~~

#### Environment:
 - OS: Windows
 - OS version: 10
 - TSDuck full version: does not matter
 - Installation type:

lelegard, 2025-02-09:
Hi @DvarfInkviz 

Thanks for your appreciation of TSDuck.

It is a bit difficult to understand your problem. Could you be more specific? Specifically, please post the exact sequence of events:
- which precise command you run and where (on the Windows system or on a peer) 
- what you observe and where (on the Windows system or on a remote one).

Apart from that, please note that multicast is very special. Receiving and sending multicast behaves differently on different systems. Some options even have a different interpretation depending on the OS (see option `--disable-multicast-loop` for instance). As a rule of thumb (and from experience), I would say that Linux works well with multicast while Windows exhibits strange behaviours when you go beyond the basic use case of either receiving one stream or sending one stream.

If I remember correctly, the various Windows versions had distinct behaviours. There is at least one configuration (can't remember which one) where, when 2 clients on the same Windows systems listening to the same multicast group received the traffic twice (duplicated) and one configuration where the traffic was "shared" (partial reception on each receiver socket). This was some old Windows configuration. Maybe before Windows 10, even though Windows 10 is old now.

Typically, the vast majority of streaming servers use Linux. Windows is mostly a personal client where the typical use case is receiving one multicast stream. 

So, the first test would be to reproduce the same experiment on a Linux system. If it works as expected, I would blame Windows. You may even try on the latest Windows 11 (24H2) to verify if it is similar to Windows 10.

lelegard, 2025-02-09:
Additional questions:

- Do you have the same behaviour with `tsp` instead of `tsswitch`?
- Do you measure the input multicast on the same machine as `tsswitch` or `tsp`? In that case, you may fall into the case I mentioned where you have two clients on same Windows machine which listen on the same multicast group (the measurement tool and `tsswitch`) and they receive a duplicated traffic.

DvarfInkviz, 2025-02-17:
Thank you for your response to our issue. Unfortunately, there is currently no way to repeat the experiment on Linux.
To illustrate the issue, we recorded a video:

[tcp_floods_into_network_HQ.mp4](https://dropmefiles.com/sRWn6)

lelegard, 2025-02-17:
Does your system have several network interfaces ? And does the multicast traffic arrive on both interfaces ?

To avoid wondering on which interface the multicast traffic arrives, TSDuck joins the multicast group on all interfaces by default. Packets will then be received on their incoming interface, whichever it is. However, if the traffic is multicast on two networks, or if two interfaces are connected to the same network, the same packets will be received on two interfaces and they look duplicated. 

If this is your configuration, use option `--local-address` to specify to join the group on that interface only. Choose the same interface as used by the other receiving applications.

lelegard, 2025-02-26:
@DvarfInkviz, I cannot reproduce your demo. I run two ffplay on the same multicast input and, when I start `tsp -I ip` on the same address, nothing special occurs. I tried joining the multicast group from all interfaces (the default), one specific interface, or the "default multicast interface". All configurations work without interference with the two concurrent ffplay on the same system.

I use Windows 11 (24H2) because I haven't seen a Windows 10 for years.

So, either there is something special in your multicast configuration or this is one of the many issues about multicast on older versions of Windows. Windows has an awful reputation about multicast. The situation has improved with each version, it's getting better with each new version. So, I recommend to try on an up-to-date version of Windows 11.

Also use the latest TSDuck build from https://tsduck.io/download/prerelease/

cus, 2025-03-21:
I think this can happen under both linux and windows due to the fact that multicast binding to an interface or to a source address (source specific multicast) does not act as packet filter from the IP stack point of view.

So if you have two network interfaces receiving the same multicast at the same time (tsswitch might do this for example), then both receiving sockets will receive packets from both interfaces.

My understanding is that the only way to restrict a socket bound to a multicast address to an interface is either using the linux specific SO_BINDTODEVICE or doing the filtering from userspace using the IP_PKTINFO struct which also contains the receiving interface index.