tsduck/tsduck #1580: [FEAT] Binary patch in all sections

vidanio, 2025-02-10:
I want to patch an SPTS on the fly in the PID 18 (EIT) just the SID, TSID and ONID without having to read or compare with their previous value.
Using SID(SSSS), TSID(TTTT) and ONID(NNNN) mask: 000000SSSS000000TTTTNNNN

Example: 
Wanna write SID = 0x1111, TSID = 0x2222, ONID = 0x3333
Just executing:
tsp -I file spts.ts -P sections --pid 18 --keep --section-write 000000111100000022223333 --section-mask 000000FFFF000000FFFFFFFF -O file output.ts

A section writer would be great. 
Thanks for your great project !!

lelegard, 2025-02-10:
Hi @vidanio 

Use XLM a XML patch file to update selected fields on the fly. If you don't want to create an external XML file, you can even inline it in the command:

~~~
tsp .. -P sections --pid 18 --patch-xml '<?xml version="1.0" encoding="UTF-8"?><tsduck><EIT x-add-service_id="0x1111" x-add-transport_stream_id="0x2222" x-add-original_network_id="0x3333"/></tsduck>' ...
~~~

See https://tsduck.io/download/docs/tsduck.html#xml-patch for more details on patch files.

vidanio, 2025-02-11:
OK, that's great. Thanks !!!

closed 2025-02-11 by vidanio