tsduck/tsduck #1581: [BUG] Can't build on Debian 12

juliusmilan, 2025-02-10:
#### Bug description:
Build fails on Debian 12 with message as below.

#### How to reproduce:
1) Cloned current master, i.e.: 95c934a6b1981d59494283d7441e18b75c084635
2) env -i bash         # otherwise I got https://github.com/tsduck/tsduck/issues/1567
3) make -j4 default

#### Expected behavior:
To compile.

#### Errors and logs:
~~~
...
  [CXX] network/tsRIST.cpp
  [CXX] plugins/private/tsRISTPluginData.cpp
In file included from /usr/include/c++/12/string:40,
                 from /home/godi/ws/maindata/tsduck/src/libtscore/cpp/tsPlatform.h:715,
                 from /home/godi/ws/maindata/tsduck/src/libtscore/types/tsUChar.h:16,
                 from /home/godi/ws/maindata/tsduck/src/libtscore/types/tsUString.h:15,
                 from /home/godi/ws/maindata/tsduck/src/libtscore/app/tsNames.h:15,
                 from dtv/signalization/tsREGID.h:16,
                 from dtv/signalization/tsREGID.cpp:9:
In static member function ‘static constexpr std::char_traits<char16_t>::char_type* std::char_traits<char16_t>::copy(char_type*, const char_type*, std::size_t)’,
    inlined from ‘static constexpr std::char_traits<char16_t>::char_type* std::char_traits<char16_t>::copy(char_type*, const char_type*, std::size_t)’ at /usr/include/c++/12/bits/char_traits.h:817:7,
    inlined from ‘static constexpr void std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::_S_copy(_CharT*, const _CharT*, size_type) [with _CharT = char16_t; _Traits = std::char_traits<char16_t>; _Alloc = std::allocator<char16_t>]’ at /usr/include/c++/12/bits/basic_string.h:423:21,
    inlined from ‘static constexpr void std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::_S_copy(_CharT*, const _CharT*, size_type) [with _CharT = char16_t; _Traits = std::char_traits<char16_t>; _Alloc = std::allocator<char16_t>]’ at /usr/include/c++/12/bits/basic_string.h:418:7,
    inlined from ‘constexpr std::__cxx11::basic_string<_CharT, _Traits, _Allocator>& std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::_M_replace(size_type, size_type, const _CharT*, size_type) [with _CharT = char16_t; _Traits = std::char_traits<char16_t>; _Alloc = std::allocator<char16_t>]’ at /usr/include/c++/12/bits/basic_string.tcc:532:22,
    inlined from ‘constexpr std::__cxx11::basic_string<_CharT, _Traits, _Alloc>& std::__cxx11::basic_string<_CharT, _Traits, _Alloc>::assign(const _CharT*) [with _CharT = char16_t; _Traits = std::char_traits<char16_t>; _Alloc = std::allocator<char16_t>]’ at /usr/include/c++/12/bits/basic_string.h:1647:19,
    inlined from ‘ts::UString& ts::UString::operator=(const ts::UChar*)’ at /home/godi/ws/maindata/tsduck/src/libtscore/types/tsUString.h:2481:64,
    inlined from ‘ts::UString ts::REGIDName(REGID, NamesFlags)’ at dtv/signalization/tsREGID.cpp:25:12:
/usr/include/c++/12/bits/char_traits.h:826:34: error: ‘void* __builtin_memcpy(void*, const void*, long unsigned int)’ accessing 9223372036854775812 or more bytes at offsets [4, 9223372036854775806] and 2 may overlap up to 9223372036854775817 bytes at offset -5 [-Werror=restrict]
  826 |                 (__builtin_memcpy(__s1, __s2, __n * sizeof(char_type))));
      |                  ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [../../Makefile.inc:66: /home/godi/ws/maindata/tsduck/bin/release-x86_64-debian/objs-libtsduck/tsREGID.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:14: default] Error 2
make: *** [Makefile:18: default] Error 2
~~~

#### Environment:
 - OS: Linux
 - OS version: Debian 12
 - TSDuck full version: git commit 95c934a6b1981d59494283d7441e18b75c084635
 - Installation type: custom build from sources

#### Additional information:
g++ version: g++ (Debian 12.2.0-14) 12.2.0

Please let me know if any other piece of information is needed.

lelegard, 2025-02-10:
Hi @juliusmilan 

See this: https://tsduck.io/download/docs/tsduck-dev.html#cpprequnix

TSDuck now requires a C++20 compliant compiler. With gcc, the minimum version is gcc 13. All previous versions which were supposed to support C++20 are buggy (as you experience).

lelegard, 2025-02-10:
Debian 12 contains only obsolete compilers, gcc 12 and clang 14. There is not even any more recent alternate package for gcc. Even the "always late" Red Hat EL has alternate packages for gcc 14. Only Debian sid has decently recent compilers.

With Debian 12, there are alternate packages for more recent versions of clang. It just tried this and it works:

~~~
sudo apt install clang-tools-19
make -j10 CXX=clang++-19 CC=clang-19 GCC=clang-19 CPP=clang-cpp-19 test
~~~

juliusmilan, 2025-02-13:
Thank you for both: explanation and build commands, worked perfectly.

closed 2025-02-13 by juliusmilan