tsduck/tsduck #1583: [BUG] Segmentation fault when attempting to use SRT output plugin

tab1293, 2025-02-13:
#### Bug description:
We are seeing `Segmentation fault (core dumped)` errors when attempting to use the SRT output plugin. 

#### How to reproduce:
```
tsp -v -I srt --caller [ip-address]:7000 --transtype live --latency 3000 --statistics-interval 2000 --rcvbuf 18750000000 -P splicemonitor -a --json -o - -P analyze -P continuity -P regulate -O srt 0.0.0.0:4000
* regulate: minimum packet burst duration is 10000000 nanoseconds
* srt: SRT statistics:
  Total received: 11,633,560 bytes, 8,702 packets, lost: 0 packets, dropped: 0 packets
  Interval received: 11,633,560 bytes, 8,702 packets, lost: 0 packets, dropped: 0 packets
  Timestamp-based delivery delay, receive: 3000 ms, RTT: 0.259000 ms
* srt: unknown initial input bitrate
Segmentation fault (core dumped)
```
vs using the same command with the `drop` output plugin
```
tsp -v -I srt --caller 35.194.65.144:7000 --transtype live --latency 3000 --statistics-interval 2000 --rcvbuf 18750000000 -P splicemonitor -a --json -o - -P analyze -P continuity -P regulate -O drop
* regulate: minimum packet burst duration is 10000000 nanoseconds
* srt: SRT statistics:
  Total received: 11,633,372 bytes, 8,702 packets, lost: 0 packets, dropped: 0 packets
  Interval received: 11,633,372 bytes, 8,702 packets, lost: 0 packets, dropped: 0 packets
  Timestamp-based delivery delay, receive: 3000 ms, RTT: 0.226000 ms
* srt: unknown initial input bitrate
* regulate: unknown bitrate, cannot regulate.
* splicemonitor: starting monitoring splice PID 0x0190 (400)
* regulate: regulated at bitrate 30,000,092 b/s
* srt: SRT statistics:
  Total received: 19,388,644 bytes, 14,502 packets, lost: 0 packets, dropped: 0 packets
  Interval received: 7,755,272 bytes, 5,800 packets, lost: 0 packets, dropped: 0 packets
  Timestamp-based delivery delay, receive: 3000 ms, RTT: 0.229000 ms
  * srt: SRT statistics:
  Total received: 27,143,916 bytes, 20,302 packets, lost: 0 packets, dropped: 0 packets
  Interval received: 7,755,272 bytes, 5,800 packets, lost: 0 packets, dropped: 0 packets
  Timestamp-based delivery delay, receive: 3000 ms, RTT: 0.226000 ms
^C* tsp: user interrupt, terminating...
```

#### Expected behavior:
Expected to see tsduck start up a SRT listener rather than seg fault

#### Environment:
 - OS: Linux Docker container
 - OS version: ubuntu22.04
 - TSDuck full version: 3.39-3918
 - Installation type:  From source (see below for how tsduck was installed in Docker container)
 ```
FROM nvidia/cuda:12.2.2-runtime-ubuntu22.04 as srt-build
RUN apt-get update
RUN DEBIAN_FRONTEND=noninteractive TZ=Etc/UTC apt-get -y install git tcl pkg-config build-essential cmake gcc g++ make automake autoconf libssl-dev zlib1g-dev \
    libass-dev libfreetype6-dev libgnutls28-dev libmp3lame-dev libsdl2-dev libtool libva-dev libvdpau-dev libvorbis-dev libxcb1-dev libxcb-shm0-dev libxcb-xfixes0-dev libunistring-dev \
    meson ninja-build texinfo yasm nasm libx264-dev libx265-dev libnuma-dev libfdk-aac-dev libopus-dev libvpx-dev \
    bash curl wget sudo lsb-release

RUN git clone --depth 1 https://github.com/tsduck/tsduck.git /tsduck
WORKDIR /tsduck
RUN ./scripts/install-prerequisites.sh
RUN make -j10
RUN make install
....
```

#### Additional information:
I am able to run this same tsp command without issue when using my local MacOS machine where tsduck was installed via brew

lelegard, 2025-02-13:
Hi @tab1293 

Hum, hard to tell without reproducing this. This looks linked to the Docker environment (it normally works correctly on Ubuntu 22.04).

Do you have the same problem without the SRT statistics? (Remove the option `--statistics-interval`)

What is the host distro, and kernel version? A container uses the kernel of its host, so this is not a "real" Ubuntu 22.04.

lelegard, 2025-02-14:
Could you also run `tsp --version=all` in the container? That will give you the version of the SRT library. I think that Ubuntu 22 had a rather old version of it. Much older than the one you use on macOS, where Homebrew always bring the latest version (1.5.4 as of today).

If you can, could you try with a container based on Ubuntu 24.04? It should have a much more recent version of the SRT library.

lelegard, 2025-02-25:
@tab1293, could you try without SRT statistics, or on an Ubuntu 24 based container? And run `tsp --version=all` in the container? 

A segmentation fault is a serious issue and shall be identified.

closed 2025-02-28 by tab1293