tsduck/tsduck #1587: step by step installation on ubuntu 22.04

Sadombaj, 2025-03-03:
Can someone walk me through tsduck installation on ubuntu 22.04. i have tried instructions am getting online several times without success. is there a detailed step by step manual for ubuntu 22.04?

lelegard, 2025-03-03:
Hi @Sadombaj,

The current pre-built binaries are produced for Ubuntu 24. The last TSDuck binaries for Ubuntu 22 were produced for [version 3.33](https://github.com/tsduck/tsduck/releases/tag/v3.33-3139) in January 2023, two years ago. The current version is 3.40.

So, if you need the latest version, you need to rebuild from sources. This is [explained here](https://tsduck.io/download/docs/tsduck-dev.html#unixbuild). Basically, the full process is:

~~~
git clone https://github.com/tsduck/tsduck.git
cd tsduck
scripts/install-prerequisites.sh
make -j10 default docs-html
sudo make install
~~~

Please note the [requirements on the C++ compiler](https://tsduck.io/download/docs/tsduck-dev.html#cpprequnix). It must be C++20 compliant, meaning gcc 13 or clang 16 at least. I don't remember which versions were installed by default on Ubuntu 22 but I think that they were slightly older. In that case, check if there are alternative packages such as `gcc-13` or `clang-16` which can be installed.

If you can't get a recent compiler on Ubuntu 22, I think you can build TSDuck 3.39 with gcc 12. Just add this command before `make` to get the corresponding sources:
~~~
git checkout v3.39-3956
~~~

If you get specific errors or problems, please post of complete description of commands and errors.

lelegard, 2025-04-10:
> gcc-13, g++-13 now available as an Ubuntu 22.04 package,

Good news. It was a pity that a stable distro such as Ubuntu, even a 3 years old one, had zero C++20 compiler.

> (LLVM=1, clang14 causes novel issues).

Clang 14 has a buggy C++20 implementation. You need clang 16 or higher. The current version is clang 19.

> Avoid NOVATEK=1 as this will remove the libusb.so requirement, which will leave a blank line in the debian control file, causing dpkg-deb to fail.

Could you be more specific? I just tried and `make -j10 NOVATEK=1 installer` worked perfectly on Ubuntu. It is Ubuntu 24.10 but it shouldn't make a difference.

TobyBorland, 2025-04-10:
Ubuntu 22.04 LTS, `make V=1 default docs-html NOVATEK=1 NOPCSC=1`

generated a ~/tsduck/pkg/installers/tmp/DEBIAN/control missing a libusb reference

```
Package: tsduck
Source: tsduck
Version: 3.41-4220.ubuntu22
Maintainer: Thierry Lelegard <thierry@lelegard.fr>
Architecture: amd64
Section: free/video
Priority: optional
Depends: libc6,
 libssl3, windscribe,
 libsrt1.4-openssl,
 libedit2,
 
 libcurl4,
 libzstd1, zlib1g,
 libstdc++6
Conflicts: tsduck-dev (<= 2.28.2441)
Description: MPEG Transport Stream Toolkit
 TSDuck, the MPEG Transport Stream Toolkit, provides some simple utilities to
 process MPEG Transport Streams (TS), either as recorded files or live streams.
```

According to the dpkg-deb error, a blank line in the "Depends" will cause package creation failure. 
Testing ldd libtsduck.so from this failed build revealed no libusb, so I am guessing that one of the make flags drops USB support and the sed script generation leaves a blank line.
Compiled and installed without issue once I removed the flags. 
I also had that anaconda (base) environment problem, giving unusual libedit issues.

This is a commendable project, thanks for your effort.

lelegard, 2025-04-11:
Which command do you use to generate the .deb packages?

The command `make V=1 default docs-html NOVATEK=1 NOPCSC=1` builds TSDuck but it does _not_ build the installers.

Using command `make NOVATEK=1 NOPCSC=1 installer`, I see the following generated command:
~~~
sed -e 's/{{VERSION}}/3.41-4241.ubuntu24/g' \
    -e 's/{{ARCH}}/amd64/g' \
    -e 's/ libssl,/ libssl3t64,/' \
    -e 's/ zlib,/ libzstd1, zlib1g,/' \
    -e 's/ libsrt,/ libsrt1.5-openssl,/' \
    -e 's/ librist,/ librist4,/' \
    -e 's/ libedit,/ libedit2,/' \
    -e '/libusb/d' \
    -e '/libpcsc/d' \
    -e 's/ libcurl,/ libcurl4t64,/' \
    tsduck.control >[...]/pkg/installers/tmp/DEBIAN/control
~~~

And the resulting control file is:
~~~
Package: tsduck
Source: tsduck
Version: 3.41-4241.ubuntu24
Maintainer: Thierry Lelegard <thierry@lelegard.fr>
Architecture: amd64
Section: free/video
Priority: optional
Depends: libc6,
 libssl3t64,
 libsrt1.5-openssl,
 librist4,
 libedit2,
 libcurl4t64,
 libzstd1, zlib1g,
 libstdc++6
Conflicts: tsduck-dev (<= 2.28.2441)
Description: MPEG Transport Stream Toolkit
 TSDuck, the MPEG Transport Stream Toolkit, provides some simple utilities to
 process MPEG Transport Streams (TS), either as recorded files or live streams.
~~~

The package names can be slightly different on Ubuntu 24.10.

The input lines about libusb (requirement for VATEK) and libpcsc are correctly removed, leaving no empty line. There is no difference between 3.41-4220 and 3.41-4241 regarding the way of building the packages. So, your result is not normal.

Your line `libssl3, windscribe,` is not normal either. Where does `windscribe` come from?

TobyBorland, 2025-04-11:
Judging by my .bash-history, `make -C tsduck installer`
Windscribe is a VPN program, I see it has both libssl.so.3 and libcrypto.so.3 in a local library.

Unfortunately as I had multiple terminal windows open, I am uncertain of the exact sequence. 
From memory, ~/tsduck/pkg/deb/Makefile contains the sed script which threw an error, namely,

`	sed -e 's/{{VERSION}}/$(VERSION)$(DISTRO)/g' \
	    -e 's/{{ARCH}}/$(DEB_ARCH)/g' \
	    $(if $(NOOPENSSL),-e '/libssl/d',-e 's/ libssl,/ $(call F_GETSODPKG,libcrypto)/') \
	    $(if $(NOZLIB),-e '/zlib/d',-e 's/ zlib,/ $(call F_GETSODPKG,libz)/') \
	    $(if $(NOSRT),-e '/libsrt/d',-e 's/ libsrt,/ $(call F_GETSODPKG,libsrt)/') \
	    $(if $(NORIST),-e '/librist/d',-e 's/ librist,/ $(call F_GETSODPKG,librist)/') \
	    $(if $(NOEDITLINE),-e '/libedit/d',-e 's/ libedit,/ $(call F_GETSODPKG,libedit)/') \
	    $(if $(NOVATEK),-e '/libusb/d',-e 's/ libusb,/ $(call F_GETSODPKG,libusb)/') \
	    $(if $(NOPCSC),-e '/libpcsc/d',-e 's/ libpcsc,/ $(call F_GETSODPKG,libpcsc)/') \
	    $(if $(NOCURL),-e '/libcurl/d',-e 's/ libcurl,/ $(call F_GETSODPKG,libcurl)/') \
	    tsduck.control >$(TMPROOT)/DEBIAN/control`

lelegard, 2025-04-12:
@TobyBorland, precisely, that's the problem: `make -C tsduck installer`

You initially built with `make default docs-html NOVATEK=1 NOPCSC=1`. So, you built the binaries _without_ support for VATEK and PCSC but then you build the installer without the same restrictions. So, the installer keeps the references to VATEK and PCSC.

The `sed` command in the Makefile _deletes_ the lines in the control file which contain references to libusb and libpcsc when `NOVATEK` and `NOPCSC` are not empty. However, when they are empty (which is the case in your `make installer` command), the references to libusb and libpcsc are _replaced_ with the exact versions which are used by the binaries you just built, without deleting the lines. Since your binaries were built _without_ reference to these libraries, the lines are empty.

Each `make` command is independent from the preceding ones. There is no storage of which build options were previously used.

As a general rule, you must keep the same consistent list of build options in all `make` commands. In that case:
~~~
make -j10 default docs-html NOVATEK=1 NOPCSC=1
make installer NOVATEK=1 NOPCSC=1
~~~

Note the `-j` option in the build command to avoid waiting for hours. Adapt the job count to your core count and memory size.

closed 2025-09-14 by github-actions[bot]