tsduck/tsduck #1589: [Q] tsp spliceinject forked from daemon stuck on ubuntu 20.04

CMThF, 2025-03-06:
Hi!
We want to start a tsp spliceinject process from a daemon. The command works with the asset and the splice command XML when we run it in the terminal. It also works when we spawn it using std::system or popen from an application. But running the application as a daemon, the process hangs. It looks like the file poller is not working/terminated.

Do you have any tipps/experience with such a setup? What am I missing?

lelegard, 2025-03-07:
Hi @CMThF 

The most significant difference between an interactive process and a daemon is the user id. Maybe the user id under which the daemon runs is denied access to the input files or directories. Maybe those files are only accessible to your interactive user id. 

Make sure you have an output log file for the daemon process and look for error messages. Also try option `--debug` in the `tsp` command in the daemon to get more verbose messages in the log file.

CMThF, 2025-03-07:
That was our first thought as well. But the files are read/writable for all. The daemon itself can read and write them, and if we start other CLI tools from it, like `cat`, those can read the files as well. 

We are running with verbose d4 logs. A colleague will retrieve them and post them here later. But there are no errors or anything.

What we also tried was calling libtsduck directly. Same thing, just sitting on the wair-first-batch CV forever. Even though the files are there. We also tried editing/writing them after the start of tsp. No luck.

```Thread 53 (Thread 0x7f97d29ce700 (LWP 1082)):
 #0  futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x55ab7d814248) at ../sysdeps/nptl/futex-internal.h:183
 #1  __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x55ab7d8141e8, cond=0x55ab7d814220) at pthread_cond_wait.c:508
 #2  __pthread_cond_wait (cond=0x55ab7d814220, mutex=0x55ab7d8141e8) at pthread_cond_wait.c:647
 #3  0x00007f97f9e574bb in ts::Condition::wait(ts::Mutex&, long, bool&) () from /lib/libtsduck.so
 #4  0x00007f97f9f7df7a in ts::GuardCondition::waitCondition(long) () from /lib/libtsduck.so
 #5  0x00007f97d89b28ab in ts::SpliceInjectPlugin::start() () from /usr/lib/tsduck/tsplugin_spliceinject.so
 #6  0x00007f97fa1a145c in ts::TSProcessor::start(ts::TSProcessorArgs const&) () from /lib/libtsduck.so```

lelegard, 2025-03-07:
Also, you did not mention the TSDuck version. From the stack dump you provided, I see old classes such as `ts::Mutex` and the like. These classes were replaced long ago by modern C++ built-in classes. So, you must run some old version. 

I don't know if it would change anything, but you should try with the latest version.

kiinz55, 2025-03-07:
Our command looks as follows: _tsp -v -d4 --add-input-stuffing 1/10 -I file /bitcodin-fast-tmp-dir/adf4-7149-f51d-3a6e/bitHLSsegment.ts -P pmt --add-pid 134/0x86 --add-programinfo-id 0x43554549 -P spliceinject --pid 134 --pts-pid 101 --files /fast-tmp-dir/6973-ce04-2cd9-0841/8846-fea6-d3ae-4034.xml --inject-count 1 --wait-first-batch -O file /fast-tmp-dir/adf4-7149-f51d-3a6e/bitHlSsegmentWithInjectedScte35.ts_

These are the only logs we are receiving until it stalls:
```
tsp: Debug[4]: debug level set to 4
tsp: Debug: ====> tsp -v -d4 --add-input-stuffing 1/10
* Debug[4]: debug level set to 4
* Debug: registering processor plugin "pmt", status: ok
* Debug: registering processor plugin "spliceinject", status: ok
```
Yes, we require the --wait-first-batch because we want to do it for an input TS file and the XML file is already present right from the start.

CMThF, 2025-03-07:
I have a minimal repro:
cpp application
```
int main()
{
    const auto parentPath = "<PATH_TO_FILES>"s;
    const auto cmd = "tsp -I file \""s + parentPath + "/<TS_FILE>.ts\" -P spliceinject --pid 134 --pts-pid 0x101 --files \""s +
        parentPath + "/<SPLICE COMMAND FILE>.xml\" -w -O file \""s + parentPath + "/output.ts\"";

    const auto sysres = std::system(cmd.c_str());
    return 0;
}
```

All input files with 666 permissions.
Started as application, works like a charm. 
Started as daemon, e.g. with `start-stop-daemon -SbCv -x <PATH_TO_EXECUTABLE>` results in hanging tsp process.

lelegard, 2025-03-07:
> Started as daemon, e.g. with start-stop-daemon -SbCv -x <PATH_TO_EXECUTABLE> results in hanging tsp process.

I cannot reproduce your test. I compiled your small C++ app with the same `tsp` command (but using TS and XML files from test-026 in the test suite).

> We are on version [Version 3.26-2349](https://github.com/tsduck/tsduck/releases/tag/v3.26-2349) as we are using ubuntu 20.04

- I first tried with the latest version, TSDuck 3.40-4165 on Ubuntu 24.10. All worked as expected, including as a daemon.
- I tried to install the same binary as you for version 3.26-2349 but couldn't because it was built for Ubuntu 20 and the library dependencies are not the same in Ubuntu 24.10.
- I rebuilt TSDuck 3.26-2349 from sources using gcc 14 on Ubuntu 24.10 and all worked correctly, including as a daemon.

So, there must be something specific in your command or system configuration (or specific to Ubuntu 20 but I doubt about it).

Do you use absolute paths in file names? Maybe the daemon is started with some other current working directory. Just trying to guess. This seems more a Unix debugging issue than a TSDuck issue.

CMThF, 2025-03-07:
Thank you @lelegard for the quick help and feedback! Looking into it!

lelegard, 2025-03-07:
Please let us know if you find the explanation. It may help other people running into a similar problem.

lelegard, 2025-03-07:
I am skeptical about your logs:

> These are the only logs we are receiving until it stalls:
> ~~~
> tsp: Debug[4]: debug level set to 4
> tsp: Debug: ====> tsp -v -d4 --add-input-stuffing 1/10
> * Debug[4]: debug level set to 4
> * Debug: [...]

You should see a trace at the start of the background thread which polls the files:
https://github.com/tsduck/tsduck/blob/73b5f7516a848a11165d0efd67839e9117abd21c/src/tsplugins/tsplugin_spliceinject.cpp#L934-L945

Or maybe the trace did not exist in the old version you use.

The file listener thread is started whenever the option `--files` is used:
https://github.com/tsduck/tsduck/blob/73b5f7516a848a11165d0efd67839e9117abd21c/src/tsplugins/tsplugin_spliceinject.cpp#L377-L378
https://github.com/tsduck/tsduck/blob/73b5f7516a848a11165d0efd67839e9117abd21c/src/tsplugins/tsplugin_spliceinject.cpp#L463-L466

CMThF, 2025-03-07:
A newer version might indeed help. My reproduction worked on Ubuntu 24.04 with the latest TSDuck, but as you mentioned, it's likely a side effect of something else.

Unfortunately, we're currently locked into the older version since we're running on an older Linux release. It'll take some time before we can update.

We will go with a workaround for now, so it is not started from a daemon. 

Thank you for digging into it!

closed 2025-03-07 by CMThF

lelegard, 2025-03-07:
> Unfortunately, we're currently locked into the older version since we're running on an older Linux release

By experience, the cost of obsolescence can become very high sometimes. You spend time to investigate, then time to implement a workaround, then investigate an new issue that appears with the workaround, etc. Then, you realized that you wasted much more time than upgrading the system (which is quite easy with Ubuntu).