tsduck/tsduck #1602: [Q] injection is not done as expected as triggers are dropped although timestamps should overlap.

PhilippSchwellenbach, 2025-03-20:
Hi,

I need help because tsp injection plugin drops my triggers although the times are matching. I have 3 different examples where I have problems with.
All these examples are based on a subsegment (no 16) from my video respectively audio stream using basically the same command for injection.

[image]

[1] injection into avc video segment:
XML file:
`<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <splice_information_table protocol_version="0" pts_adjustment="900,000" tier="0x0FFF">
    <metadata PID="134"/>
    <time_signal pts_time="5,401,646"/>
    <splice_segmentation_descriptor identifier="0x43554549" segmentation_event_id="0x00000001" segmentation_event_cancel="false" web_delivery_allowed="false" no_regional_blackout="false" archive_allowed="true" device_restrictions="3" segmentation_type_id="0x34" segment_num="0" segments_expected="1" sub_segment_num="0" sub_segments_expected="0">
      <segmentation_upid type="0x09"/>
    </splice_segmentation_descriptor>
  </splice_information_table>
</tsduck>`

[avcVideoSegment16](https://storage.googleapis.com/scte35injection_schwellenbach/segment_16_avc.ts)

the tsp command for the avc segment looks like this::
`tsp -v --add-input-stuffing 1/10 -I file segment_16_avc.ts  -P pmt --add-pid 134/0x86 --add-programinfo-id 0x43554549 -P spliceinject --pid 134 --pts-pid 257 --files 'segment_16.xml' --inject-count 1 --wait-first-batch -O file inj.ts`
but what I understand from the command's output is that the trigger is ignored because the "end" equals the "current pts":
[image]
so i modified the pts_adjustment by +1 to get the trigger injected. Not really nice but acceptable workaround.
I still wanted to put up this example because it builds of for the next two and it brings already up the first question:

[Q] Why is the trigger not injected here because the timestamps of trigger and segment are perfectly matching?

[2] injection into aac audio segment (this is from the same input the audio segment that corresponds to the video segment from example [1]
Here we can see that the audio segment actually starts a bit earlier then the trigger. The trigger XML is:
`<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <splice_information_table protocol_version="0" pts_adjustment="900,000" tier="0x0FFF">
    <metadata PID="134"/>
    <time_signal pts_time="5,400,000"/>
    <splice_segmentation_descriptor identifier="0x43554549" segmentation_event_id="0x00000001" segmentation_event_cancel="false" web_delivery_allowed="false" no_regional_blackout="false" archive_allowed="true" device_restrictions="3" segmentation_type_id="0x34" segment_num="0" segments_expected="1" sub_segment_num="0" sub_segments_expected="0">
      <segmentation_upid type="0x09"/>
    </splice_segmentation_descriptor>
  </splice_information_table>
</tsduck>`

[aacAudioSegment16](https://storage.googleapis.com/scte35injection_schwellenbach/segment_16_aac.ts)

the used command is:
`tsp -v --add-input-stuffing 1/10 -I file segment_16_aac.ts  -P pmt --add-pid 134/0x86 --add-programinfo-id 0x43554549 -P spliceinject --pid 134 --pts-pid 258 --files 'segment_16.xml' --inject-count 1 --wait-first-batch -O file audioInj.ts`

for this i get the following result:

[image]
here we can see that the difference from end to current is much bigger and I tested around with the pts_adjustment how much I offset I need to add to the trigger to get it injected. The offset is 961. So 960 and the +1 that we hacked for video too. The coincidence here is that the 960 is exactly half an audio packet - respectively the time that the audio stream starts earlier then the 70sec where the trigger is supposed to be. But from scratch I don't see a reason why there is a need to further delay the trigger to match with the start of the 2. audio packet (as audio streams starts half a packet before 70sec and trigger injection works with half a packet after 70 sec).

[Q] I was thinking maybe there is an additional offset needed due to aac's priming feature?

So i tried to workaround this by just adding 1 packet of additional offset to the trigger to get it injected into my aac audio segments.
But unfortunately this doesn't work consistently ->

[3] i have another version of the audio segment 16 where it is necessary to add 2.5 audio packets of delay to get the trigger injected:
`<tsduck>
  <splice_information_table protocol_version="0" pts_adjustment="900,000" tier="0x0FFF">
    <time_signal pts_time="5400000"/>
    <splice_segmentation_descriptor identifier="0x43554549" segmentation_event_id="0x00000001" segmentation_event_cancel="false" web_delivery_allowed="false" no_regional_blackout="false" archive_allowed="true" device_restrictions="3" segmentation_type_id="0x34" segment_num="0" segments_expected="1" sub_segment_num="0" sub_segments_expected="0">
      <segmentation_upid type="0x09"/>
    </splice_segmentation_descriptor>
  </splice_information_table>
</tsduck>`

[aacAudioSegment16_2](https://storage.googleapis.com/scte35injection_schwellenbach/segment_16_aac_2.ts)

command:
`tsp -v --add-input-stuffing 1/10 -I file segment_16_aac_2.ts  -P pmt --add-pid 134/0x86 --add-programinfo-id 0x43554549 -P spliceinject --pid 134 --pts-pid 258 --files 'segment_16_2.xml' --inject-count 1 --wait-first-batch -O file audioInj.ts`

resulting in this:

[image]

here i need to add additional 4800 to pts_adjustment (corresponding to 2.5 audio packets delay) and the +1 "hack" to get the trigger injected as I want it to. When checking the difference between the two audio segments I see that the 2. audio segment example is stereo instead of mono and has higher bitrate.
So maybe there is a correlation between the necessary offset and the number of channels here.

So in the end my most important question here is the following:
[Q] Is there a way to change my command to automatically adapt to the "audio delay" OR can you provide a recommendation how to consistently calculate the delay that is necessary for aac audio to make sure that the injection always works?

lelegard, 2025-03-21:
Hi @PhilippSchwellenbach 

I am not sure to understand what you try to achieve.

Inside a receiver, sections and audio/video follow distinct paths. The sections are analyzed by the application software while the audio/video is processed by a codec (hardware or software).

Therefore, the synchronization of SCTE-35 sections and the content PTS is loose. The section are injected well before a given event (determined by its PTS). The PTS is an hypothetical point in the future (unlike the PCR which is a precise time in the TS) where audio and video will be resynchronized by the player.

Inside the stream, at the point of an audio or video packet, the event is neither the audio PTS nor the video PTS. The actual point in time where the player will synchronize the audio and video PTS is later. However, at injection level, we shall not rely on that, we must inject sufficiently in advance so that the application software has the time to grab the section, decode it, process it and synchronize what should be done with the player at the given future point.

You can use either the audio PTS or the video PTS, it doesn't matter. Whichever you use, you just need to keep the same reference and plan everything well in advance before the event PTS. "Well in advance" is a matter of seconds while PTS is expressed in 1/90000 second (0.00001 second).

For non-immediate splice_insert() and time_signal() commands, the algorithm is explained in the documentation as follow, with the default values:

- Start to insert the first section 2,000 milliseconds before the specified splice PTS value, as seen in the reference PID (option `--start-delay`).
- Repeat the injection 2 times (option `--inject-count`).
- The interval between injections of the same section is 800 milliseconds (option `--inject-interval`).
- No need to inject after the event PTS since the event is past.

Note that "after the event PTS" is a loose concept because the actual point of time of the PTS is later, after the audio or video packet containing that PTS value. The interval between that packet and the actual event is unknown, not predicabale, and not even constant, especially because it changes with the type of video frame (I/D/B/etc) and potentially variable encoding bit rate.

So, again, plan "well in advance". Once we reach "somewhere around the PTS value of the event", it's too late, the receiver won't have time to process the section, so we just drop it.

Also consider that a section is always "inserted" by replacing a null packet. That null packet is, by definition, not the same packet as the audio or video one containing the last PTS. Time has ticked between the last packet with a PTS and the next null packet where the section can be injected. So, the estimated PTS at the point of potential insertion is larger than the last extracted PTS value.

In file segment_16_avc.ts:
- First PTS in PID 257: 6,301,646
- PTS in time_signal: 5,401,646 + PTS adjustment 900,000 = 6,301,646

In files segment_16_aac.ts and segment_16_aac_2.ts:
- First PTS in PID 258: 6,299,040
- PTS in time_signal: 5,400,000 + PTS adjustment 900,000 = 6,300,000 (10 ms after first PTS)

In all cases, the very start of the file is at, or just after the event. When the next null packet arrives, it is too late to inject the section, the event is past, it is too late the send the splice command, the splice section becomes useless, and we drop it.

As a last point, note that the "PTS adjustment" in a splice section is not supposed for such hack. It is used in case of PTS restamping. From SCTE-35:

> Any device that re-stamps PCR/PTS/DTS and that passes these cue messages to a downstream device should modify the pts_time field or the pts_adjustment field in the message in all PIDs conforming to this standard. Modifying the pts_adjustment field is preferred because the restamping device will not have to be knowledgeable of the pts_time field that may occur in multiple commands (and possibly in future commands).

lelegard, 2025-03-28:
> in some cases I can have null packets at the start of the stream before the first video or audio packet and then the injection would work with my +1 hack at least.

Initially, the plugin `spliceinject` has no idea of the clock reference. It cannot insert anything before the first packet containing a PTS. This first PTS packet starts the clock and the plugin knows which events are to be inserted in the future and which events are already obsolete.

> But in some cases I have the content packets first and then the trigger can only be injected if I target the pts of a "later packet" with the end time of the event.
Is that correct?

Yes. However, as I explained in my previous response, the target event PTS in the SCTE-35 section is _not_ the point where the section will be injected. It is the point where the receiver device will have something to do. This means that the section must be inserted several times well before that point, typically a few seconds. If the first PTS in the stream is equal to the target PTS of the event (or close to it by a few milliseconds), it is too late to insert the section.

> And I am assuming that the null packets are generated by the stuffing (of -add-input-stuffing 1/10 in my case), is this correct?

Yes, it is correct. As explained [here](https://tsduck.io/download/docs/tsduck.html#tsp-reference), in section "Modifying, inserting and deleting packets", `tsp` never "creates" packets, it can only replace "free space" - null packets - with meaningful packets.

In the case of broadcast TS, there are always some null packets to stuff up to the target constant bitrate. However, in the case of streaming TS, there is usually no null packets on input and you need to insert some using option `--add-input-stuffing`. Note that, in that case, it is recommended to remove the remaining null packets using `-P filter -n -p 0x1FFF` just before the output plugin.

PhilippSchwellenbach, 2025-03-28:
> -P filter -n -p 0x1FFF just before the output plugin
this is a really good hint thanks!
I will proceed with this.
I think we can close this now. In case I have another question I can reopen a new one.
Thank you very much for your help. And fast reply today!

closed 2025-03-28 by lelegard