tsduck/tsduck #1605: [Q] How to switch files when sending data in TSDuck?

xengine-qyt, 2025-04-01:
I want to implement this using the C++ API. I want TSDuck to control the transmission rate and automatically stream to a specific UDP address and port. At a specific time point during the streaming process, I want to switch files and continue streaming without interruption, ensuring that counters and timestamps remain continuous. How can this be achieved?

xengine-qyt, 2025-04-01:
I tested using the command: ./tsp -I file input.ts -P regulate -O ip 10.0.1.101:8890 and found that the bitrate fluctuation is very small. How is this achieved? How should the C++ interface be implemented? For example, if I have a TS with a fixed bitrate of 8000 kbps, and it fluctuates within 1 kbps, how can this be achieved?

lelegard, 2025-04-01:
Hi @xengine-qyt 

> At a specific time point during the streaming process, I want to switch files and continue streaming without interruption, ensuring that counters and timestamps remain continuous. How can this be achieved?

This is easily possible only if all files have the same structure: same signalization, same PID's, same services and same audio/video codecs. Otherwise, it is still possible but more complicated because you have to rebuild the signalization and manage its transformation when you switch files.

Also make sure that the files start at the beginning of an intra-frame and stop at the end of a frame. Audio transitions may be out of sync from the video stream.

> For example, in the following code, how can I send data at a specific rate, and ensure the sending bitrate is exactly the fixed bitrate? My media file is packaged with a fixed bitrate of 8MB. I want to send it at this same bitrate; how should I do it?

The input file is read at the speed of the disk. You need to suspend the program the appropriate time before sending packets over UDP. The class `ts::BitRateRegulator` does this. See the source code of the `regulate` plugin for an example.

I do not want to review your code in details. I just noticed that your restamp PCR and PTS but not DTS. When the restamping offset is non zero, the receiver will probably decode nothing because the unmodified decoding time stamps will be too far away from the PCR clock.

>  the bitrate fluctuation is very small. How is this achieved? How should the C++ interface be implemented? For example, if I have a TS with a fixed bitrate of 8000 kbps, and it fluctuates within 1 kbps, how can this be achieved?

I don't understand what is the question.

xengine-qyt, 2025-04-02:
hello @lelegard 
Is it like this? But I don't know how this method controls the switching of files. Do I need to read the TSPacket myself? How can I read the TSPacket on my own, like the custom read callback in ffmpeg?
```

	ts::TSProcessorArgs m_TSArgs;
	ts::TSProcessor m_TSProcess(m_TSReport);

	m_TSArgs.app_name = u"xx";
	m_TSArgs.input = { u"file", {u"D:\\windows-ffmpeg\\x64\\test.ts"} };
	m_TSArgs.output = { u"ip", {u"10.0.1.101:8890"} };
	m_TSArgs.plugins = {
		//{u"regulate", {u"--pcr"}},
		{u"regulate", {u"--bitrate", u"8000000"}},
	};

	if (!m_TSProcess.start(m_TSArgs))
	{
		return 1;
	}
	m_TSProcess.waitForTermination();
```
Is it possible to implement a custom callback and rewrap the TSPacket? This way, I can switch files freely, as long as all the video parameters in my TS files are consistent.
Or is custom plugin development still required?

xengine-qyt, 2025-04-02:
> The input file is read at the speed of the disk. You need to suspend the program the appropriate time before sending packets over UDP. The class `ts::BitRateRegulator` does this. See the source code of the `regulate` plugin for an example.

and What about the RegulatePlugin plugin you said? Can I reuse it directly, or do I need to refer to this code implementation?

xengine-qyt, 2025-04-02:
@lelegard  thanks

> This is easily possible only if all files have the same structure: same signalization, same PID's, same services and same audio/video codecs. Otherwise, it is still possible but more complicated because you have to rebuild the signalization and manage its transformation when you switch files.

Yes, my TS files are processed according to unified encoding parameters. I noticed that when I switch files and push the stream again, it causes errors in the PMT and PAT intervals, as well as PRC precision interval errors. Do I need to repack the TS package? Or should I reset it? Does TSDuck have an interface for this?

xengine-qyt, 2025-04-03:
"I found that it can be achieved by rewriting the input file, but after rewriting, I noticed that the playback is somewhat choppy and not smooth. Could it be a performance issue caused by m_TSFile.readPackets(&m_TSPacket, &m_TSMetadata, 1, m_TSReport)?
```
void InputHandler::handlePluginEvent(const ts::PluginEventContext& context)
{
	ts::PluginEventData* data = dynamic_cast<ts::PluginEventData*>(context.pluginData());
	if (data != nullptr)
	{
		ts::TSPacket m_TSPacket;
		ts::TSPacketMetadata m_TSMetadata;
		ts::AsyncReport m_TSReport(ts::Severity::Info);
		
		if (m_TSFile.readPackets(&m_TSPacket, &m_TSMetadata, 1, m_TSReport))
		{
			data->append(m_TSPacket.b, ts::PKT_SIZE);
		}
		else 
		{
			_report.info(u"end of input");
			m_TSFile.rewind(m_TSReport);
			m_TSFile.readPackets(&m_TSPacket, &m_TSMetadata, 1, m_TSReport);
			data->append(m_TSPacket.b, ts::PKT_SIZE);
		}
	}
}
bool CALLBACK XEngine_ConvertStream_ThreadSecond()
{
	ts::AsyncReport m_TSReport(ts::Severity::Info);
	
	ts::TSPacket m_TSPacket;
	ts::UDPSocket m_TSUDPSocket;
	ts::PCRRegulator m_TSRegulator;
	ts::TSPacketMetadata m_TSMetadata;
	ts::ContinuityAnalyzer m_TSCCMaker;
	
	// 打开 TS 文件
	if (!m_TSFile.openRead("D:\\windows-ffmpeg\\x64\\test.ts", 0, m_TSReport))
	{
		return 1;
	}

	ts::TSProcessorArgs m_TSArgs;
	ts::TSProcessor m_TSProcess(m_TSReport);
	InputHandler input(m_TSReport);
	m_TSProcess.registerEventHandler(&input, ts::PluginType::INPUT);

	m_TSArgs.app_name = u"xx";
	//m_TSArgs.input = { u"file", {u"D:\\windows-ffmpeg\\x64\\test.ts"} };
	m_TSArgs.input = { u"memory", {} };
	m_TSArgs.output = { u"ip", {u"127.0.0.1:8890"} };
	m_TSArgs.plugins = {
		{u"pcradjust", {}},
		{u"continuity", {u"--fix"}},
		{u"regulate", {u"--pcr"}},
		//{u"regulate", {u"--bitrate", u"8000000"}},
	};

	if (!m_TSProcess.start(m_TSArgs))
	{
		return 1;
	}
	m_TSProcess.waitForTermination();
	return 0;
```

closed 2025-04-03 by xengine-qyt