tsduck/tsduck #1606: [Q] read EPG with DVB-T

marcoTelecolor, 2025-04-01:
Hi,
I would like to monitor the presence of epg data within a TV channel and read the related program names.
From the tsduck manual I tried to analyze the mux and it turns out that it is enabled on some channels but with zero EPG days even if I see them on television

`c:\tsduck\bin\tsp -I dvb -u 38 -P until -s 30 -P eit -O drop`

Summary
-------
TS id:         930 (0x03A2)
Last UTC:      2025/04/01 13:43:59
EITp/f actual: 618
EITp/f other:  0
EITs actual:   238
EITs other:    0

TS      Services  With EITp/f  With EITs  EPG days
------  --------  -----------  ---------  --------
Actual        25           15          6         0
Other          0            0          0         0

A/O  TS Id   Srv Id  Name               EITp/f  EITs  EPG days
---  ------  ------  -----------------  ------  ----  --------
Act  0x03A2  0x001F  Servizio 31        no      no           0
Act  0x03A2  0x0020  Servizio 32        no      no           0
Act  0x03A2  0x0021  Servizio 33        no      no           0
Act  0x03A2  0x0022  Servizio 34        no      no           0
Act  0x03A2  0x0023  Servizio 35        no      no           0
Act  0x03A2  0x0024  Servizio 36        no      no           0
Act  0x03A2  0x0025  Servizio 37        no      no           0
Act  0x03A2  0x0026  Servizio 38        no      no           0
Act  0x03A2  0x0027  Servizio 39        no      no           0
Act  0x03A2  0x0063  Ghost              no      no           0
Act  0x03A2  0x0BBC  Rete4 HD           yes     yes          0
Act  0x03A2  0x0BBD  Canale5 HD         yes     yes          0
Act  0x03A2  0x0BBE  Italia1 HD         yes     yes          0
Act  0x03A2  0x0BCC  20Mediaset HD      yes     yes          0
Act  0x03A2  0x0BEF  Mediaset Extra HD  yes     yes          0
Act  0x03A2  0x0C20  Rete4 HD           yes     no           0
Act  0x03A2  0x0C21  Canale5 HD         yes     no           0
Act  0x03A2  0x0C22  Italia1 HD         yes     no           0
Act  0x03A2  0x0C30  20Mediaset HD      yes     no           0
Act  0x03A2  0x0C80  Test HEVC Main10   yes     yes          0
Act  0x03A2  0x0DB0  Rete4 HD           yes     no           0
Act  0x03A2  0x0DB1  Canale5 HD         yes     no           0
Act  0x03A2  0x0DB2  Italia1 HD         yes     no           0
Act  0x03A2  0x0DC0  20Mediaset HD      yes     no           0
Act  0x03A2  0x0DE4  Mediaset Extra HD  yes     no           0`

and if I analyze the single channel I get this:

`c:\TSDuck\bin>tsp -I dvb -u 38 -P zap "20Mediaset HD" -P until -s 30 -P eit -O drop

Summary
-------
TS id:         930 (0x03A2)
Last UTC:      2025/04/01 13:47:40
EITp/f actual: 0
EITp/f other:  0
EITs actual:   0
EITs other:    0

TS      Services  With EITp/f  With EITs  EPG days
------  --------  -----------  ---------  --------
Actual         3            0          0         0
Other          0            0          0         0

A/O  TS Id   Srv Id  Name           EITp/f  EITs  EPG days
---  ------  ------  -------------  ------  ----  --------
Act  0x03A2  0x0BCC  20Mediaset HD  no      no           0
Act  0x03A2  0x0C30  20Mediaset HD  no      no           0
Act  0x03A2  0x0DC0  20Mediaset HD  no      no           0`

I tried increasing the time to 300 seconds but I still don't see the EPG data. Am I doing the syntax wrong or what? Thanks a lot

Marco

lelegard, 2025-04-01:
Hi @marcoTelecolor 

In the first command, you have EPG for the current TS. All channels which are probably real TV channels have EIT p/f and six of them have EIT schedule. The report mentions 618 sections of EITp/f actual and 238 sections of EITs actual. So, there are EPG data.

According to the [user's guide](https://tsduck.io/download/docs/tsduck.html#_eit), _"The EPG depth in days is also reported by service (number of days in advance an event is signaled by an EIT schedule)"_. If the EPG days is zero, this does not mean that there is no EPG. This means that all EIT schedule describe events which start within 24 hours from the time in the last TDT, so "zero day in advance". One day in advance means that there are EIT schedule for events which start between 24 and 48 hours in the future, etc.

When you see the EPG on TV, do you see later events?

In the second command, you have no EIT because they are removed by default by the `zap` plugin. From the [user's guide](https://tsduck.io/download/docs/tsduck.html#_zap):

> `--eit`
> Keep EIT sections for the specified services. EIT sections for other services are removed. By default, all EIT’s are removed.

Also note that there are three distinct services in the TS with the same name "20Mediaset HD", so you'd better specify a service id and not the ambiguous service name.

marcoTelecolor, 2025-04-02:
> Hi [@marcoTelecolor](https://github.com/marcoTelecolor)
> 
> In the first command, you have EPG for the current TS. All channels which are probably real TV channels have EIT p/f and six of 
> [...]
> When you see the EPG on TV, do you see later events?

I actually only see those within 24 hours.

> In the second command, you have no EIT because they are removed by default by the `zap` plugin. From the [user's guide](https://tsduck.io/download/docs/tsduck.html#_zap):
>
>
> Also note that there are [...]

you're right! Now I added the command --eit after zap and used the service instead of the name

`c:\TSDuck\bin>tsp -I dvb -u 38 -P zap --eit 0x0BEF -P until -s 10 -P eit -O drop
Summary
-------
TS id:         930 (0x03A2)
Last UTC:      2025/04/02 08:36:52
EITp/f actual: 17
EITp/f other:  0
EITs actual:   18
EITs other:    0

TS      Services  With EITp/f  With EITs  EPG days
------  --------  -----------  ---------  --------
Actual         1            1          1         0
Other          0            0          0         0

A/O  TS Id   Srv Id  Name               EITp/f  EITs  EPG days
---  ------  ------  -----------------  ------  ----  --------
Act  0x03A2  0x0BEF  Mediaset Extra HD  yes     yes          0`

But now ho can I send epg to file and read it? I tried to save the file but it doesn't save the epg, it saves the whole ts.

I searched in the manual but I only found the section for the inject but I didn't understand how to extract them
`tsp -I dvb -u 38 -P zap --eit 0x0BEF -P until -s 10 -P eit -O file epg_data.txt`

thankyou very much!
Marco

lelegard, 2025-04-02:
> But now ho can I send epg to file and read it?

It depends on which format you want it.

For a human-readable text format, to collect EIT's for service id 0x0BEF:
~~~
tstables file.ts --tid 0x50-0x5F --tid-ext 0x0BEF --all-sections --no-deep-duplicate
~~~

To get an XML file:
~~~
tstables file.ts --tid 0x50-0x5F --tid-ext 0x0BEF --pack-all-sections --no-deep-duplicate --xml-output out.xml
~~~

Or a JSON file:
~~~
tstables file.ts --tid 0x50-0x5F --tid-ext 0x0BEF --pack-all-sections --no-deep-duplicate --json-output out.json
~~~

I let you read the user's guide for the exact meaning of each option and the format of XML and JSON files.

marcoTelecolor, 2025-04-02:
ok, thankyou!!
It works!!
I also managed to combine the two commands so that I don't have to save the file but analyze it directly from the antenna. 
Thanks again
Marco

lelegard, 2025-04-06:
I also added option `--epg-dump` to the plugin `eit` to dump a timeline of events per service, in human-readable form instead of EIT sections.

closed 2025-09-07 by github-actions[bot]