tsduck/tsduck #1607: [Q] cut out a time period of a .ts file

ToniTram81, 2025-04-01:
Hello,
I have a ts file (e. g. duration 1h) and want to cut out a time period of 10 minutes from timeline 00:15:00 to 00:25:00.
Is this possible with tsp? In the user´s guide I can not find any parameters for this. The original broadcast time is not relevant at this point.

tsp -I original.ts  -P ..... -O file period.ts

lelegard, 2025-04-02:
Hi @ToniTram81 

The question is how to match the start and stop time stamps in the TS file? Once you have the answer, then you know where to cut.

If your stream is CBR, then you can map durations to numbers of TS packets (use `tsbitrate` first if you know it's CBR but you don't know the bitrate). Then use plugins `skip` and `until` to cut the TS file.

ToniTram81, 2025-05-15:
Hello,
the plugin skip s working fine, but the plugin until not: 

I want to have only the first 15 seconds of the ts file:
tsp -v -I file video.ts -P until --second 15 -O file video15.ts
but both ts files have the same length and file size...

lelegard, 2025-05-15:
Using `-P until --second 15` means stop after 15 seconds, wall clock time, not playout time. Processing a file, even a large one, takes less than 15 seconds.

See the [tsp documentation](https://tsduck.io/download/docs/tsduck.html#tsp-reference), specifically the section named "Rendering speed and transmission speed".

lelegard, 2025-06-06:
@ToniTram81 

I just added option `--pcr-based` to plugin `until` so that `--second 15` is interpreted as 15 seconds of playout time, based on PCR. Without `--pcr-based`, it keeps the default interpretation of 15 seconds of real time.

For consistency, the plugin `skip` has been enhanced with the same options as `until` (including the new `--pcr-based`).