tsduck/tsduck #1618: [Q]  how to exit waitForTermination

xengine-qyt, 2025-04-23:
After ts::TSProcessor is started using start, it waits for termination using waitForTermination, but calling abort has no effect.

I am using a pipe to read and write data. TSProcessor reads data from the pipe and pushes it via UDP. When I don't write any data to the pipe, calling the abort function doesn't cause waitForTermination to return. However, if I continuously write data to the pipe and then call abort, the waitForTermination function works and exits immediately.

Why does this happen? How can I properly terminate it?

lelegard, 2025-04-23:
Hi @xengine-qyt 

This is the expected behaviour. There is often a misunderstanding on the word "abort". In most multithreading environments, "abort" means "set threads in an abort state that will be taken into account when threading activities take place".

With TSDuck, the abort state is processed by each plugin when this plugin has finished its processing on a packet and wait for more. In the case of the input plugin, the abort is processed when packets are received and passed to the next plugin.

In practice, each plugin runs in its own thread. The main loop of the thread is part of the TSDuck library. The plugin is repeatedly called from that main loop, to read some packets (input plugin), process one packet (packet processing plugin), or write some packets (output plugin). When the thread runs inside the plugin, we cannot "abort" it because we don't know what it is doing. The abort takes place when the execution returns from the plugin into `tsp`. In the case of an input plugin, this means when some packets were read. If the input plugin is infinitely waiting for packets which never come, the execution never returns to `tsp` and the abort is infinitely delayed.

The only way to avoid an infinite deadlock is to specify an input timeout, if the input plugin has such option.

Independently of TSDuck, abruptly aborting a thread is usually a very bad idea because we have no idea about which resources are still in use, lost, or whatever (open files, pipes, allocated memory, etc). In most multithreading environments, an unconditional and immediate abort either does not exist or is documented as a last resort and unrecoverable method because of the potential resources still in limbo.

xengine-qyt, 2025-04-24:
ok thanks

closed 2025-04-24 by xengine-qyt