tsduck/tsduck #1619: [Q] ipv6 question

dbhat-peacock, 2025-04-23:
I am referring to (https://github.com/tsduck/tsduck/issues/1551 where it is suggested that ipv6 works with IP feed (no SRT support yet). But when I used tsp (TSDuck - The MPEG Transport Stream Toolkit - version 3.38-3822), 

tsp -I ip  2a05:d018:151a:e700:7363:95bc:96b1:bd8f:7086  -P until --seconds 10 -P analyze -O drop

I got an error:  (Error: ip: 2a05:d018:151a:e700:7363:95bc:96b1:bd8f: Address family for hostname not supported). 

The ipv6 address is good and receiving an MPEG stream as I checked with ffmpeg.

My question: what works with ipv6 as far as tsp is concerned? Any pointers will be great 

Thanks
Dinkar

dbhat-peacock, 2025-04-23:
Sorry, I think the feed on the above address is an MPEG-2 TS over SRT stream which you already mentioned does not work as yet. @lelegard But I will keep the question open just to be sure that the error is indeed because of SRT nonsupport and not that ipv6 is not supported at all.

lelegard, 2025-04-23:
Hi @dbhat-peacock 

Version 3.38 is older that our discussion in #1551. Could you try with the latest version, 3.40? IPv6 should be fully supported, including with SRT.

closed 2025-10-05 by github-actions[bot]