tsduck/tsduck #1621: [Q] listen srt question...

xengine-qyt, 2025-04-25:
tsp comand:tsp -I srt --listener :10080 --stream --streamid=live/livestream -P regulate --pcr -P continuity --fix -P pcradjust -O ip 10.0.4.69:8890

ffmpeg command:ffmpeg -re -stream_loop -1 -i "./1.ts" -c:a copy -c:v copy -f mpegts "srt://10.0.7.1:10080?streamid=live/livestream,m=publish"

tsp version:tsduck_3.26
 report error:
16:26:34.927567/SRT:RcvQ:worker*E:SRT.c: HSREQ/rcv: Agent uses STREAM API, but the Peer declares MESSAGE API. Not compatible transmission type, rejecting.
16:26:34.927629/SRT:RcvQ:worker*E:SRT.c: interpretSrtHandshake: process HSREQ returned unexpected value 0
16:26:34.927748/SRT:RcvQ:worker*E:SRT.c: @1021768730:newConnection: connection rejected due to: CONNECTION REJECTED
16:26:34.927780/SRT:RcvQ:worker*E:SRT.c: UU:newConnection: rsp(REJECT): 1012 - MessageAPI/StreamAPI collision

tsp version:tsduck_3.40 no error report ,it is work...
Is this a bug in tsduck_3.26 version or is it not used correctly?

lelegard, 2025-04-25:
Hi @xengine-qyt 

These messages are reported by libsrt, the SRT library. I am not an SRT expert, so I cannot tell, although the command looks legitimate to me.

Considering the difference in versions, there are also different versions of libsrt between Ubuntu 20 and 24. You can view it using `tsp --version=srt` (or `=all`). The latest version of libsrt is 1.5.4. If it works fine with a recent version of libsrt, I would say fine, looks like a fixed problem with libsrt.

In case of suspicious error messages from libsrt, I would suggest to ask the question in the libsrt issues (https://github.com/Haivision/srt/issues). The guys from Haivision are cool and helpful. They are aware of TSDuck (they have committed a few PR's in TSDuck and I have committed a few in libsrt). If the problem is about how SRT is used, they will help.

xengine-qyt, 2025-04-26:
> Hi [@xengine-qyt](https://github.com/xengine-qyt)
>
> These messages are reported by libsrt, the SRT library. I am not an SRT expert, so I cannot tell, although the command looks legitimate to me. [...]

The previous version of SRT was 1.4. I upgraded libsrt to 1.5.4, but the same error still occurred. Only upgrading TSDuck fixed it...

lelegard, 2025-04-26:
> The previous version of SRT was 1.4. I upgraded libsrt to 1.5.4, but the same error still occurred. Only upgrading TSDuck fixed it...

OK, so that must be a TSDuck improvement. Version 3.26 is very old (exactly 4 years ago). It has been designed for an old version of libsrt. There have been a lot of changes up to version 3.40.

I cannot really investigate an issue that occurred in a 4 years old version but appears to be fixed. I can only investigate issues in the current version.

closed 2025-04-28 by xengine-qyt