tsduck/tsduck #1625: [FEAT] Support for parsing/modifying ASTRA Service Guide Table (SGT) and related descriptors

nicholas-fr, 2025-04-29:
#### Is your feature request related to a problem?
As part of the broadcast team at TP Vision, we regularly use TSDuck and other tools for TS analysis and modification.
Transport streams from certain operators such as HD+ make use of private signalling defined by SES/ASTRA.
Currently we are unable to read and modify the private signalling using TSDuck, aside from reading out the raw data.

#### Proposed solution:
In order to analyse and modify such Transport Streams for testing purposes, ideally TSDuck would be able to parse the Service Guide Table (SGT) and related private descriptors (service_list_name_descriptor, bouquet_list_descriptor, and virtual_service_id_descriptor).

#### Alternatives:
Currently we use a combination of TSDuck and other tooling (e.g. from Tektronix), but we would prefer to use TSDuck fully for stream analysis and modification, as it works really well for scripting and testing automation.

#### Additional information:
We have obtained the consent of SES to share the latest relevant specification document publicly here, that you can find attached to this issue.

[ASTRA_LCN_v2_6.pdf](https://github.com/user-attachments/files/19958750/ASTRA_LCN_v2_6.pdf)

lelegard, 2025-04-29:
Hi @nicholas-fr 

This is feasible. It's just a matter of time and resources.

Could you provide a transport stream captured file which contains all the relevant signalisation? Using a real-world reference is always necessary to implement an existing signalisation.

TSDuck is free and open source and this will not change. Nonetheless, commercial companies which use TSDuck in their business operations are encouraged to [contribute](https://tsduck.io/donate/), especially when requesting new features. No obligation, fair participation only.

lelegard, 2025-05-02:
The Astra SGT and associated descriptors have been implemented in TSDuck.

A user from the [TSDuck Signal group](https://tsduck.io/signal/) kindly captured the transport stream id 1111 which contains the Astra SGT's on Astra 19.2 E. We can see that the SGT are correctly implemented in TSDuck.

However, as a side observation, I noted that Astra violates several DVB rules in some SGT tables. Since TP Vision produces receivers, you may encounter issues (or enforce some "bug-compatibility" mode).

The details are provided below. Since you are in contact with Astra, you may direct them to this issue for the details.

<hr/>

## Incorrect signalization in Astra SGT

The following analysis is performed on a capture of Astra 19.2 E, 12604H, on 2025-05-02. The captured file is available [here](https://tsduck.io/streams/?name=astra-sgt).

### Incorrect string encoding

ETSI EN 300 468 specifies that the default character set in DVB tables is ISO-6937. If characters which are not available in ISO-6937 are used, another character set shall be used. In that case, the first byte of the string is a character set selector.

In the Astra SGT, we see German characters such as 'Ö' which are not available in ISO-6937. The character set which is used appears to be ISO-8859-9. However, the initial selector byte is omitted (it should be 0x05). Therefore, a DVB-compliant receiver interprets it as ISO-6937 and the corresponding character, encoded as 0xD6, is interpreted as '¬' instead of 'Ö'.

DVB-compliant interpretation:
~~~
$ tstables astra-12604H.ts --tid 0x91 | grep 'Astra Bouquet List (0x93, 147), 11 bytes' -A 1 -m 1
      - Descriptor 3: Astra Bouquet List (0x93, 147), 11 bytes
        Bouquet name: "¬sterreich"
~~~

Using the option `--default-charset DUMP`, we dump the binary content of the string. We see that there is no initial selector byte:
~~~
$ tstables astra-12604H.ts --tid 0x91 --default-charset DUMP | grep 'Astra Bouquet List (0x93, 147), 11 bytes' -A 1 -m 1
      - Descriptor 3: Astra Bouquet List (0x93, 147), 11 bytes
        Bouquet name: "D6 73 74 65 72 72 65 69 63 68"
~~~

If we force the interpretation of all strings as ISO-8859-9 when no selector byte is present, we found the expected string:
~~~
$ tstables astra-12604H.ts --tid 0x91 --default-charset ISO-8859-9 | grep 'Astra Bouquet List (0x93, 147), 11 bytes' -A 1 -m 1
      - Descriptor 3: Astra Bouquet List (0x93, 147), 11 bytes
        Bouquet name: "Österreich"
~~~

### Missing private_data_specifier_descriptor

In some SGT, but not all, the private_data_specifier_descriptor with Astra PDS are missing.

In [this file](https://tsduck.io/streams/astra-sgt/astra-12604H-2025-05-02.sgt.txt), line 27, the SGT from PID 0x0776 has the correct private_data_specifier_descriptor:
~~~
    - Service id: 0x6D67 (28007), TS id: 0x0437 (1079), Network id: 0x0001 (1)
      LCN: 150, visible: true, new: false, genre code: 0x0000 (0)
      - Descriptor 0: Service (0x48, 72), 16 bytes
        Service type: 0x01 (Digital television service)
        Service: "3sat", Provider: "ZDFvision"
      - Descriptor 1: Private Data Specifier (0x5F, 95), 4 bytes
        Specifier: 0x00000001 (SES Astra)
      - Descriptor 2: Astra Bouquet List (0x93, 147), 8 bytes
        Bouquet name: "ARD/ZDF"
      - Descriptor 3: Astra Virtual Service Id (0xD1, 209), 2 bytes
        Virtual service id: 0x041A (1050)
~~~

However, in the SGT from PID 0x0777, line 7741 of the same text file, the private_data_specifier_descriptor is missing:
~~~
    - Service id: 0x6DCA (28106), TS id: 0x044D (1101), Network id: 0x0001 (1)
      LCN: 1, visible: true, new: false, genre code: 0x0000 (0)
      - Descriptor 0: Service (0x48, 72), 15 bytes
        Service type: 0x01 (Digital television service)
        Service: "Das Erste", Provider: "ARD"
      - Descriptor 1: Astra Virtual Service Id (0xD1, 209), 2 bytes
        Virtual service id: 0x6DCA (28106)
~~~

In the file, the astra_virtual_service_id_descriptor is still correctly interpreted because the analysis command used the option `--default-pds astra`, meaning "if a descriptor with a tag higher than 0x80 is found without previous private_data_specifier_descriptor in the descriptor list, then assume that the Astra PDS is used".

Without this option, using the default DVB behaviour, we get this:
~~~
    - Service id: 0x6DCA (28106), TS id: 0x044D (1101), Network id: 0x0001 (1)
      LCN: 1, visible: true, new: false, genre code: 0x0000 (0)
      - Descriptor 0: Service (0x48, 72), 15 bytes
        Service type: 0x01 (Digital television service)
        Service: "Das Erste", Provider: "ARD"
      - Descriptor 1: unknown (0xD1, 209), 2 bytes
        0000:  6D CA                                      m.
~~~

It shall be noted that, in the same section, there is a private_data_specifier_descriptor in the outer descriptor loop. However, it is missing in the inner service-specific descriptor loops. As specified in ETSI TS 101 211, section 4.2.7.1, "The scope of a private_data_specifier_descriptor in a descriptor loop does not propagate to any following descriptor loop or into any following section." This clearly specifies that the private_data_specifier_descriptor shall be repeated in all descriptor loops (this is different from the MPEG-defined registration_descriptor).

lelegard, 2025-05-08:
> Unfortunately, as you've probably seen, the 2 issues you mentioned are not uncommon in broadcast deployments.
> Our receivers use other criteria to ensure the correct character set and private data context are used.

Yes, this is not uncommon, far from that. This is why TSDuck contains so many options to force the interpretation of PDS, charsets, or other specific feature, when not correctly implemented in the stream. I would even say that Digital TV is a domain where the number of bugs, both on the head-end and receiver sides, are the highest in the industry. Much more than Internet and network protocols for instance. They have their bugs and pitfalls, but not as many as in Digital TV. This is why a debugging environment such as TSDuck can be useful.

lelegard, 2025-05-09:
Looking closely at the SGT, there are other problems in addition to incorrect character sets or missing private data specifier descriptors.

In the service list id 1 ("Channel List Germany 19,2°E") on PID 0x0777, we see repeated identical descriptors in several services, for instance this one:

~~~
<service service_id="0x33F5" transport_stream_id="0x03ED" original_network_id="0x0001" logical_channel_number="1,116" visible_service_flag="true" new_service_flag="false" genre_code="0x0000">
  <service_descriptor service_type="0x19" service_provider_name="ORF" service_name="ORF2St HD"/>
  <CA_identifier_descriptor>
    <CA_system_id value="0x1762"/>
    <CA_system_id value="0x0D05"/>
    <CA_system_id value="0x1801"/>
  </CA_identifier_descriptor>
  <astra_virtual_service_id_descriptor virtual_service_id="0x33F5"/>
  <service_descriptor service_type="0x19" service_provider_name="ORF" service_name="ORF2St HD"/>
  <CA_identifier_descriptor>
    <CA_system_id value="0x1762"/>
    <CA_system_id value="0x0D05"/>
    <CA_system_id value="0x1801"/>
  </CA_identifier_descriptor>
  <astra_virtual_service_id_descriptor virtual_service_id="0x33F5"/>
</service>
~~~

The same suite of 3 descriptors is repeated twice. Several other services have the same problem, but not all.

I don't know which tool is used to generate it but I think it is time for Astra to maintain a list of clean SGT in XML files and use TSDuck to generate it 😉 

The character sets will always be correct and the recent option `--fix-missing-pds` ensures that the required `<private_data_specifier_descriptor>` are present (because `<astra_virtual_service_id_descriptor>` is known to require an Astra PDS).