tsduck/tsduck #1626: [Q] How to replace an audio stream in a video TS segment

parmy, 2025-05-04:
Hello

For a client, I hope to work directly with their Video TS Streams. The steps are:

1. Using ffprobe count the number of audio streams ( they typically have 4, we're only concerned with the first )
2. Using ffmpeg extract the primary audio stream PID 257, in wav format
3. Encode and upscale the audio. Convert this encoded wav using ffmpeg into a Interim Audio TS file
4. Use TSDuck to replace with Audio Stream PID 257 in the Original VIdeo TS file with the Interim Audio TS File

Simplied:

```
Stream #0:0[0x100]: Video: h264
Stream #0:1[0x101]: Audio: aac (LC), 48000 Hz, stereo  <==== [I want to replace this]
Stream #0:2[0x102]: Audio: aac (LC), 48000 Hz, stereo
Stream #0:3[0x103]: Audio: aac (LC), 48000 Hz, stereo
Stream #0:4[0x104]: Audio: aac (LC), 48000 Hz, stereo
```

I tried with FFmpeg, however a number of issues occur:

1. It replaces all Audio Streams in original TS
2. Even if I work around point 1, the Duration, Start and Service-Provider meta data are updated

My first attempt was to work with the RAW file directly and replace the Audio Packets in Original.ts with InterimAudio.ts. I end up with one of two issues:

1. Duration inflates
2. If I fix the Duration issue, the Audio Stream is not valid

From my understanding TSDuck allows fine grain control. I've tried all manner of commands - I've provided those below.

I've attached three files:

- original.ts - the oringal TS segment where I'm trying to replace the audio stream
- input.wav - wav format of the audio stream I'm trying to insert back into original.ts
- audio.ts - ts format  of the audio stream I'm trying to insert back into original.ts

I would be extremely grateful for any assistance. 

Warm regards, Parmy

---
TSDuck commands I have tried:

**Attempt 1: Basic Merge with Interleave (First Attempt)**
```
tsp -I file original.ts --interleave audio.ts --label-base 1 -P psimerge --main-label 1 --merge-label 2 -O file final_muxed.ts
```

Outcome: Error with service conflict, service 0x0001 (1) exists in both streams, so the audio stream was dropped.

**Attempt 2: Merge with Remap PID on Audio Stream**
```
tsp -I file original.ts -P remap 257=261 -P pat -P pmt -O file final_muxed.ts
```
Outcome: Did not produce the expected results.

**Attempt 3: Merge with Audio Remap in Another Process**
```
tsp -v -I file original.ts -P merge "tsp -I file audio.ts -P remap 257=261 -O file -" -O file final_muxed.ts
```
Outcome: Streams were not merged as expected.

**Attempt 4: Interleave with Audio Remapped Stream**
```
tsp -I file original.ts --interleave audio_remapped.ts --label-base 1 -P psimerge --main-label 1 --merge-label 2 -O file final_muxed.ts
```
Outcome: Service conflict, audio stream dropped due to service ID conflict.

**Attempt 5: Using modify Plugin for Service ID Adjustments**
```
tsp -I file audio.ts -P modify --service-id 2 -O file audio_adjusted.ts
```
Outcome: Error with "processor plugin modify not found."

**Attempt 6: Using psimultiplex Plugin (Not Found)**
```
tsp -I file original.ts --interleave audio_remapped.ts --label-base 1 -P psimultiplex --main-label 1 --merge-label 2 -O file final_muxed.ts
```
Outcome: Error with psimultiplex.dylib not found.

**Attempt 7: Filter and Interleave with Remapped Audio**
```
tsp -I file filtered_original.ts --interleave audio_remapped.ts --label-base 1 -P psimerge --main-label 1 --merge-label 2 -O file final_muxed.ts
```
Outcome: Still service conflict issues and audio dropped.

**Attempt 8: Remap Audio, Filter Video, and Interleave**
```
tsp -I file audio.ts -P remap 1=2 -O file audio_remapped.ts
tsp -I file filtered_original.ts --interleave audio_remapped.ts --label-base 1 -P psimerge --main-label 1 --merge-label 2 -O file final_muxed.ts
```
Outcome: Service conflict, still losing the audio stream.

**Attempt 9: Direct Muxing with Interleaving**
```
tsp -I file original.ts --interleave audio_remapped.ts --label-base 1 -O file final_muxed.ts
```
Outcome: Empty muxed file with no data encoded (0 KB output).

**Attempt 10: Merging Without PID Remap**
```
tsp -I file filtered_video.ts -P merge "tsp -I file audio.ts -O file -" -O file final_muxed.ts
```
Outcome: Error with audio not merged as expected.

**Attempt 11: Encoding with FFmpeg (Timestamp Issues)**
```
ffmpeg -i final_muxed.ts -c:v libx264 -c:a aac -ss 59799.467000 -t 00:00:02.00 -y final_muxed_corrected.ts
```
Outcome: Errors due to timestamp discontinuities and no successful encoding (empty file output).

**Attempt 12: FFmpeg Output After Timestamp Adjustments**
```
ffmpeg -i final_muxed.ts -c:v libx264 -c:a aac -ss 59799.467000 -t 00:00:02.00 -y final_muxed_corrected.ts
```
Outcome: Still empty output with timestamp discontinuities.

**Key Observations:**
Service Conflict: A recurring issue with services conflicting across streams, causing streams to be dropped.
Unmodifed Files: Merge had no impact
Timestamp Issues: In the case of FFmpeg encoding, timestamp discontinuities led to invalid outputs.

[Archive.zip](https://github.com/user-attachments/files/20025708/Archive.zip)

lelegard, 2025-05-04:
Hi @parmy

Replacing/injecting an audio track is a muxing task and TSDuck is not a MUX. You need a MUX to do this. FFmpeg can do this. If you already use ffmpeg, do the muxing in the ffmpeg process.

TSDuck is transport stream processing tool. `-I file --interleave` just picks N TS packets on each input file in a round-robin fashion. `-P merge` replaces null packets from the main stream with packets from the merged stream, when available. None if these is muxing.

Differences between "merging" and "muxing":
- Audio/video muxing means mixing video and audio packets from the same service on a time stamp basis, based on a unique PCR, restamping PTS and DTS when necessary, regardless of TS packets and free space in the stream.
- Merging two TS with TSDuck means mixing distinct services coming from distinct TS into one TS containing all services from the two TS. Each service needs to be already correctly muxed in it own TS and this intra-service muxing won't be changed by TSDuck. There is no precise timing synchronization between distinct services. They use distinct PCR.

In short, ffmpeg is very good at audio/video processing, transcoding and muxing, but rather bad at transport stream and signalization processing. TSDuck is just the opposite. That is why the two are needed together to produce a well crafted transport stream. However, each one shall be used for what it is good at, and muxing is ffmpeg's job.

Regarding the issues you encountered with ffmpeg commands, I recommend to ask the question in some ffmpeg forum. FFmpeg is an extremely poweful tool but its commands can become complicated for sophisticated processing (arguably, the same remark applies to TSDuck too). So you need to ask the right experts.

> Outcome: Error with "processor plugin modify not found."
> Outcome: Error with psimultiplex.dylib not found.

Why did you think that these plugins existed? Please read the [TSDuck user's guide](https://tsduck.io/download/docs/tsduck.html) which contains a list of all available plugins and their reference documentation.

The first step is to precisely describe your input and the required output. You described processing steps without describing the required output. Therefore, we do not know if those processing steps are the right ones for your required output. And they probably aren't since you do not succeed in getting what you want. Describing input and output also means describing the required signalization in terms of services and additional metadata (if any).

parmy, 2025-05-04:
Hi @lelegard 

I'm extremely grateful that you took time out to provide a details response, thank you very much. Thanks for setting me straight on the purpose of TSDuck and wasting anymore time chasing my tail on that.

My bad on the following:

> Outcome: Error with "processor plugin modify not found."
> Outcome: Error with psimultiplex.dylib not found.

I either got confused reading a forum or mixed up commends from different tools. 

Your excellent feedback refocused my efforts with ffmpeg ( which I had given up on ) and I've managed to do exactly what I was after, create a clone of the original TS segment and replacing the single primary audio stream.

Thanks very much again,

Parmy

closed 2025-05-04 by parmy