tsduck/tsduck #1628: [FEAT] DVB-T U1 support

Sicand, 2025-05-07:
Hi , some year ago i bought this modulator but work only whit his program on windows (blocked at 5Mhz Bandwidht) and the driver gave me a com port on windows.

Recently i seen Vatek made a U3 Modulator , and my question is this : is it supported and is a previuos version of vatek chip modulator ? 

Escuse me if is not directly related to TSduck

[image]

lelegard, 2025-05-07:
Hi @Sicand 

I don't know this modulator. There is no brand. Where does it come from?

> Recently i seen Vatek made a U3 Modulator , and my question is this : is it supported and is a previuos version of vatek chip modulator ?

U3 is the name of a modulator device by [Suntechtv](https://www.suntechtv.com/h/ProductDetail?key=e593s&productId=34917). VATek produces the chip which is inside. As far as I remember, the VATek chip name is A20.

So, I do not know if there is any relationship between your U1 and this U3, but if there is one, it must be the reference of a Suntechtv device, based on the naming scheme. There is no guarantee that the chip was a VATek one. When I was in contact with VATek, the API that they developed was specifically for the A20. I am not aware of a previous modulator chip they would have made.

Sicand, 2025-05-20:
Hi , escuse me for the time; lusb gave me this ID 040b:0101 Weltrend Semiconductor . On windows is a com port and whit their program it transmit at 6mhz bandwidth

lelegard, 2025-05-21:
The U1 is unknown on the Weltrend Semiconductor web site. All their chips are apparently named WTxxx. Your chip must be some experimental one, maybe an internal proof of concept. I doubt that anyone with the appropriate knowledge of that chip will invest time to support it in TSDuck. You may check with the people you got that device from.

Sicand, 2025-05-21:
I think it go in trashcan  or i'll search in my hard disk archive if find something old than ten year . I will update . Thanks for your time :)