tsduck/tsduck #1629: [Q] Need help with message : tsp: buffer failed to lock into physical memory (12: Cannot allocate memory), risk of real-time issue

cgourves, 2025-05-13:
Hello,
I'm using TSDuck on automatic tests, I upgraded from 3.35-3258 to 3.40-4165 and I get some issues : 
I typically run this kind of command, it was working perfectly on 3.35 version : 
 tsp -I ip 239.4.14.7:5000 -P until --seconds 2 -P analyze --normalized -O drop 
Since the upgrade, this always returns bitrates with values 0, which is wrong.
I tried to add option debug -d2, and I get this message : 
* Debug: tsp: buffer failed to lock into physical memory (12: Cannot allocate memory), risk of real-time issue
Can this be a reason for the bitrates = 0 ?
I run tsp on a Ubuntu 24.04 VM.
Tell me if you need more informations
Thanks,
Cyrille

cgourves, 2025-05-13:
Sorry I missed something : I run tsp on a Ubuntu docker container running on a Debian VM

lelegard, 2025-05-13:
Hi @cgourves 

The message about "failed to lock into physical memory" is just for debug and completely harmless. TSDuck tries to lock the global buffer into physical memory to avoid paging/swapping. This is relevant only on systems with very small memory. On Linux, this is allowed only if you are root. So, this message has probably always been present in your case.

What kind of bitrates are zero? All of them? Some of them?

What is the rest of the analysis? Could you post the output of `-P analyze --normalized`?

The plugin `analyze` computes bitrates based on PCR from the stream. Do you have enough PCR? Is there any change in the input stream that can explain the difference?

cgourves, 2025-05-13:
Sure, this is the output : 
`tsduck@tm-testtoolbox-vm-191:/$ tsp -I ip 239.4.14.7:5000 -P until --seconds 10 -P analyze --normalized -O drop
title:
ts:id=0:services=1:clearservices=1:scrambledservices=0:pids=7:clearpids=7:scrambledpids=0:pcrpids=1:unreferencedpids=0:packets=398952:invalidsyncs=0:transporterrors=0:suspectignored=0:bytes=75002976:bitrate=0:bitrate204=0:userbitrate=0:userbitrate204=0:pcrbitrate=0:pcrbitrate204=0:duration=0:
time:utc:system:first:date=13/05/2025:time=08h21m42s:secondsince2000=800439702:
time:utc:system:last:date=13/05/2025:time=08h21m52s:secondsince2000=800439712:
time:local:system:first:date=13/05/2025:time=08h21m42s:secondsince2000=800439702:
time:local:system:last:date=13/05/2025:time=08h21m52s:secondsince2000=800439712:
global:pids=3:clearpids=3:scrambledpids=0:packets=184923:bitrate=0:bitrate204=0:access=clear:pidlist=0,8187,8191:
unreferenced:pids=0:clearpids=0:scrambledpids=0:packets=0:bitrate=0:bitrate204=0:access=clear:pidlist=:
service:id=1:tsid=0:access=clear:pids=4:clearpids=4:scrambledpids=0:packets=214029:bitrate=0:bitrate204=0:servtype=0:pmtpid=48:pcrpid=49:pidlist=48,49,52,53:provider=ATSC 2.1:name=eas_rpl
pid:pid=0:access=clear:servcount=0:global:bitrate=0:bitrate204=0:packets=224:clear=224:scrambled=0:invalidscrambling=0:af=0:pcr=0:pts=0:dts=0:pcrleap=0:ptsleap=0:dtsleap=0:discontinuities=0:duplicated=112:unitstart=224:description=PAT
pid:pid=48:pmt:access=clear:servcount=1:servlist=1:bitrate=0:bitrate204=0:packets=56:clear=56:scrambled=0:invalidscrambling=0:af=0:pcr=0:pts=0:dts=0:pcrleap=0:ptsleap=0:dtsleap=0:discontinuities=0:duplicated=28:unitstart=56:description=PMT
pid:pid=49:access=clear:streamid=224:video:servcount=1:servlist=1:bitrate=0:bitrate204=0:packets=206989:clear=206989:scrambled=0:invalidscrambling=0:af=1196:pcr=598:pts=602:dts=198:pcrleap=0:ptsleap=0:dtsleap=0:discontinuities=28431:duplicated=47:pes=602:invalidpesprefix=0:firstpcr=1202216127525:lastpcr=1202484594038:firstpts=4007424185:lastpts=4008328088:firstdts=4007415176:lastdts=4008316076:description=MPEG-2 Video (1920x1080i, @25 Hz, 16:9, 4:2:0)
pid:pid=52:access=clear:streamid=189:audio:language=eng:servcount=1:servlist=1:bitrate=0:bitrate204=0:packets=5234:clear=5234:scrambled=0:invalidscrambling=0:af=78:pcr=0:pts=78:dts=0:pcrleap=0:ptsleap=0:dtsleap=0:discontinuities=28:duplicated=2561:pes=78:invalidpesprefix=0:firstpts=4007389737:lastpts=4008265257:description=ATSC AC-3 Audio (eng, AC-3, 3/2 (L,C,R,SL,SR), @48,000 Hz, complete main, bsid 6)
pid:pid=53:access=clear:streamid=189:audio:language=spa:servcount=1:servlist=1:bitrate=0:bitrate204=0:packets=1750:clear=1750:scrambled=0:invalidscrambling=0:af=124:pcr=0:pts=124:dts=0:pcrleap=0:ptsleap=0:dtsleap=0:discontinuities=1:duplicated=873:pes=124:invalidpesprefix=0:firstpts=4007397788:lastpts=4008276188:description=ATSC AC-3 Audio (spa, AC-3, mono, @48,000 Hz, complete main, bsid 6)
pid:pid=8187:access=clear:servcount=0:global:bitrate=0:bitrate204=0:packets=242:clear=242:scrambled=0:invalidscrambling=0:af=0:pcr=0:pts=0:dts=0:pcrleap=0:ptsleap=0:dtsleap=0:discontinuities=0:duplicated=121:unitstart=242:description=ATSC PSIP (ATSC TVCT-current)
pid:pid=8191:access=clear:servcount=0:global:bitrate=0:bitrate204=0:packets=184457:clear=184457:scrambled=0:invalidscrambling=0:af=0:pcr=0:pts=0:dts=0:pcrleap=0:ptsleap=0:dtsleap=0:discontinuities=0:duplicated=0:unitstart=0:description=Stuffing
table:pid=0:tid=0:tidext=0:tables=112:sections=112:repetitionpkt=3591:minrepetitionpkt=3586:maxrepetitionpkt=3595:firstversion=1:lastversion=1:versions=1:
table:pid=48:tid=2:tidext=1:tables=27:sections=27:repetitionpkt=14362:minrepetitionpkt=14356:maxrepetitionpkt=14363:firstversion=0:lastversion=0:versions=0:
table:pid=8187:tid=199:tidext=0:tables=77:sections=77:repetitionpkt=5186:minrepetitionpkt=5149:maxrepetitionpkt=5191:firstversion=13:lastversion=13:versions=13:
table:pid=8187:tid=200:tidext=0:tables=33:sections=33:repetitionpkt=11968:minrepetitionpkt=11961:maxrepetitionpkt=11970:firstversion=12:lastversion=12:versions=12:
table:pid=8187:tid=205:tidext=0:tables=11:sections=11:repetitionpkt=35904:minrepetitionpkt=35899:maxrepetitionpkt=35908:firstversion=0:lastversion=0:versions=0:
`

cgourves, 2025-05-13:
on the same stream, bitrates are not equal to 0 when I run the command on version 3.35 : 
pid:pid=48:pmt:access=clear:servcount=1:servlist=1:bitrate=4211:bitrate204=4569:packets=28:clear=28:scrambled=0:invalidscrambling=0:af=0:pcr=0:pts=0:dts=0:pcrleap=0:ptsleap=0:dtsleap=0:discontinuities=0:duplicated=0:unitstart=28:description=PMT

cgourves, 2025-05-13:
Sorry it's probably a hardware issue on my side, the 2 versions 3.40 and 3.35 are running on different hosts

lelegard, 2025-05-13:
This is due to bad reception. There are too many lost packets to compute the bitrate.

Look at the description of the video PID:
~~~
pid:pid=49:access=clear:streamid=224:video:servcount=1:servlist=1:bitrate=0:bitrate204=0:packets=206989:clear=206989:scrambled=0:invalidscrambling=0:af=1196:pcr=598:pts=602:dts=198:pcrleap=0:ptsleap=0:dtsleap=0:discontinuities=28431:duplicated=47:pes=602:invalidpesprefix=0:firstpcr=1202216127525:lastpcr=1202484594038:firstpts=4007424185:lastpts=4008328088:firstdts=4007415176:lastdts=4008316076:description=MPEG-2 Video (1920x1080i, @25 Hz, 16:9, 4:2:0)
~~~

And specifically these values in the line:
~~~
pcr=598
pcrleap=0
discontinuities=28431
~~~

The bitrate is computed by counting packets between 2 PCR values. A PCR value is a clock value, units of 27 MHz. The number of TS packets between 2 PCR values is a number of bits during a given duration, giving a bitrate. During analysis, an average is computed on all PID's containing PCR, giving the final bitrate value.

There are 598 PCR values, more than enough to compute a bitrate. No PCR "leap" means no abrupt change of PCR reference, with PCR values steadily increasing without bizarre leap.

However, there are 28431 discontinuities, meaning missing packets. When computing the bitrate between 2 PCR values, we need to be confident that all TS packets were counted between those 2 PCR. If at least one discontinuity is present, we know that packets are missing, but we don't know how many for sure. This means that the computed bitrate will be incorrect. So, we drop that interval. However, here we have an average of 47 discontinuities per interval. This means that we cannot find not even one single interval with a valid bitrate evaluation.

This explains the result.

cgourves, 2025-05-13:
Thanks a lot for this explanation, we will investigate on network issue then.

cgourves, 2025-05-13:
As mentionned, this issue seems to come from the container host; Might it be possible that an ubuntu24:04 image with the 3.40 tsduck release installed have issues due to the host?
The container host used are:
Debian10 Docker 20.10.21 Common KVM processor => Works like a charm
Debian12 Docker 27.3.1 AMD EPYC 7543P 32-Core Processor => Unknown bitrate returned

lelegard, 2025-05-13:
Hard to tell. Network issues are never the result of the choice of one OS or Docker version. Or even a CPU, especially a 32-core EPYC monster. It is a global problem that needs to be globally analyzed on your systems. There are a lot of network troubleshooting tools to analyze this. This is not a TSDuck issue that can be solved here.

The only time I got a similar problem was 20 years ago. An IP source injected UDP datagrams in a 100 Mbps network, which then went through a switch and then to a 10 Mbps network down to the receiver (multiply by 100 nowadays). When the source was too bursty, too many packets accumulated in the switch and some were dropped because of the 100->10 Mbps bottleneck. Smoothing the bursts at the source solved the issue. Just one example to show you that the problem can be anywhere on the network. We cannot help here.

cgourves, 2025-05-13:
Thanks for your help