tsduck/tsduck #1631: [Q] PID Values Change

wcarman1, 2025-05-16:
I'm trying to take in an SRT Feed and output it as multicast. I have it working but no matter what I do it doesn't carry over the PID values form the srt input. It assigns new PIDs no matter what I've tried.

The incoming SRT has the following PIDs and I'd like to retrain them on the output multicast:

|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Undefined .................................... C    5,339,762 b/s  |
|  0x0030  PMT .......................................... C       24,751 b/s  |
|  0x0031  AVC video (1280x720, main profile, level 3.2,  C    4,359,235 b/s  |
|  0x0034  ATSC AC-3 Audio (eng, AC-3, 3/2 (L,C,R,SL,SR)  C      552,829 b/s  |
|  0x0035  ATSC AC-3 Audio (spa, AC-3, stereo (L,R), @48  C      276,636 b/s  |
|  0x0036  MPEG-2 PES private data ...................... C      126,311 b/s  |
|          (C=Clear, S=Scrambled, +=Shared)                                   |

Here's the basic command I'm running (sanitized)

 /usr/bin/tsp
      -I srt ipaddress:port --passphrase **password**
      -P regulate
      -O ip 224.200.101.151:6000 --local-address 10.20.20.171 --ttl 10

Here is what the output looks like:

|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Digital television service ................... C    3,718,440 b/s  |
|  0x0100  AVC video (1280x720, main profile, level 3.2,  C    3,128,826 b/s  |
|  0x0101  ATSC AC-3 Audio (eng, AC-3, 3/2 (L,C,R,SL,SR)  C      569,487 b/s  |
|  0x1000  PMT .......................................... C       20,127 b/s  |
|          (C=Clear, S=Scrambled, +=Shared)                                   |

I've been searching for an answer for this all day, is this normal that the PID gets re-assigned? I was under the impression the PIDs would stick if they were present. It also seems to drop the second audio pid 35. 

Am I missing something? Any pointers would be greatly appreciated.

Thank you!

lelegard, 2025-05-16:
Hi @wcarman1 

The command you mention will never change anything inside the stream. Nothing. Even less doing complicated things such as remapping PID's.

The two analysis excerpts you provide do not describe the same stream (or not the same service in the stream). The PID's are different, the bitrates are different, the number of audio components are different.

The problem must be in the way you performed the analysis. Your analysed something else. The only thing I can guarantee is that the two analysis excerpts are _not_ the input and output of the command you included.

wcarman1, 2025-05-20:
@lelegard 
I got it figured out, It was the same stream but on the output side, for some reason, I could not use tsp to directly analyze the stream so I would do a tcpdump to a file and use ffmpeg to convert the dump file into a format that TSP could analyze and FFMPEG was also changing the PID Values. Now that I have the server running TSP (SRT in to Multicast out), I decided to put it in a network that I can test externally, and the PIDs do follow and I confirmed that on our StreamScope.
 
Still not sure why `tsp -I ip 224.200.101.151:6000 --local-address 10.20.20.171 -P until --seconds 10 -P analyze -O drop` won't analyze the output side. I'm not sure if it's something OS (Rocky 9) related or not. But it's working so that's good!!

closed 2025-05-20 by wcarman1

lelegard, 2025-05-20:
> FFMPEG was also changing the PID Values

OK, that explains everything. Since the input and output container formats may be different, ffmpeg drops any information on the structure of the input TS and builds a new TS structure from scratch on output. New PID's, new identifiers, new everything in the signalization. It also transcodes one audio stream only by default.

> I could not use tsp to directly analyze the stream so I would do a tcpdump to a file and use ffmpeg to convert the dump file into a format that TSP could analyze

TSDuck can read a UDP/IP transport stream directly from a pcap file. See the [pcap input plugin](https://tsduck.io/download/docs/tsduck.html#_pcap_input) and the [tspcap information utility](https://tsduck.io/download/docs/tsduck.html#_tspcap)

> Still not sure why tsp -I ip 224.200.101.151:6000 --local-address 10.20.20.171 -P until --seconds 10 -P analyze -O drop won't analyze the output side. 

It should work. Try with '-O file something.ts" to see if you receive something. Also use Wireshark on that interface to see if you get the packets or not.

wcarman1, 2025-05-20:
@lelegard That is all great info to know! I didn't realize FFMPEG would do all of that.

PCAP was also the way to go.

`tshark -i ens224 -f "udp and dst host 224.200.101.151 and port 6000" -a duration:10 -w capture.pcap`
`tsp -I pcap capture.pcap -P analyze -O drop`

Gave me an exact match input to output.

Thank you very much!!