tsduck/tsduck #1632: [Q] Problem in installing binaries or making tsduck on raspberry 2pi

Sicand, 2025-05-20:
Hi , i tried to install tsduck binaries for raspberry present in homepage but it can't be installable becouse require 2 64bit  packet instead of 32bit that are the binaries. 

So i tried to compile on my raspberry 2pi but it stop on this error :

 [CXX] zip/tsZlib.cpp
In file included from zip/tsZlib.cpp:58:
/media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/src/libtscore/zip/private/sinfl.h:417:15: error: comparison of integers of different signs: 'unsigned int' and 'int' [-Werror,-Wsign-compare]
  417 |       if (len > (e - s.bitptr))
      |           ~~~ ^  ~~~~~~~~~~~~
1 error generated.
make[2]: *** [../../Makefile.inc:66: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/objs-libtscore/tsZlib.o] Error 1
make[1]: *** [Makefile:14: default] Error 2
make: *** [Makefile:18: default] Errore 2

I used clang 20.1.0 becouse debian bookworm gcc is on version 12 , and used this command : make LLVM=1 NOPCSC=1 NODEKTEC=1
Escuse for my poor english

lelegard, 2025-05-20:
Hi @Sicand 

The code which creates the error (`sinfl.h`) is external to TSDuck. It is known to generate many warnings and the TSDuck source code which includes it mutes these warnings before including the code.

However, this warning was not reported. Specifically, I just tried with clang 20.1.3 for 64-bit Arm (in a Fedora 42 VM on a Mac) and this warning is not reported. This should be almost the same compiler as yours. I haven't seen it with previous versions of clang either.

Anyway, I added this warning in the list of warnings to mute before including this external code. I don't understand why you have it and I don't but it does not hurt to add it.

Please pull the repo and retry. Let us know if it compiles.

Independently from this issue, `NODEKTEC=1` is implicit on Arm targets because Dektec does not provide a binary API for Arm. Specifying `NODEKTEC=1` is not necessary in that case.

Sicand, 2025-05-20:
Thanks , i try late. The (mine) raspberry Pi2' CPU is an Arm32 . Whit Debian bookworm & Clang19 on a Intel 64bit processor i had no problem

lelegard, 2025-05-20:
> The (mine) raspberry Pi2' CPU is an Arm32 . 

Is the original problem on an Arm32 distro for Raspberry Pi? From your original post, I thought it was an Arm64 distro.

Sicand, 2025-05-20:
No , is an arm 32 distro ( pi2 is only 32bit) ; in fact, I wrote that I could not install your precompiled package because even if declared as 32 bit, it depends on 2 64 bit libraries (libssl64 and another one that I don't remember), for this reason I tried to compile a 32 bit version. When i end to work i return home to compile whit your modification.

lelegard, 2025-05-20:
>  even if declared as 32 bit, it depends on 2 64 bit libraries (libssl64 and another one that I don't remember), 

I see. You probably refer to dependencies libssl3t64 and libcurl4t64. They are NOT 64-bit libraries (simply because 64-bit libraries cannot be used on a 32-bit platform, just impossible). They are 32-bit libraries.

When a package name ends in "t64" (note the "t"), this means that this is a version which is compiled with time values stored on 64 bits instead of the traditional 32-bit UNIX time, which ends in Jan 2038. This is still 13 years ahead, but the transition will be long and needs to be started.

On tsduck.io, the 32-bit binaries for Raspberry Pi are, however, compiled for Raspbian 13 (trixie). This is why you get dependencies on these upgraded versions of some libraries, using enlarged time values.

Debian and Raspbian 12 come with obsolete compilers and are not supported. However, if you had installed a clang 20, it should work. I am just surprised by your compilation warning.

lelegard, 2025-05-20:
More details on "t64" libraries here: https://lwn.net/Articles/938149/

Sicand, 2025-05-20:
Next error

tv/epg/tsEITRepetitionProfile.cpp:79:61: error: implicit conversion loses integer precision: 'rep' (aka 'long long') to 'size_t' (aka 'unsigned int') [-Werror,-Wshorten-64-to-32]
   79 |     return EIT::SegmentToTableId(actual, prime_days.count() * EIT::SEGMENTS_PER_DAY);
      |            ~~~                           ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
dtv/epg/tsEITRepetitionProfile.cpp:84:53: error: implicit conversion loses integer precision: 'rep' (aka 'long long') to 'size_t' (aka 'unsigned int') [-Werror,-Wshorten-64-to-32]
   84 |     return EIT::SegmentToSection(prime_days.count() * EIT::SEGMENTS_PER_DAY);
      |            ~~~                   ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
2 errors generated.
make[2]: *** [../../Makefile.inc:66: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/objs-libtsduck/tsEITRepetitionProfile.o] Error 1
make[1]: *** [Makefile:14: default] Error 2
make: *** [Makefile:18: default] Errore 2

lelegard, 2025-05-21:
Hi,

On my Rasperry Pi 3 with 32-bit Raspbian 13 (trixie), I have gcc 14.2 and clang 19.1. I can reproduce that warning with clang 19. I can understand it, even though other 32-bit compilers, gcc on Arm32 and msvc on Win32, do not report it. I have fixed that warning and pushed the result.

_However_ (and this is a big "however"), I got a probable code generation issue with clang 19. One of the unitary tests fails with a bus error. The code is straighforward, linear, simple, local, single-threaded, and without error (link below). More suspect, if I add a debug trace message in the middle of the sequence, the bus error crash disappears. In such a simple sequence, in a single-threaded environment, it smells the code generation issue.

https://github.com/tsduck/tsduck/blob/e36c9592c7bfb05fa8e0d04a50523fb33c02fe24/src/utest/utestBuffer.cpp#L1889-L1895

Clang is usually a good to very good compiler. I use is on Intel x64 and Arm64, especially on macOS where it is the default compiler. However, I never used it on Arm32.

On 32-bit Raspbian, I only compile official versions, using gcc. I don't really use my Raspberry Pi 3 otherwise. Compiling TSDuck on it takes several hours, while it compiles in 2 minutes on my iMac M3. 

The other problem with Debian / Raspbian is the very long cycle time of versions, up to the point where the current version (Debian 12) mostly contains obsolete software and is not even able to compile C++20 code in 2025. This is why I migrated Raspbian and all my Debian VM's (Arm64 and x64) to trixie, future Debian 13.

If you want to keep Raspbian 12, I recommend to install [TSDuck 3.39](https://github.com/tsduck/tsduck/releases/tag/v3.39-3956). This is the last version which can be compiled on Raspbian 12 with the default compilers. You won't get any new feature but that's the rule of the game on a distro which is full of obsolete software.

Sicand, 2025-05-21:
Hi , i try for another Clang session  whit your correction but  if it stop again hower i wrote again here but try to port and compile whit gcc13/14 .  If  have other problem i use previous tsduck version . No problem on pc Bookworm based and last tsduck version. 

I'm blocked whit raspbian/debian 12 becouse pi2 is armv6 , pi3 is armv7 ;) .

Sicand, 2025-05-22:
bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
  [CXX] vatek/tsVatekControl.cpp
vatek/tsVatekControl.cpp:13:10: fatal error: 'vatek_sdk_device.h' file not found
   13 | #include <vatek_sdk_device.h>
      |          ^~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [../../Makefile.inc:66: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/objs-libtsduck/tsVatekControl.o] Error 1
make[1]: *** [Makefile:14: default] Error 2
make: *** [Makefile:18: default] Errore 2

Restarted whit NOVATEK=1

Now got this

 [CXX] tsanalyze.cpp
bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
  [LD] /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/tsanalyze
/usr/bin/ld: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/libtscore.so: undefined reference to `curl_multi_info_read'
/usr/bin/ld: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/libtscore.so: undefined reference to `curl_multi_wakeup'
/usr/bin/ld: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/libtscore.so: undefined reference to `curl_slist_append'
/usr/bin/ld: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/libtscore.so: undefined reference to `curl_multi_add_handle'
/usr/bin/ld: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/libtscore.so: undefined reference to `curl_multi_poll'
/usr/bin/ld: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/libtscore.so: undefined reference to `curl_multi_perform'
/usr/bin/ld: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/libtscore.so: undefined reference to `curl_multi_cleanup'
/usr/bin/ld: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/libtscore.so: undefined reference to `curl_easy_init'
/usr/bin/ld: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/libtscore.so: undefined reference to `curl_global_init'
/usr/bin/ld: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/libtscore.so: undefined reference to `curl_easy_cleanup'
/usr/bin/ld: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/libtscore.so: undefined reference to `curl_multi_remove_handle'
/usr/bin/ld: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/libtscore.so: undefined reference to `curl_multi_strerror'
/usr/bin/ld: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/libtscore.so: undefined reference to `curl_easy_setopt'
/usr/bin/ld: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/libtscore.so: undefined reference to `curl_slist_free_all'
/usr/bin/ld: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/libtscore.so: undefined reference to `curl_multi_init'
/usr/bin/ld: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/libtscore.so: undefined reference to `curl_version_info'
/usr/bin/ld: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/libtscore.so: undefined reference to `curl_easy_strerror'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [../../Makefile.inc:68: /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/bin/release-arm-raspberrypi-clang/tsanalyze] Error 1
make[1]: *** [Makefile:14: default] Error 2
make: *** [Makefile:18: default] Errore 2
andrea@raspberrypi:/media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck $ 

This whit GCC13 

 [CXX] crypto/tsAES128.cpp
In file included from /home/andrea/cmake-test/tools/native-pi-gcc-13.3.0-1/include/c++/13.3.0/semaphore:37,
                 from /home/andrea/cmake-test/tools/native-pi-gcc-13.3.0-1/include/c++/13.3.0/stop_token:39,
                 from /home/andrea/cmake-test/tools/native-pi-gcc-13.3.0-1/include/c++/13.3.0/thread:42,
                 from /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/src/libtscore/cpp/tsPlatform.h:749,
                 from /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/src/libtscore/types/tsUChar.h:16,
                 from /media/andrea/c7227d69-9a36-4c3e-a646-7c8b09ae4022/tsduck/src/libtscore/types/tsUString.h:15,
                 from crypto/tsBlockCipher.h:15,
                 from crypto/tsAES128.h:15,
                 from crypto/tsAES128.cpp:9:
/home/andrea/cmake-test/tools/native-pi-gcc-13.3.0-1/include/c++/13.3.0/bits/semaphore_base.h:59:41: error: '_POSIX_SEM_VALUE_MAX' was not declared in this scope; did you mean '_SC_SEM_VALUE_MAX'?
   59 |     static constexpr ptrdiff_t _S_max = _POSIX_SEM_VALUE_MAX;
      |                                         ^~~~~~~~~~~~~~~~~~~~
      |                                         _SC_SEM_VALUE_MAX

lelegard, 2025-05-22:
Curl is probably not installed on your system:

1. Downloading the VATek lib fails because it's done using curl => 'vatek_sdk_device.h' file not found
2. All HTTP code in TSDuck calls libcurl => undefined reference to curl_XXX' on link.

Did you run `scripts/install-prerequisites.sh` ?

Sicand, 2025-05-24:
Yes , executed first install-prerequisite.sh script . 
Next i manually installed 2 libcurl cpp wrapper (find with synaptic) and tried to make all with clang 21 installed and now my raspberry pi 2 have last version of tsduck (made successful) . 

With gcc 14.2.0 make stop at this
'_POSIX_SEM_VALUE_MAX' was not declared in this scope; did you mean '_SC_SEM_VALUE_MAX'?
59 | static constexpr ptrdiff_t _S_max = _POSIX_SEM_VALUE_MAX;
| ^~~~~~~~~~~~~~~~~~~~
| _SC_SEM_VALUE_MAX

lelegard, 2025-05-24:
I am glad that you succeeded to build with clang 21.

Your error with gcc 14.2.0 on Raspbian 12 is weird and is probably a configuration problem of the compiler (I assume it is a custom build). On Raspbian 13 trixie, using gcc 14.2.0-19+rpi1, which is the default compiler, it builds correctly.

Sicand, 2025-05-25:
I'm on standard raspberry image , only installed compiled versione of updated compilers.
I open a issue on GCC raspberry compiler GitHub page , i wait now but i think is a standard problem

lelegard, 2025-05-26:
What do you mean by a "standard problem"? Is it a "common problem" or a "problem with the C++20 standard"?

The build works with GCC 13, 14, and 15 on all other Linux distros. The error is clearly some misconfiguration of the GCC 14 package for Raspbian 12. So, you are right to report the issue to the GCC packaging team.

lelegard, 2025-05-28:
> Anyone kwon a way tò pass the -std=c++20 tò GCC in make command ? I compiled via clang but i want to test GCC now

There is no general response. It depends on the makefiles and the build system for a given project. With TSDuck, C++20 is already the default. To specify another language level, use `make CXXFLAGS_STANDARD=-std=c++23` for instance. However, this is specific to the TSDuck makefiles, this is not applicable to another project.

Sicand, 2025-05-29:
Thanks for your support, tried but nothing , now i will compile myself gcc . I think this "question" can be closed . Only one question : the compiled version report an erroneous alert of cpu limit (150% instead of a real 50% CPU used) when use hides's output. Any option to silent the alert ?

lelegard, 2025-05-30:
> the compiled version report an erroneous alert of cpu limit (150% instead of a real 50% CPU used) when use hides's output.

What do you mean by that? Is this tsp displaying it? What is the exact message? Do you use option `tsp --monitor` ?

Keep in mind that CPU percentage has two distinct meanings, depending on tools which use it. It can be:
- A percentage of the total CPU power of the system. Hence, this is never higher than 100%.
- A percentage of the CPU power of one core. On a 4-core system, the value can go up to 400%. If a process is multithreaded, as `tsp` is, when several threads of that process simultaneously run on distinct cores, the CPU usage of the process can be higher than 100%.

Sicand, 2025-05-31:
This is tsp command :
`tsp  --buffer-size-mb 16 --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 --realtime -m --ignore-joint-termination --bitrate 6032085 --control-port 4445 \
-I null \
-P regulate --bitrate 6032085 --packet-burst 14 \
-P merge --acceleration-threshold 4 --restart --incremental-pcr-restamp  "tsp --realtime --buffer-size-mb 2 --initial-input-packets 4 --max-input-packets 4 --max-flushed-packets 4 -I ip :4001" \
-P filter -n -p 0 -p 17 -p 100 -p 4096 --stuffing \
-P merge --acceleration-threshold 4 --restart --incremental-pcr-restamp  "tsp --realtime --buffer-size-mb 2 --initial-input-packets 4 --max-input-packets 4 --max-flushed-packets 4 -I ip :4002" \
-P filter -n -p 0 -p 17 -p 100 -p 4096 --stuffing \
-P merge --acceleration-threshold 4 --restart --incremental-pcr-restamp "tsp --realtime --buffer-size-mb 2 --initial-input-packets 4 --max-input-packets 4 --max-flushed-packets 8 -I ip :4003" \
-P filter -n -p 0 -p 17 -p 100 -p 4096 --stuffing \
-P merge --acceleration-threshold 4 --restart --incremental-pcr-restamp  "tsp --realtime --buffer-size-mb 2 --initial-input-packets 4 --max-input-packets 4 --max-flushed-packets 4 -I ip :4004" \
-P filter -n -p 0 -p 17 -p 100 -p 4096 --stuffing \
-P merge --acceleration-threshold 4 --restart --incremental-pcr-restamp  "tsp --realtime --buffer-size-mb 2 --initial-input-packets 4 --max-input-packets 4 --max-flushed-packets 4 -I ip :4005" \
-P filter -n -p 0 -p 11 -p 17 -p 100 -p 4096 -p 4097 --stuffing \
-P duplicate 259=359 \
-P mux --bitrate 250000 ts/service.ts \
-P filter -n -p 0 -p 17 -p 100 -p 4096 --stuffing \
-P inject pat.xml --pid 0 --bitrate 4500 \
-P inject sdt.xml --pid 17 --bitrate 10000 \
-P inject nit.xml --pid 16 --bitrate 3000 \
-P inject tdt.xml=1000 tot.xml=15000 --pid 20 --bitrate 3000 \
-P inject pmt-street1.xml --pid 100 --bitrate 15000 \
-P inject pmt-street2.xml --pid 200 --bitrate 15000 \
-P inject pmt-street3.xml --pid 300 --bitrate 15000 \
-P inject pmt-street4.xml --pid 110 --bitrate 15000 \
-P inject pmt-street5.xml --pid 120 --bitrate 15000 \
-P inject pmt-street6.xml --pid 130 --bitrate 15000 \
-P inject pmt-street7.xml --pid 86 --bitrate 15000 \
-P inject pmt-street9.xml --pid 201 --bitrate 15000 \
-P inject pmt-street10.xml --pid 140 --bitrate 15000 \
-P inject pmt-street12.xml --pid 111 --bitrate 15000 \
-P inject pmt-street13.xml --pid 112 --bitrate 15000 \
-P timeref --start system \
-P eitinject --file 'epgtemp/*.xml' --delete-files --pf --terrestrial --europe \
-P continuity -f --no-replicate-duplicated \
-O hides -c QPSK -f 474000000 -h 1/2 -g 1/32 -t 8k`

[image]

lelegard, 2025-05-31:
The display is done by the monitor option `-m`:
~~~
  -m[filename]
  --monitor[=filename]
      Continuously monitor the system resources which are used by tsp. This
      includes CPU load, virtual memory usage. Useful to verify the stability
      of the application. The optional file is an XML monitoring configuration
      file.
~~~

The XML file defines the monitoring profile and alarm thresholds. It is documented in the TSDuck user's guide [appendix C](https://tsduck.io/download/docs/tsduck.html#chap-monconfig). Note that it mentions _"The max_cpu attribute indicates the maximum CPU percentage that the TSDuck application should use during a monitoring interval. If the application CPU usage exceeds this value, an alarm is triggered. Keep in mind that multi-threaded applications such as tsp or tsswitch, running on multi-core systems, may use more than 100% of a CPU."_

The default file is `/usr/share/tsduck/tsduck.monitor.xml`.

Keep in mind that the objective of monitoring is not to make warning messages disappear (in which case you can simply remove the monitoring options), it is to be notified of relevant information. Your command is relatively complex, with 34 plugins, resulting in 35+ threads inside the process, and your RPi 2 computer is excessively small. Therefore, 150% CPU is probably valid in this context. With a high-end Arm processor such as Apple M3 or M4, it would probably remain below 5%. If you want to monitor this command on a RPi 2 and get valid alarms, use a larger threshold in a custom monitoring configuration file.

closed 2025-11-02 by github-actions[bot]