tsduck/tsduck #1633: [Q] Trying to insert PSI data into video and convert to .ts

SixGunx86, 2025-05-23:
I will preface this by saying I am a network engineer first and am nothing but a noob with any form of development or encoding. This is for my job and my developers are all swamped. I was hoping to get this task taken care of. I will also be honest about the fact that I have been trying to use chatgpt for this issue and I'm not sure it has lead me down the right path.

I am trying to turn a .mp4 video into a .ts that has PSI data. It is going to be streamed via multicast from a server to a modulator that will turn it into QAM RF. The modulator is from a 3rd party that allows limited access to us to configure but it always complains about "No PSI Data".

I have already used ffmpeg to convert the .mp4 into a mpegts .ts file.

Chatgpt has been trying to get me to create a tables.xml file in order to use tstabcomp to compile a tables binary and then inject it with tsp to the .ts using "tsp -I file output.ts -P inject tables.bin -O file final.ts".

I just need a basic minimal PSI table config for this. Here is what chatgpt currently has me trying as a tables.xml:

<tsduck>
  <pat transport_stream_id="1">
    <program program_number="100" pid="256"/>
  </pat>
  <pmt program_number="100" pcr_pid="256">
    <elementary_stream pid="256" type="0x1B"/>
  </pmt>
</tsduck>

I have tried multiple variations of this xml and always get errors and tstabcomp wont compile the bin.

Any help would be greatly appreciated.

lelegard, 2025-05-23:
Hi @SixGunx86 

It is hard to say what "No PSI Data" means to your modulator. There are minimal PSI in ffmpeg output with mpegts format. Otherwise, the TS wouldn't be usable at all.

FFmpeg only outputs some minimal forms of PSI. On the other hand, TSDuck can tweak every single details of the signalization but this should be necessary only for fine tuning of the broadcast.

The only thing that maybe your modulator needs in addition to ffmpeg's default PSI is the NIT. You may force it using ffmpeg's output option `-mpegts_flags nit`.

If you want to precisely control ffmpeg's output PSI, I would recommend the following set of options:
~~~
-mpegts_flags nit
-mpegts_service_id 2
-mpegts_transport_stream_id 3
-mpegts_original_network_id 4
-metadata service_name="Foo"
-metadata service_provider="Bar"
-muxrate 5000000
~~~

Note that the id values 2, 3, 4 and the names are only examples. Use your own values. The `-muxrate` is required only if you want a precise constant bitrate, which is usually required by a modulator.

No need for TSDuck so far.

If you need more PSI tweaks, then TSDuck will definitely help you. But some proficiency with the MPEG and DVB standards are required. No need to use `tstabcomp`, as ChatGPT suggests, apparently stupidly copying an example from the Web site which is not relevant to your case. You can directly inject XML files. You can also patch existing PSI on the fly without injecting a new one. However, you need to precisely understand what you need, hence the required proficiency with the standards.

<hr/>

More generally, I _strongly_ suggest to never use ChatGPT - or any similar Artificial Idiot - for that kind of problem. All answers are incorrect, sometimes completely insane, 100% of the time. The more precise a question is, the more incorrect the response is. I verified this 100% of the time. When you ask a generic or vague question, you get a generic and vague answer which impresses lots of people, just because it is written in plain human language with correct sentences, instead of the useable computer's mishmash. However, when you need a precise answer, they are useless or even counter-productive.

Each time a user has asked ChatGPT, DeepSeek, etc. a question about TSDuck, the answer has always been wrong. Always. Using Natural Intelligence (aka "brain"), people reads the documentation, they _learn_ something, they improve, they find solutions which work. For this reason, I am seriously considering stopping responding to AI-generated problems (meaning standard problems which turned into other problems just because some AI produced an incorrect answer).

SixGunx86, 2025-05-23:
Thanks for the info and for helping someone with my limited knowledge. I'd like to also tell you that I'm not actually using ffmpeg to create the stream. We are using an in house solution to stream but my devs tell me its based on something open source out there. They tell me it should be able to handle passing psi data but I'm not entirely sure that's the case.

I am just using ffmpeg to convert the video to a ts. When I try doing anything with mpegts arguments, I get ts invalid errors.

I have done some more research and it may be possible that the source video has codecs not suitable for ts. I may have to convert it to h264 and aac to test this when I get a chance.

Is there anyway you could show me what a simple ffmpeg command to convert an MP4 to TS and preserve psi data looks like?

lelegard, 2025-05-23:
Ffmpeg is very verbose. What looks like errors may not be errors in fact. When using a TS file which a fragment of broadcast, the start of the file is typically in the middle of a frame and ffmpeg reports errors. But you may safely ignore them. 

If you get real errors, make sure you have the latest version, which must be 7.1.something. 

Finally, ffmpeg never _preserves_ the PSI. Ffmpeg works on audio and video but drops the input container (the TS). On output, it recreates a new container from scratch. The options I mentioned are used to configure that output container.

closed 2025-11-02 by github-actions[bot]