tsduck/tsduck #1634: [Q] Injecting TDT doesn't work

mkucmpro, 2025-05-28:
I have a TS file generated with FFmpeg. I want to make it DVB-compatible, therefore I'm trying to use `tsp` to inject the last missing table, which is TDT.

I tried:
`tsp --debug=4 -I file TSTEST.ts -P inject tdt.xml=100 --pid 20 --stuffing -O file TSTEST_with_tdt.ts`

But the resulting output file doesn't have the TDT in it.

The standard output I get is:
```
tsp: Debug: ====> tsp --debug=4
* Debug: file: ====> tsp -I file TSTEST.ts
* Debug: file: ====> tsp -O file TSTEST_with_tdt.ts
* Debug: trying to load "C:\TSDuck\bin\tsplugin_inject.dll"
* Debug: registering processor plugin "inject", status: ok
* Debug: inject: ====> tsp -P inject xd.xml=100 --pid 20 --stuffing
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: inject: loading XML file xd.xml
* Debug: inject: loading XML file C:\TSDuck\bin\tsduck.tables.model.xml
* Debug: inject: adding standards DVB to none
* inject: loaded 1 sections from xd.xml, repetition rate: 100 ms
* Debug[2]: inject: schedule section: PID 0x0014, TID 0x70, TIDext 0x0000, section 0/0, cycle: 0, packet: 0, due packet: 0
* inject: target bitrate from repetition rates: 15,040 b/s
* Debug: file: detected TS file format TS
* Debug: file: initial buffer load: 44,620 packets, 8,388,560 bytes
* file: initial input bitrate is 8,222,556 b/s
* file: creating file TSTEST_with_tdt.ts
* Debug: control commands: found 1 packet processor plugins
* Debug: file: input thread started
* Debug: inject: packet processing thread started
* inject: transport bitrate: 8,222,556 b/s, packet interval: 547
* Debug: file: output thread started
* Debug: file: stopping the input plugin
* Debug: file: input thread terminated after 115,894 packets
* Debug: inject: packet processing thread terminated after 115,894 packets, 115,894 passed, 0 dropped, 0 nullified
* Debug: inject: stopping the plugin
* Debug: file: stopping the output plugin
* Debug: file: output thread terminated after 115,894 packets (115,894 output)
* Debug: Report logging thread terminated

```

What am I doing wrong? Can anyone point me in the right direction?

lelegard, 2025-05-28:
Hi @mkucmpro

There are three points in your question:

- Why is the TDT not inserted?
- What is a valid repetition rate for the TDT?
- How to adjust the content of the TDT on the fly?

The most common reason for not inserting anything is the absence of "free space" (i.e. stuffing, null packets) in the transport stream. See the [`tsp` reference documentation](https://tsduck.io/download/docs/tsduck.html#tsp-reference), especially the section named ["modifying, inserting and deleting packets"](https://tsduck.io/download/docs/tsduck.html#inserting-packets).

Here, I assume that your stream contains no stuffing at all.

The typical repetition rate of a TDT is 10 to 20 seconds. Using 100 ms is way too high. One packet is 188 * 8 = 1504 bits. If you want one TDT (one TS packet) every 10 seconds, the TDT bitrate is 150 b/s.

Finally, the TDT shall contain the current time. Each TDT section is consequently different. If you inject the same table content all the time, from a XML file, the time does not increase. The idea is to inject a fixed TDT with a dummy date and time and then update the time on the fly using the plugin `timeref`.

So, here is a complete example. Note that the table is so short and simple that we can even avoid creating a separate XML file and provide the XML text instead of the XML file name (TSDuck recognizes this automatically).

~~~
tsp --add-input-stuffing 1/10 \
    -I file in.ts \
    -P inject --pid 20 --bitrate 150 --stuffing "<?xml?><tsduck><TDT UTC_time='2000-01-01 00:00:00'/></tsduck>" \
    -P timeref --start 2025/05/01:08:57:00 \
    -P filter --negate --pid 0x1FFF \
    -O file out.ts
~~~

mkucmpro, 2025-05-28:
@lelegard Thank you for extremely detailed and helpful answer. Solved!