tsduck/tsduck #1636: [Q] tsfclean  PCR error

enricovittorini, 2025-05-31:
Version: 3.41-4299
Windows 10

when running tsfclean on a file, pcrverify return error. Am I doing something wrong?

```
tsp -I file test.ts -P pcrverify -O drop
* pcrverify: 603 PCR OK, 0 with jitter > 27,000 (1,000 micro-seconds), 1 unchecked
```

`tsfclean test.ts -o test_clean.ts`

```
tsp -I file test_clean.ts -P regulate -P pcrverify -O drop
 pcrverify: PID 0x0030 (48), PCR jitter: 947,659 = 35,098 micro-seconds = 387 packets + 116 bytes + 0 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 940,573 = 34,836 micro-seconds = 384 packets + 135 bytes + 1 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 947,659 = 35,098 micro-seconds = 387 packets + 116 bytes + 0 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 945,213 = 35,007 micro-seconds = 386 packets + 115 bytes + 7 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 947,659 = 35,098 micro-seconds = 387 packets + 116 bytes + 0 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 947,659 = 35,098 micro-seconds = 387 packets + 116 bytes + 0 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 945,213 = 35,007 micro-seconds = 386 packets + 115 bytes + 7 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 948,944 = 35,146 micro-seconds = 388 packets + 26 bytes + 6 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 945,213 = 35,007 micro-seconds = 386 packets + 115 bytes + 7 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 947,659 = 35,098 micro-seconds = 387 packets + 116 bytes + 0 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 947,659 = 35,098 micro-seconds = 387 packets + 116 bytes + 0 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 945,213 = 35,007 micro-seconds = 386 packets + 115 bytes + 7 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 947,659 = 35,098 micro-seconds = 387 packets + 116 bytes + 0 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 945,214 = 35,007 micro-seconds = 386 packets + 116 bytes + 0 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 946,498 = 35,055 micro-seconds = 387 packets + 26 bytes + 6 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 947,659 = 35,098 micro-seconds = 387 packets + 116 bytes + 0 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 945,213 = 35,007 micro-seconds = 386 packets + 115 bytes + 7 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 947,659 = 35,098 micro-seconds = 387 packets + 116 bytes + 0 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 947,659 = 35,098 micro-seconds = 387 packets + 116 bytes + 0 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 945,213 = 35,007 micro-seconds = 386 packets + 115 bytes + 7 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 947,659 = 35,098 micro-seconds = 387 packets + 116 bytes + 0 bits (tsp time)
* pcrverify: PID 0x0030 (48), PCR jitter: 945,214 = 35,007 micro-seconds = 386 packets + 116 bytes + 0 bits (tsp time)
```

lelegard, 2025-06-02:
Hi @enricovittorini 

The plugin `pcrverify` works well with constant bitrate only because if evaluates the difference (jitter) between each PCR and a hypothetical "perfect" clock. This perfect clock does not exist and the plugin needs to "invent" it. So, it uses the evaluated average bitrate and counts packets. This gives a clock.

The tool `tsfclean` turns a file into something that is a valid TS file from the beginning, using similar requirements as a HLS segment. It heavily transforms the beginning of the file: start with PSI cycles, drop video packets before the first intra-frame, drop audio and subtitles packets with a PTS lower than the PTS of the first video intra-frame. As a consequence, the file is "clean" but its leading part is no longer CBR (if the input was).

Now, when the plugin `pcrverify` analyses the start of the stream, the evaluated bitrate is incorrect (same clock, less packets). The bitrate is constantly re-evaluated, so it quickly stabilises. However, the positions of the first PCR's do not match the original input file. Since the hypothetical "perfect" clock starts with the first PCR, there is an initial jitter. 

In that case, we see that this initial jitter is 35 ms. Then, all PCR have the same 35 ms jitter. This means that they are all synchronised with each other, but they are all together shifted by 35 ms from the initial PCR due to the reorganisation of the start of the file by `tsfclean`.

The problem with verifying PCR is to verify compared to what? In a specialised tool working on real-time broadcast streams, the wall clock can be used as reference. By `tsp` works on undifferentiated streams - here a file - and there is no such external reference. The plugin tries to find one but it cannot work all the time. As a general rule, it works well on a stable stream (especially at the beginning) where it accurately detects jitters in the middle of the stream. Here, we have the exact opposite: a perfect stream in the middle (as the analysis of the input demonstrated) but unstable at the very beginning. Not the right use case...

enricovittorini, 2025-06-12:
thanks @lelegard 
FYI, the programming reference on the website is not working well anymore. search bar disappears and some pages return "No found"

lelegard, 2025-06-12:
> the programming reference on the website is not working well anymore

Thank you for reporting this. It's fixed. For some reason, the automatic update of the documentation failed. After rerunning it, it seems fixed. 

Every night, when some changes occurred in the TSDuck repo since the night before, the doc and binaries are regenerated in a scheduled GitHub workflow. In case of success, a trigger is sent to tsduck.io. In response, a PHP script in the tsduck.io server connects to GitHub to collect the artifacts of the latest run of the nightly build workflow. The binaries and documentation - including the huge set of doxygen files for the TSDuck Programming Reference - are published on tsduck.io. Looking at the [last log](https://github.com/tsduck/tsduck/actions/runs/15574582696/job/43859972280) for the nightly build, it seems that an Azure server in the GitHub infrastructure interrupted the download of the `doxy.zip` file, leading to a corrupted set of files.

I forced a rerun of the nightly build workflow. There was no error in the log and the online documentation seems fine.