tsduck/tsduck #1641: [BUG] Build failure: `tsUString.h: error: no matching function for call to 'ts::UString::erase(std::basic_string<char16_t>::const_iterator&)'`

barracuda156, 2025-06-05:
#### Bug description:
Build fails with numerous errors:
```
tsUString.h: error: no matching function for call to 'ts::UString::erase(std::basic_string<char16_t>::const_iterator&)'
```
etc.

#### How to reproduce:
Run the build :) But I do not know which configurations are affected. Looks like not only mine (see below).

#### Expected behavior:
The build should work.

#### Errors and logs:

~~~
[DEP] xml/tsxmlUnknown.cpp
  [CXX] crypto/tsAES128.cpp
  [CXX] system/tsAbortInterface.cpp
  [CXX] crypto/tsAES256.cpp
  [CXX] network/tsAbstractNetworkAddress.cpp
  [CXX] types/tsAbstractNumber.cpp
g++-mp-14: warning: Macintosh-Sergey-Fedorovs-Mac/objs-libtscore/tsxmlUnknown.o: linker input file unused because linking not done
g++-mp-14: error: Macintosh-Sergey-Fedorovs-Mac/objs-libtscore/tsxmlUnknown.o: linker input file not found: No such file or directory
  [CXX] algo/tsAbstractOutputStream.cpp
In file included from /opt/local/var/macports/build/_opt_local_ppcports_multimedia_tsduck/tsduck/work/tsduck-3.41-4299/src/libtscore/report/tsReport.h:16,
                 from network/tsAbstractNetworkAddress.h:16,
                 from network/tsAbstractNetworkAddress.cpp:9:
/opt/local/var/macports/build/_opt_local_ppcports_multimedia_tsduck/tsduck/work/tsduck-3.41-4299/src/libtscore/types/tsUString.h: In member function 'std::basic_string<char16_t>::iterator ts::UString::erase(std::basic_string<char16_t>::const_iterator)':
/opt/local/var/macports/build/_opt_local_ppcports_multimedia_tsduck/tsduck/work/tsduck-3.41-4299/src/libtscore/types/tsUString.h:2488:75: error: no matching function for call to 'ts::UString::erase(std::basic_string<char16_t>::const_iterator&)'
 2488 |         iterator erase(const_iterator position) { return SuperClass::erase(position); }
      |                                                          ~~~~~~~~~~~~~~~~~^~~~~~~~~~
In file included from /opt/local/include/gcc14/c++/bits/basic_string.h:57,
                 from /opt/local/include/gcc14/c++/string:54,
                 from /opt/local/var/macports/build/_opt_local_ppcports_multimedia_tsduck/tsduck/work/tsduck-3.41-4299/src/libtscore/cpp/tsPlatform.h:723,
                 from /opt/local/var/macports/build/_opt_local_ppcports_multimedia_tsduck/tsduck/work/tsduck-3.41-4299/src/libtscore/types/tsStringifyInterface.h:15,
                 from network/tsAbstractNetworkAddress.h:15:
/opt/local/include/gcc14/c++/bits/cow_string.h:1752:7: note: candidate: 'std::basic_string<_CharT, _Traits, _Alloc>& std::basic_string<_CharT, _Traits, _Alloc>::erase(size_type, size_type) [with _CharT = char16_t; _Traits = std::char_traits<char16_t>; _Alloc = std::allocator<char16_t>; size_type = long unsigned int]'
 1752 |       erase(size_type __pos = 0, size_type __n = npos)
      |       ^~~~~
/opt/local/include/gcc14/c++/bits/cow_string.h:1752:23: note:   no known conversion for argument 1 from 'std::basic_string<char16_t>::const_iterator' to 'std::basic_string<char16_t>::size_type' {aka 'long unsigned int'}
 1752 |       erase(size_type __pos = 0, size_type __n = npos)
      |             ~~~~~~~~~~^~~~~~~~~
/opt/local/include/gcc14/c++/bits/cow_string.h:1768:7: note: candidate: 'std::basic_string<_CharT, _Traits, _Alloc>::iterator std::basic_string<_CharT, _Traits, _Alloc>::erase(iterator) [with _CharT = char16_t; _Traits = std::char_traits<char16_t>; _Alloc = std::allocator<char16_t>; iterator = std::basic_string<char16_t>::iterator]'
 1768 |       erase(iterator __position)
      |       ^~~~~
/opt/local/include/gcc14/c++/bits/cow_string.h:1768:22: note:   no known conversion for argument 1 from '__normal_iterator<const char16_t*,[...]>' to '__normal_iterator<char16_t*,[...]>'
 1768 |       erase(iterator __position)
      |             ~~~~~~~~~^~~~~~~~~~
/opt/local/include/gcc14/c++/bits/cow_string.h:3406:6: note: candidate: 'std::basic_string<_CharT, _Traits, _Alloc>::iterator std::basic_string<_CharT, _Traits, _Alloc>::erase(iterator, iterator) [with _CharT = char16_t; _Traits = std::char_traits<char16_t>; _Alloc = std::allocator<char16_t>; iterator = std::basic_string<char16_t>::iterator]'
 3406 |      basic_string<_CharT, _Traits, _Alloc>::
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/local/include/gcc14/c++/bits/cow_string.h:3406:6: note:   candidate expects 2 arguments, 1 provided
/opt/local/var/macports/build/_opt_local_ppcports_multimedia_tsduck/tsduck/work/tsduck-3.41-4299/src/libtscore/types/tsUString.h: In member function 'std::basic_string<char16_t>::iterator ts::UString::erase(std::basic_string<char16_t>::const_iterator, std::basic_string<char16_t>::const_iterator)':
/opt/local/var/macports/build/_opt_local_ppcports_multimedia_tsduck/tsduck/work/tsduck-3.41-4299/src/libtscore/types/tsUString.h:2489:93: error: no matching function for call to 'ts::UString::erase(std::basic_string<char16_t>::const_iterator&, std::basic_string<char16_t>::const_iterator&)'
 2489 |         iterator erase(const_iterator first, const_iterator last) { return SuperClass::erase(first, last); }
      |                                                                            ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
/opt/local/include/gcc14/c++/bits/cow_string.h:1752:7: note: candidate: 'std::basic_string<_CharT, _Traits, _Alloc>& std::basic_string<_CharT, _Traits, _Alloc>::erase(size_type, size_type) [with _CharT = char16_t; _Traits = std::char_traits<char16_t>; _Alloc = std::allocator<char16_t>; size_type = long unsigned int]'
 1752 |       erase(size_type __pos = 0, size_type __n = npos)
      |       ^~~~~
/opt/local/include/gcc14/c++/bits/cow_string.h:1752:23: note:   no known conversion for argument 1 from 'std::basic_string<char16_t>::const_iterator' to 'std::basic_string<char16_t>::size_type' {aka 'long unsigned int'}
 1752 |       erase(size_type __pos = 0, size_type __n = npos)
      |             ~~~~~~~~~~^~~~~~~~~
/opt/local/include/gcc14/c++/bits/cow_string.h:1768:7: note: candidate: 'std::basic_string<_CharT, _Traits, _Alloc>::iterator std::basic_string<_CharT, _Traits, _Alloc>::erase(iterator) [with _CharT = char16_t; _Traits = std::char_traits<char16_t>; _Alloc = std::allocator<char16_t>; iterator = std::basic_string<char16_t>::iterator]'
 1768 |       erase(iterator __position)
      |       ^~~~~
/opt/local/include/gcc14/c++/bits/cow_string.h:1768:7: note:   candidate expects 1 argument, 2 provided
/opt/local/include/gcc14/c++/bits/cow_string.h:3406:6: note: candidate: 'std::basic_string<_CharT, _Traits, _Alloc>::iterator std::basic_string<_CharT, _Traits, _Alloc>::erase(iterator, iterator) [with _CharT = char16_t; _Traits = std::char_traits<char16_t>; _Alloc = std::allocator<char16_t>; iterator = std::basic_string<char16_t>::iterator]'
 3406 |      basic_string<_CharT, _Traits, _Alloc>::
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... 54 lines cut ...]
      |       ^~~~~~
/opt/local/include/gcc14/c++/bits/cow_string.h:1674:7: note:   candidate expects 3 arguments, 2 provided
/opt/local/include/gcc14/c++/bits/cow_string.h:1692:7: note: candidate: 'std::basic_string<_CharT, _Traits, _Alloc>::iterator std::basic_string<_CharT, _Traits, _Alloc>::insert(iterator, _CharT) [with _CharT = char16_t; _Traits = std::char_traits<char16_t>; _Alloc = std::allocator<char16_t>; iterator = std::basic_string<char16_t>::iterator]'
 1692 |       insert(iterator __p, _CharT __c)
      |       ^~~~~~
/opt/local/include/gcc14/c++/bits/cow_string.h:1692:23: note:   no known conversion for argument 1 from '__normal_iterator<const char16_t*,[...]>' to '__normal_iterator<char16_t*,[...]>'
 1692 |       insert(iterator __p, _CharT __c)
      |              ~~~~~~~~~^~~
/opt/local/var/macports/build/_opt_local_ppcports_multimedia_tsduck/tsduck/work/tsduck-3.41-4299/src/libtscore/types/tsUString.h: In member function 'std::basic_string<char16_t>::iterator ts::UString::insert(std::basic_string<char16_t>::const_iterator, std::basic_string<char16_t>::size_type, ts::UChar)':
/opt/local/var/macports/build/_opt_local_ppcports_multimedia_tsduck/tsduck/work/tsduck-3.41-4299/src/libtscore/types/tsUString.h:2534:99: error: no matching function for call to 'ts::UString::insert(std::basic_string<char16_t>::const_iterator&, std::basic_string<char16_t>::size_type&, ts::UChar&)'
 2534 |         iterator insert(const_iterator pos, size_type count, UChar ch) { return SuperClass::insert(pos, count, ch); }
      |                                                                                 ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~

~~~

#### Environment:
 - OS: macOS
 - OS version: 10.6.8
 - TSDuck full version: v3.41-4299
 - Installation type: MacPorts

#### Additional information:

Looks like the issue was reported ages ago on Linux: https://github.com/tsduck/tsduck/issues/125#issuecomment-421323588
But nothing was done about it.

barracuda156, 2025-06-05:
@lelegard I use gcc14, which supports C++20 and C++23. There is nothing about the OS specifically here, since gcc uses its own libstdc++.

The error could be related to C++ ABI, which is independent of C++ standard. It should be certainly possible to make the code compatible with both C++ ABIs, since both can be used with C++20, and nobody will switch to a different ABI just to build one specific library or app.

lelegard, 2025-06-05:
> I use gcc14, which supports C++20 and C++23. There is nothing about the OS specifically here, since gcc uses its own libstdc++.

The TSDuck makefiles for macOS are based on clang and libraries from Homebrew. So I assume that you had to change a number of things. 

The changes in GCC ABI are quite old, around GCC 4 or 5. I didn't notice any change since GCC 7 at least. With GCC 14, this is really old story and should't matter.

According to the error messages, the problems are in the API, not the ABI. This may be a language level issue. The TSDuck makefiles force C++20 standard. Make sure you do the same.

C++23 is also accepted but most compilers, gcc, clang and msvc, still have bugs with it and I had to remove it from the CI workflows. Previously, the CI did the tests in C++11 and C++14, then C++17 and C++20, then C++20 and C++23. But I removed C++23 because of numerous compiler bugs.

barracuda156, 2025-06-05:
@lelegard Well, let me see if I can reproduce the issue on a modern macOS. If it is only a C++ standard issue, this can work on my powerpc macOS – it has a newer C++ supported than most of macOS versions on x86.

P. S. TBH, it is a bit sad that a lot of upstreams assume that macOS == Homebrew. It is not. There are MacPorts and pkgsrc (NetBSD’s originally, but macOS and Linux are supported systems). There are reasons not to go with Homebrew, and if a user has a different package manager, it is undesirable to add another one.

barracuda156, 2025-06-06:
> This symbol is not user anywhere in the TSDuck build files or source code. So, there is no need to "not pass it", whatever it means.

@lelegard Toolchain may add it, as you could see from the empirical examples, including the earlier on Linux. MacPorts build system adds it with gcc and libstdc++.

> uname -m is definitely more appropriate

Given that the only thing what it does is to set a string used for directory name, it should really be just dropped. It is either useless or breaking.
But anyway, I simply mentioned that, it was not meant to be a request to fix the script. It is trivial for me to keep reinplace line in the port.

There are few minor issues with macOS code, otherwise this builds neatly (takes some time to, though):

[image]

lelegard, 2025-06-06:
Concerning `uname -m`, I will replace all spaces with a dash. It doesn't hurt for other platforms.

About the bin directory name, the host name and architecture are used so that the work area can be shared among several systems (typically a host and several virtual machines) and all binaries are created in distinct directories by default. Otherwise, you can just `make BINDIR=/some/where/else`.

About the build time, it can be awfully long, yes, there are [400,000 lines of C++ code](https://github.com/tsduck/tsduck/blob/master/METRICS.md) to compile. With 4 cores, use `make -j6` (or `-j10` if there use hardware threading). On a 8-core Apple M3, `make -j10` completes in 2 minutes.

About MacPort, whatever they setup can be a problem. Nowadays, MacPorts is rather deprecated (although still active). The mainstream package manager on macOS is now Homebrew, with a very active community and good automated processes (when I publish a new release of TSDuck on GitHub, the HomeBrew binaries automatically appears a couple of hours later). Whether you like or not, this is the one to go for current Macs. They no longer support PPC and old macOS versions, though.

barracuda156, 2025-06-06:
> Concerning `uname -m`, I will replace all spaces with a dash. It doesn't hurt for other platforms.

Thank you!

> Otherwise, you can just make BINDIR=/some/where/else.

This should do indeed.

> Whether you like or not, this is the one to go for current Macs.

IDK, I use MacPorts on macOS arm64 and it works fine. It is also well-supported for current macOS. (PowerPC support is nominal there, nobody tests anything, and since I gave up on fixing stuff in their upstream, a lot of ports will be broken. Still, it is better because it is configurable – you can just do more with MacPorts.) It probably also has more software than Brew, though specific software which a given user wants may be available here or there.