tsduck/tsduck #1645: [Q] tslsdvb display mac address

tasoss, 2025-06-17:
Hello.
I would like to ask if it is easy to display the mac address of the adapter in tslsdvb , somehow.
Thanks for the great work!

lelegard, 2025-06-18:
Hi @tasoss 

What kind of MAC address are you referring to? For which kind of device?

The command `tslsdvb` lists the broadcast receivers, tuners / demodulators, for DVB, ATSC and ISDB modulation standards. Broadcast receivers do not have a MAC address, simply because an "address" of whatever kind means a way to receive targeted unicast traffic, not broadcast.

tasoss, 2025-06-18:
Hello.
I am talking about dvb-s adapters.
For example (from dmesg)

> [Fri May 30 09:18:38 2025] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6908 DVB-S/S2 )...
[Fri May 30 09:18:38 2025] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
[Fri May 30 09:18:38 2025] TBSECP3 driver 0000:01:00.0: MAC address 00:22:ab:92:95:39

Thanks!

lelegard, 2025-06-18:
This is displayed by the TBSECP3 driver and specific to that TBS card. In the TBS device, I do not know how this MAC address is used, maybe for higher level protocols. We find MAC addresses in encapsulated protocols such as MPE and a few others. However, there is no concept of MAC address in DVB-S reception (or any other broadcast reception). That's why MAC address is an unknown concept for `tslsdvb`.

lelegard, 2025-06-19:
Hi @tasoss 

The [TBS User Guide](http://www.tbsdtv.com/download/document/common/tbs-quick-install-guide.zip) mentions that the MAC address in the TBS card is a "Virtual MPE Decoder Adapter".

That means that the card (or its driver) can extract a Multi-Protocol Encapsulation (MPE) stream from the received DVB-S transport stream and forwards it "as if it comes from an Ethernet adapter". The MAC address in the card is the one of this "virtual adapter". A MPE stream is a stream of IP/UDP packets which are encapsulated into TS packets, in a dedicated PID of DSM-CC sections.

This is the equivalent of the `mpe` plugin in TSDuck which extracts a MPE stream from the TS and forwards it as if it comes from the host system (which acts as an IP router in practice).

As I anticipated, this specific device implements "other" functions than pure DVB reception and its MAC address is dedicated to these other functions.

tasoss, 2025-06-19:
Hello.
Thank you very much for taking the time to read about it and letting me know.
Once more ,  thanks for the great work in this project!