tsduck/tsduck #1646: Feedback request on Grafana / InfluxDB

lelegard, 2025-06-21:
Hi TSDuck users,

I plan to add "some way" to log real time data to Grafana for visualisation.

This discussion is opened to collect feedback on the data to visualise and the way to send them to Grafana. Specifically, feedback from experienced users in Grafana would be welcome. As TSDuck users, you may not be one (I am not one either) but if some of your colleagues are, please encourage them to participate in this discussion and share theirs insights.

So far, the initial idea is:

- Add a plugin which periodically sends data to InfluxDB, used as data source for Grafana.
- Send bitrate information, defined as InfluxDB "measurement".
- Send raw bitrate data for whole TS, per service, per PID type (video, audio, PSI, etc), per PID, etc.

Open questions:
- Use any other kind of Grafana data source? Why and how to?
- Send any other measurement?
- Best way to structure data series?
- Recommendations on how to setup and operate Grafana, InfluxDB.
- Examples of useful InfluxDB queries to create useful visualisations on Grafana.
- Etc.

Thank you in advance for your feedbacks.

lelegard, 2025-06-29:
I have added a plugin named `influx` which sends various bitrate metrics to InfluxDB, which can be used as data source for Grafana.

In the TSDuck User's Guide, see:
- The reference for the [plugin `influx`](https://tsduck.io/download/docs/tsduck.html#_influx).
- A [sample usage with Grafana](https://tsduck.io/download/docs/tsduck.html#sample-grafana) with stacked data types (audio, video, etc) of a transport stream:

[image]

Any feedback is welcome, especially:

- Other metrics to send to InfluxDB.
- Other sample Grafana dashboards with nice and useful displays.
- Other data sources for Grafana.

lelegard, 2025-07-07:
Support for ETSI TR 101 290 has been added to plugin `influx`. See the [documentation of this plugin](https://tsduck.io/download/docs/tsduck.html#_influx) for more details.

Most logical errors from TR 101 290 are detected. Physical errors on the modulation cannot be detected in a TSDuck plugin.

Some users have expressed interest for TR 101 290. Please try it. Also please share your Grafana dashboards in the form of JSON export.

Sample data to Influxdb:
~~~
bitrate,scope=ts,tsid=4 value=22936796 1548161530912
counter,name=ts_sync_loss,severity=1 value=0 1548161530912
counter,name=sync_byte_error,severity=1 value=0 1548161530912
counter,name=pat_error,severity=1 value=0 1548161530912
counter,name=pat_error_2,severity=1 value=0 1548161530912
counter,name=continuity_count_error,severity=1 value=220 1548161530912
counter,name=pmt_error,severity=1 value=0 1548161530912
counter,name=pmt_error_2,severity=1 value=0 1548161530912
counter,name=pid_error,severity=1 value=0 1548161530912
counter,name=transport_error,severity=2 value=0 1548161530912
counter,name=crc_error,severity=2 value=0 1548161530912
counter,name=crc_error_2,severity=2 value=0 1548161530912
counter,name=pcr_error,severity=2 value=0 1548161530912
counter,name=pcr_repetition_error,severity=2 value=0 1548161530912
counter,name=pcr_discontinuity_indicator_error,severity=2 value=0 1548161530912
counter,name=pcr_accuracy_error,severity=2 value=0 1548161530912
counter,name=pts_error,severity=2 value=10 1548161530912
counter,name=cat_error,severity=2 value=0 1548161530912
counter,name=nit_error,severity=3 value=0 1548161530912
counter,name=nit_actual_error,severity=3 value=0 1548161530912
counter,name=nit_other_error,severity=3 value=0 1548161530912
counter,name=si_repetition_error,severity=3 value=0 1548161530912
counter,name=si_pid_error,severity=3 value=0 1548161530912
counter,name=buffer_error,severity=3 value=0 1548161530912
counter,name=unreferenced_pid,severity=3 value=0 1548161530912
counter,name=sdt_error,severity=3 value=0 1548161530912
counter,name=sdt_actual_error,severity=3 value=0 1548161530912
counter,name=sdt_other_error,severity=3 value=0 1548161530912
counter,name=eit_error,severity=3 value=0 1548161530912
counter,name=eit_actual_error,severity=3 value=0 1548161530912
counter,name=eit_other_error,severity=3 value=1 1548161530912
counter,name=eit_pf_error,severity=3 value=31 1548161530912
counter,name=rst_error,severity=3 value=0 1548161530912
counter,name=tdt_error,severity=3 value=0 1548161530912
counter,name=empty_buffer_error,severity=3 value=0 1548161530912
counter,name=data_delay_error,severity=3 value=0 1548161530912
counter,name=packet_count,severity=4 value=43220 1548161530912
counter,name=error_count,severity=4 value=262 1548161530912
~~~

lelegard, 2025-08-08:
Hi @bog 

What kind of PCR data would like to be logged?

InfluxDB is a database. It may be overkill to store all PCR/PTS/DTS values. Moreover, each call to InfluxDB is a web API, with a huge overhead. It is not appropriate to call it 10 or 20 times per second, with each timestamp on all services. It seems that the standard usage is one call every 2 or 5 seconds, typically to log statistics over the previous period. 

The plugin `influx` is in version 3.42, not yet released, but you can grab nightly builds here: https://tsduck.io/download/prerelease/

lelegard, 2025-08-13:
Hi @bog 

> Updating the values (PTS/DTS/PCR) for a given stream would be the ideal.

This is done. Options `--pcr`, `--pts`, `--dts` have been added to report the last value of the corresponding timestamp in each measurement period. The selection can be by PID or for services (using the declared PCR PID of the service).

lelegard, 2025-09-13:
Thank you @bugre, this is exactly the kind of feedback which is useful to improve the product.

If the segmentation using distinct buckets is not appropriate, there are several possible improvements. The first two are trivial. The third one needs slightly more work - although very doable.

First, for all types of metrics, we can add a generic option `--tag name=value`, with several occurrences if necessary. In that case, for that occurrence of plugin `influx`, we add several user-specified tags. This is not limited to TR 101 290.

For instance, for the current metrics:
~~~
counter,name=continuity_count_error,severity=1 value=220 1548161530912
~~~

with `-P influx ... --tag region=foo --tag operator=bar`, we would get this:
~~~
counter,name=continuity_count_error,severity=1,region=foo,operator=bar value=220 1548161530912
~~~

This would let you identify metrics of a precise origin using InfluxDB filters.

Second, specific to TR 101 290, we can systematically add the TS id in each metrics:
~~~
counter,name=continuity_count_error,severity=1,tsid=4 value=220 1548161530912
~~~

This would let you identify the TS when several occurrences of plugin `influx` log error counters for different TS.

Third, specific to TR 101 290 again, we can add scoping information, for instance:
~~~
counter,name=continuity_count_error,severity=1,scope=ts,tsid=4 value=220 1548161530912
counter,name=continuity_count_error,severity=1,scope=service,service=M6 value=220 1548161530912
counter,name=continuity_count_error,severity=1,scope=pid,pid=110 value=180 1548161530912
counter,name=continuity_count_error,severity=1,scope=pid,pid=120 value=40 1548161530912
~~~

This requires some more work. Moreover, it is incompatible with the exiting implementation. Currently, a filter `name=continuity_count_error` would retun one value with 220 errors. With the new implementation, it would return 4 counters with a total of 660 errors, which is obviously incorrect.

So, this cannot be the default behaviour. It would be activated using some specific option. When this option is used, the InfluxDB filter would need to be modified to use the appropriate criteria.

bugre, 2025-09-14:
Thank you for the option evaluation @lelegard .

> If the segmentation using distinct buckets is not appropriate, there are several possible improvements. The first two are trivial. The third one needs slightly more work - although very doable.

It will depend a lot on how one implements the solution. But i think most situation, we'll have a more "global" influx instance, sometimes, not even managed by us, where we get a specific bucket created and assigned based on a service request. And in this case, being able to report to the same bucket, with a way to differentiate metrics for different "monitoring" will be necessary.

> First, for all types of metrics, we can add a generic option --tag name=value, with several occurrences if necessary. In that case, for that occurrence of plugin influx, we add several user-specified tags. This is not limited to TR 101 290.

I would start with this option, it will allow to group all metrics that relate to each other more directly. 

> Second, specific to TR 101 290, we can systematically add the TS id in each metrics:
Systematically adding ts_id to each TR 101 290 metric, would also ensure those metrics have some correlation context and so avoid having two different origins being mixed up.

And for the third option, i would suggest to wait until we have some more experience and maybe some more imput on use-cases and if it really is needed. For now, if we need pid information on CCE for example we can add the continuity plugin and get it more detailed info.

lelegard, 2025-09-14:
The first two options have just been added.

The option `--tag` is applicable to all metrics. Sample metrics with `-P influx --tag operator=tvsat --tag satellite=sat1`:
~~~
bitrate,scope=ts,tsid=4,operator=tvsat,satellite=sat1 value=22936796 1548161530912
~~~

With `--tr-101-290`, we use a `scope` tag as with bitrate metrics. Its only possible value for now is `ts` and the secondary tag `tsid` gives the TS id. More scopes can be added later, with their corresponding secondary tags. I do not know if TS 101 290 allows to report error counters in a subset of the TS, service or PID, but as long as it is useful, this is fine for me.

Example:
~~~
counter,name=ts_sync_loss,severity=1,scope=ts,tsid=4 value=0 1548161530912
counter,name=sync_byte_error,severity=1,scope=ts,tsid=4 value=0 1548161530912
counter,name=pat_error,severity=1,scope=ts,tsid=4 value=0 1548161530912
~~~

lelegard, 2025-09-15:
I have added the third solution. When using `--tr-101-290` with `--services`, `--pid`, or `--all-pids`, the counters are also reported by PID or service.

Note the following:

- All global counters (`scope=ts`) are always sent, even if zero.
- The specific counters with `scope=pid` or `scope=service` are sent only when non-zero. This is done to reduce the amount of logged data.
- When using `--services`, `--pid`, or `--all-pids`, it is therefore recommended to also use `--max-severity 3`. Otherwise, the informational counters (severity 4) are also included and there are not zero (`packet_count` for instance).
- The tag `tsid` has been unconditionally added to all metrics to differentiate data from distinct TS which are logged in the same bucket.
- As a consequence, to filter data for complete TS, also add the filter `scope=ts`. Filtering only on `tsid=XXX` would return all metrics, including for PID's or services.

The [user's guide](https://tsduck.io/download/docs/tsduck.html#_influx) has been updated accordingly (will be updated in the next nightly build).

bugre, 2025-09-18:
Quick question, does it make sense to have the "v2" metrics of pat_error and pmt_error? On all feeds that i've monitored and that did trigger this counter increase they always increased together, presenting the same values.

I'm still experimenting and working on dashboards, but i've one for the tr-101-290 that i consider usable. Here is a screenshot.

The 3 dropdown i've edited so they all show, allow to select a specific channel, severity level a if needed a specific metric from this severity.

[image]

I've a influx with grafana running as containers, and do run individual tsduck containers for each channel i'm monitoring (at least for now).

```
## example of tsduck command used.
CH=channelName; MCAST=239.121.121.121:1234;
docker run -d --name ${CH} --net=host --restart unless-stopped bugre/tsduck tsp -v -I ip ${MCAST} -P influx --host-url "influx-url" --bucket "tsmetrics" --tag channel=${CH} --token "influx-token" --org "influx-organization" --tr-101-290 -O drop
```

The influx query isn't very elaborated yet, but with a few adjustments of the dashboard settings it works.
```
import "strings"

channels = if "${channel}" == "/.*/" then [] else strings.split(v: "${channel}", t: ",")

from(bucket: "tsmetrics")
  |> range(start: v.timeRangeStart, stop: v.timeRangeStop)
  |> filter(fn: (r) => r.severity == "${severity}" )
  // |> filter(fn: (r) => r._field == "value")
  |> filter(fn: (r) => "${counter}" == "All"  or r.name == "${counter}"  )
  |> filter(fn: (r) => "${channel}" == "/.*/" or contains(value: r.channel, set: channels))
  |> map(fn: (r) => ({ r with _field: r.name }))
  |>  group(columns: ["_field"])  // cleaner legend info.
  |> aggregateWindow(every: v.windowPeriod, fn: sum, createEmpty: false)
  
``` 
The Dashboard:

It's available on grafana.com as id: 24091 (link -- https://grafana.com/grafana/dashboards/?pg=docs-grafana-latest-dashboards-share-dashboards-panels&search=tr-101-290 -- isn't working yet, as i've just uploaded it: )

Meanwhile i'll also share dashboard json here.  Hope it's ok to share it here @lelegard 
[TSDuck - TR 101 290 Metrics-1758231600221.json](https://github.com/user-attachments/files/22415667/TSDuck.-.TR.101.290.Metrics-1758231600221.json)

Comments, fixes and improvements area welcome.

makaroshkj, 2025-09-22:
@lelegard Thanks for that feature since it no longer requires using custom json -- prometheus exporters. I've already tested this feature for monitoring udp/spts streams, and everything works perfectly. I have a couple of questions:
* Are there any plans to add `scope=pid` for `continuity_count_error` counter?
* Are there any plans to implement a solution for monitoring Inter-packet Arrival Time (IAT)? - This can be useful for monitoring ts/udp streams and assessing network jitter.

lelegard, 2025-09-22:
@makaroshkj 

> Are there any plans to add scope=pid for continuity_count_error counter?

It is already there if you use `--all-pids`. By default, `--ts-101-290` only reports global TS counters (`scope=ts`).

Look at the last example in the [plugin influx documentation](https://tsduck.io/download/docs/tsduck.html#_etsi_tr_101_290_error_counter_metrics):
~~~
counter,name=continuity_count_error,severity=1,scope=ts,tsid=4 value=220 1548161530912
counter,name=continuity_count_error,severity=1,scope=service,tsid=4,service=1025 value=44 1548161530912
counter,name=continuity_count_error,severity=1,scope=service,tsid=4,service=1026 value=40 1548161530912
counter,name=continuity_count_error,severity=1,scope=service,tsid=4,service=1031 value=42 1548161530912
counter,name=continuity_count_error,severity=1,scope=service,tsid=4,service=1045 value=45 1548161530912
counter,name=continuity_count_error,severity=1,scope=service,tsid=4,service=1046 value=38 1548161530912
counter,name=continuity_count_error,severity=1,scope=service,tsid=4,service=global value=11 1548161530912
counter,name=continuity_count_error,severity=1,scope=pid,tsid=4,pid=0 value=1 1548161530912
counter,name=continuity_count_error,severity=1,scope=pid,tsid=4,pid=18 value=3 1548161530912
counter,name=continuity_count_error,severity=1,scope=pid,tsid=4,pid=100 value=1 1548161530912
counter,name=continuity_count_error,severity=1,scope=pid,tsid=4,pid=120 value=13 1548161530912
counter,name=continuity_count_error,severity=1,scope=pid,tsid=4,pid=130 value=13 1548161530912
counter,name=continuity_count_error,severity=1,scope=pid,tsid=4,pid=132 value=6 1548161530912
counter,name=continuity_count_error,severity=1,scope=pid,tsid=4,pid=140 value=2 1548161530912
counter,name=continuity_count_error,severity=1,scope=pid,tsid=4,pid=170 value=3 1548161530912
~~~

You can have the counter by TS, service and/or PID.

> Are there any plans to implement a solution for monitoring Inter-packet Arrival Time (IAT)? 

It could be added, depending on the definition of that metrics. I didn't see it in TR 101 290. Do you have a precise definition for it? What should be precisely reported?

makaroshkj, 2025-09-29:
@lelegard I've just tested the latest tsduck version, and CC counter works with `scope=pid` as expected. Many hanks.

We consider IAT as a maximum between UDP packets arrival:

[image]

I couldn't find any specs for IAT, but it's often discussed in conjunction with the MDI described in the [RFC 4445](https://www.rfc-editor.org/rfc/rfc4445.html).
Actually, it's not part of the TR 101290, and it seems it is only make sense in the network-level monitoring. So I'm not sure tsduck should support this.

One more question. We currently use a splicemonitor plugin to parse scte35 events. We then export the event_type/start_time/duration in the prometheus format to display each event on the timeline. This helps us detect when scte35 events disappear from input streams. Are there any plans to add this for export to influx as well?

makaroshkj, 2025-12-10:
@lelegard Understood. Thanks for the clarification. I can’t to split the bond into individual interfaces for now since it requires maintenance. I will find the another server without bonding and check it out. Will keep you posted.